m-labs/artiq #1714: Vivado version

jbqubit, 2021-07-06:
Please add the recommended version of Vivado to the [release notes](https://github.com/m-labs/artiq/blob/master/RELEASE_NOTES.rst) for tagged releases. Going forward this will enable the community to reliably build older versions of ARTIQ. At the top of the release notes please add a note about the Vivado version for the development branch. 

This was [discussed](https://github.com/m-labs/artiq/issues/910) some time ago. It remains difficult to parse the [Hydra build logs](https://nixbld.m-labs.hk/jobset/artiq/board-generated) to determine which version of Vivado M-Labs is using.

sbourdeauducq, 2021-07-06:
Seriously, please just see nix-scripts. Hydra gives you its commit ID (e.g. ``nixScripts → 591ef2b``) and then it is not difficult to determine the Vivado version:
https://git.m-labs.hk/M-Labs/nix-scripts/src/commit/591ef2b/artiq-fast/vivado.nix#L4
You can even derive the version of every single piece of software involved (not just Vivado), based on the Hydra info.

closed 2021-07-06 by sbourdeauducq

dnadlinger, 2021-07-07:
Perhaps we should just a tutorial like this to the docs somewhere, if it doesn't already exist (stating that there are many dependencies, of which Vivado tends to be critical, and then giving the above steps). It's a bit hard to know where to look if you are a Nix-oblivious user – figuring out the version by glancing at the derivations in the tree is easy after that.