m-labs/artiq #1719: satellite hangs at I2C initialization with `SDA is stuck low`

jbqubit, 2021-07-13:
## One-Line Summary

Kasli `satellite` runtime hangs at I2C initialization with `SDA is stuck low` while `master` is fine.

## Issue Details

### Steps to Reproduce

Here's the satellite .json that fails and transcript of failure. 
```
{
    "target": "kasli",
    "min_artiq_version": "6.0",
    "variant": "brittonlab-laserlab-right",
    "hw_rev": "v1.1",
    "base": "satellite",
    "peripherals": [
        {
            "type": "dio",
            "ports": [0],
            "bank_direction_low": "output",
            "bank_direction_high": "output"
        }
   ]
}

```

```
$ artiq_flash -t kasli start; flterm /dev/ttyUSB2
Open On-Chip Debugger 0.10.0-snapshot (2021-07-12-19:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 25000 kHz
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 25000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
Info : gdb server disabled
TEMP 55.09 C
VCCINT 0.997 V
VCCAUX 1.783 V
VCCBRAM 1.000 V
VPVN 0.000 V
VREFP 0.000 V
VREFN 0.000 V
VCCPINT 0.000 V
VCCPAUX 0.000 V
VCCODDR 0.000 V
Bootloader CRC passed
Gateware ident 7.7663.51d2861e.beta;brittonlab-laserlab-right
Initializing SDRAM...
Read leveling scan:
Module 1:
00000000000111111111110000000000
Module 0:
00000000000111111111111000000000
Read leveling: 16+-5 16+-6 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000009s]  INFO(runtime): ARTIQ runtime starting...
[     0.003932s]  INFO(runtime): software ident 7.7663.51d2861e.beta;brittonlab-laserlab-right
[     0.012220s]  INFO(runtime): gateware ident 7.7663.51d2861e.beta;brittonlab-laserlab-right
[     0.020500s]  INFO(runtime): log level set to INFO by default
[     0.026229s]  INFO(runtime): UART log level set to INFO by default
panic at src/libcore/result.rs:945:5: I2C initialization failed: "SDA is stuck low and doesn\'t get unstuck"
backtrace for software version 7.7663.51d2861e.beta;brittonlab-laserlab-right:
0x400321c8
0x4002341c
0x4000362c
0x400296a0
0x40031908
halting.
use `artiq_coremgmt config write -s panic_reset 1` to restart instead
```

Here's the master .json that works and transcript of success. 
```
{
    "target": "kasli",
    "min_artiq_version": "6.0",
    "variant": "brittonlab-laserlab-right",
    "hw_rev": "v1.1",
    "base": "master",
    "peripherals": [
        {
            "type": "dio",
            "ports": [0],
            "bank_direction_low": "output",
            "bank_direction_high": "output"
        }
   ]
}
```

```
$ artiq_flash -t kasli start; flterm /dev/ttyUSB2
Open On-Chip Debugger 0.10.0-snapshot (2021-07-12-19:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 25000 kHz
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 25000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
Info : gdb server disabled
TEMP 49.99 C
VCCINT 0.993 V
VCCAUX 1.778 V
VCCBRAM 0.998 V
VPVN 0.000 V
VREFP 0.000 V
VREFN 0.000 V
VCCPINT 0.000 V
VCCPAUX 0.000 V
VCCODDR 0.000 V

 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2021 M-Labs Limited

Bootloader CRC passed
Gateware ident 7.7666.2204fd2b.beta;brittonlab-laserlab-right
Initializing SDRAM...
Read leveling scan:
Module 1:
00000000001111111111110000000000
Module 0:
00000000000111111111110000000000
Read leveling: 15+-6 16+-5 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000009s]  INFO(runtime): ARTIQ runtime starting...
[     0.003931s]  INFO(runtime): software ident 7.7666.2204fd2b.beta;brittonlab-laserlab-right
[     0.012219s]  INFO(runtime): gateware ident 7.7666.2204fd2b.beta;brittonlab-laserlab-right
[     0.020499s]  INFO(runtime): log level set to INFO by default
[     0.026228s]  INFO(runtime): UART log level set to INFO by default
[     0.032607s]  INFO(runtime::rtio_clocking): using internal RTIO clock (by default)
[     0.311206s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[     2.649734s]  INFO(board_artiq::si5324):   ...locked
[     2.680565s]  INFO(runtime): network addresses: MAC=80-1f-12-47-a8-3d IPv4=192.168.1.70 IPv6-LL=fe80::821f:12ff:fe47:a83d IPv6=no configured address
[     2.695690s]  WARN(board_artiq::drtio_routing): could not read routing table from configuration, using default
[     2.704443s]  INFO(board_artiq::drtio_routing): routing table: RoutingTable { 0: 0; 1: 1 0; 2: 2 0; }
[     2.718282s]  INFO(runtime::mgmt): management interface active
[     2.732594s]  INFO(runtime::session): accepting network sessions
[     2.748008s]  INFO(runtime::session): running startup kernel
[     2.752472s]  INFO(runtime::session): no startup kernel found
[     2.758297s]  INFO(runtime::session): no connection, starting idle kernel
[     2.765116s]  INFO(runtime::session): no idle kernel found
[     2.770544s]  INFO(runtime::rtio_mgt::drtio): [DEST#0] destination is up
```

No other changes to the setup between flashing satellite and master. This reproduces on two Kasli with no EEM peripherals connected. Kasli is v1.1. ARTIQ is tracking m-labs master. Build environment is 

```$ nix-shell -I artiqSrc=artiq nix-scripts/artiq-fast/shell-dev.nix```

sbourdeauducq, 2021-07-13:
Try flashing one of the existing satellite variants from Hydra.

jbqubit, 2021-07-13:
We've tracked down the origin of the above confusing results. My json file specifies ```"variant", "brittonlab-laserlab-right"``` and I switched between ```"base": "satellite" ``` and ```"base": "master"``` while debugging my crates. The ```artiq.gateware.kasli_generic.py``` builds based on the json ```base``` specifier and puts the results in a directory named after the json ```variant``` -- my build folder contained both master and satellite build outputs. `artiq_flash.py` flashes different files based on the name of the auto-generated build directory. `artiq_flash.py` branches as follows based on the ```variant``` argparse.
- if argparse variant string ends `satellite` it looks for files corresponding to the `satellite` variant
- if argparse variant string ends in `master` it looks for files corresponding to the `master` variant
- else it looks for files corresponding to the `standalone` variant

https://github.com/m-labs/artiq/blob/6ce9c26402e62e44797b9398c30f872a2074d3dc/artiq/frontend/artiq_flash.py#L449

The broader issue is that there is not consistency across the build-flash stack for specifying {satellite, master, standalone}. It also looks like there's mangling of two concepts.
1) ARTIQ core device variants: satellite, master, standalone
2) end-user core device configuration variants: mycrate1.json, mycrate2.json

sbourdeauducq, 2021-07-14:
> `artiq_flash.py` branches as follows based on the `variant` argparse.
>
>     * if argparse variant string ends `satellite` it looks for files corresponding to the `satellite` variant
>
>     * if [...]

No, it does not. And the only part relevant to this problem here is:

```python
            if variant.endswith("satellite"):
                firmware = "satman"
            else:
                firmware = "runtime"
```

This could perhaps be changed to code that checks which of ``satman`` or ``runtime`` is present, and errors out if both are present.

The other code is for handling the Sayma special case with the RTM.

closed 2021-07-14 by sbourdeauducq

jbqubit, 2021-07-15:
Thank you. I like the added guidance "more than one firmware file, please clean up your build directory. "

commit 4fb8ea5b (2021-07-14): artiq_flash: determine which firmware to flash by looking at filesystem

Closes #1719