m-labs/artiq #1726: Random Kasli v2.0 panic

sotirova, 2021-07-19:
We are currently getting frequent (a couple per day) random panics from a Kasli 2.0 while running experiments that have been running successfully for a few months. The UART output is as follows:

```
@ 0x000060                                                                                                           
+0000: 00000000 20000000 30008001 00000007                                                                           
+0010: 20000000 20000000 3001a001 99aab749                                                                          
+0020: 30026001 00000000 30012001 02423fe5                                                                          
+0030: 3001c001 00000000 30018001 03631093                                                                           
@ 0x4015b7dc                                                                                                         
+0000: 00000060 00000000 40009eb0 4002c3ec                                                                           
+0010: b0001822 0000001f 00003b75 00000001                                                                           
+0020: 4002ef48 00000015 00000ab3 0000000a                                                                           
+0030: 4016f2d0 00000008 40161dc4 4015fa2d                                                                           
panic at runtime/main.rs:290:13: exception IllegalInsn at PC 0x60, EA 0x4015b7dc                                     
backtrace for software version 6.0.dev+73.g7b931c3d;trapmaster:                                                      
0x40030334                                                                                                           
0x400241fc                                                                                                           
0x4002fbec                                                                                                           
0x400010d0                                                                                                           
halting.                                                                                                            
use `artiq_coremgmt config write -s panic_reset 1` to restart instead 
```

This is normally fixed by rebooting the kasli. It doesn't seem deterministic as it has been caused by different experiments, including very trivial ones, e.g. switching on an Urukul channel.

sbourdeauducq, 2021-07-20:
Did your gateware meet timing?

sotirova, 2021-07-20:
Yes, the timing passed

sbourdeauducq, 2022-08-29:
Is this still happening with ARTIQ-7?

dnadlinger, 2022-12-18:
Closing this as effectively non-reproducible on RISC-V (how would one know it's the same crash?).

Of course, please open a new issue for any panics still occurring (with the runtime.elf for the gateware attached, or the backtrace symbolized).

closed 2022-12-18 by dnadlinger