m-labs/artiq #1727: MonInj can't handle >64 TTL channels

lriesebos, 2021-07-21:
# Bug Report

## One-Line Summary

MonInj seems to break when we have more than 64 TTL channels to monitor, see https://forum.m-labs.hk/d/207-moninj-failure-and-disabled-outputs .

## Issue Details

### Steps to Reproduce

1. Step 1. Flash gateware with more than 64 TTL channels (e.g. [this json file](https://git.m-labs.hk/M-Labs/sinara-systems/src/branch/master/duke6.json) with just over 64 TTL channels including the TTL channels used for the Urukul boards)
2. Step 2. Start an artiq session
3. Step 3. Get `struct.error: unpack requires a buffer of 6 bytes`

### Expected Behavior

All TTL channels work fine with MonInj, both for Kasli and KC705.

### Actual (undesired) Behavior

Both for Kasli and KC705, the error shown below occurs when starting the dashboard which initiates the MonInj connection. TTL outputs of the device become unreliable, rendering the coredevice useless. When MonInj is not started (i.e. using `artiq_run`), the gateware runs fine. I would expect this is some addressing limitation somewhere, causing a 6 bit address to overflow, but I am not familiar enough with the firmware/gateware to point out the exact lines.

```
[nix-shell:~/red-chamber]$ artiq_dashboard 

(process:23064): Gtk-WARNING **: 16:41:29.782: Locale not supported by C library.
	Using the fallback 'C' locale.
Gtk-Message: 16:41:29.844: Failed to load module "canberra-gtk-module"
Gtk-Message: 16:41:29.845: Failed to load module "canberra-gtk-module"
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
No XVisualInfo for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
Falling back to using screens root_visual.
INFO:dashboard:root:ARTIQ dashboard 6.7602.ec4270fb connected to ::1
ERROR:dashboard:artiq.dashboard.moninj:lost connection to core device moninj
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/nix/store/35pv6fwf7xykj8dqym8p5rjp5jk0lxi6-python3-3.8.8-env/lib/python3.8/site-packages/artiq/dashboard/moninj.py", line 472, in stop
    await self.dm.close()
  File "/nix/store/35pv6fwf7xykj8dqym8p5rjp5jk0lxi6-python3-3.8.8-env/lib/python3.8/site-packages/artiq/dashboard/moninj.py", line 421, in close
    await asyncio.wait_for(self.core_connector_task, None)
  File "/nix/store/qy5z9gcld7dljm4i5hj3z8a9l6p37y81-python3-3.8.8/lib/python3.8/asyncio/tasks.py", line 455, in wait_for
    return await fut
  File "/nix/store/35pv6fwf7xykj8dqym8p5rjp5jk0lxi6-python3-3.8.8-env/lib/python3.8/site-packages/artiq/dashboard/moninj.py", line 396, in core_connector
    await self.core_connection.close()
  File "/nix/store/35pv6fwf7xykj8dqym8p5rjp5jk0lxi6-python3-3.8.8-env/lib/python3.8/site-packages/artiq/coredevice/comm_moninj.py", line 54, in close
    await asyncio.wait_for(self._receive_task, None)
  File "/nix/store/qy5z9gcld7dljm4i5hj3z8a9l6p37y81-python3-3.8.8/lib/python3.8/asyncio/tasks.py", line 455, in wait_for
    return await fut
  File "/nix/store/35pv6fwf7xykj8dqym8p5rjp5jk0lxi6-python3-3.8.8-env/lib/python3.8/site-packages/artiq/coredevice/comm_moninj.py", line 91, in _receive_cr
    channel, override, value = struct.unpack(
struct.error: unpack requires a buffer of 6 bytes
```

### Your System (omit irrelevant parts)

* Operating System: Ubuntu 20.04
* ARTIQ version: ARTIQ 6
* Hardware involved: Same for KC705 and Kasli

sbourdeauducq, 2021-07-21:
@SidaChen1999 Is there anything in the core device log (Kasli UART) when the problem occurs?

SidaChen1999, 2021-07-22:
Nope, there is only an info: `INFO(runtime::moninj): new connection from 192.168.1.111:35756`

PR #1729 (2021-07-22): moninj: change read to readexactly to avoid unpack failure

# ARTIQ Pull Request

## Description of Changes 
Use `readexactly` so that `struct.unpack` will always have the right length to unpack. 

### Related Issue
Closes #1727

## Type of Changes

|   | Type |
| ------------- | ------------- |
| ✓  | :bug: Bug fix  |

## Steps (Choose relevant, delete irrelevant before submitting)

### All Pull Requests

- [x] Use correct spelling and grammar.
- [x] Close/update issues.

### Code Changes

- [x] Run `flake8` to check code style (follow PEP-8 style). `flake8` has issues with parsing Migen/gate [... cut]
closed 2021-07-22 by sbourdeauducq