m-labs/artiq #1728: List corruption in kernel list assignment with slicing

steine01, 2021-07-21:
# Bug Report

## One-Line Summary

When assigning a large list to another large list in a kernel function and slicing the list being assigned, the new list is corrupted after a certain index.

## Issue Details

### Steps to Reproduce

1. Execute the following experiment:

```
class ListCorruption(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.list_a = [1]*100
        self.list_b = [0]*100
        
    def run(self):
        self.copy_list()
        print(self.list_a)
        print(self.list_b)
        
    @kernel
    def copy_list(self):
        self.core.reset()
        self.list_b = self.list_a[:]
```

### Expected Behavior

Either `list_a` is equal to `list_b` elementwise or an error is raised (as it is with `self.list_b = self.list_a.copy()`).

### Actual (undesired) Behavior

`list_b` is equal to `list_a` only up to a certain index, afterwards there appear large semi-random numbers. This is an example output of the above experiment:

```
INFO:worker(41939,tests.py):print:[1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1]
INFO:worker(41939,tests.py):print:[1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1082186704, 1342176952, 1342176984, 1342177016, 1, 1342176876, 0, 1082186704, 1342176980, 204, 1, 2, 1342176804, 1082525612, 1, 1082189636, 1677747200, 64, -2105183231, 1342176596, 0, 0, 0, 1082295668, 1082144748, 1342177060, 4, 1342177192, 1077940224, 1082525609, 2, 0, 24, 1082137176, 0, 1342177184, 151258256, 1082525612, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1077936132, 4092, 11, 1082525381, 0, 234946561, 0, 1077940228, 4092, 418, 1082525613]
```

This seems similiar to issue #1394 but there is no RPC involved.

### Your System (omit irrelevant parts)

* Operating System: Windows 10
* ARTIQ version: ARTIQ v6.7480.a97b4633.beta

pca006132, 2021-07-28:
I think this is not list corruption, but a problem due to unsound escape analysis.

It seems that the generated code is not copying the content of `list_a[:]` into `list_b`, but rather *allocates* a temporary list, store the slice result in the list, and assign the list to `list_b`. As the list is allocated on the stack, this is a dangling pointer after the function returns, and causes the corruption. It should be correct if you write `list_b[:] = list_a[:]` instead.

It seems to me that the easiest way to fix this would be to forbid such assignment by fixing the escape analysis.

steine01, 2021-07-28:
If I change the last line to the one you suggest (`list_b[:] = list_a[:]`), the error disappears. Increasing the length of the list (up to 10000 elements) also appears to be no issue then.