m-labs/artiq #1730: Kasli v2.0 <-> v1.1 DRTIO ERRORs

pmldrmota, 2021-07-26:
# Bug Report

## One-Line Summary

DRTIO link from Master Kasli v2.0 to Satellite Kasli v1.1 breaks down after few minutes of operation.

## Issue Details

### First example
Master UART log:
```
[  3410.855634s] ERROR(runtime::rtio_mgt::drtio): [LINK#0] error(s) found (0x01):
[  3410.861591s] ERROR(runtime::rtio_mgt::drtio): [LINK#0] received packet of an unknown type
[  3410.870582s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] communication failed (link went down)
[  3410.911220s]  INFO(runtime::session): no connection, starting idle kernel
[  3411.079467s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link is down
[  3411.084747s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] destination is down
[  3490.817225s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link RX became up, pinging
[  3497.472802s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] remote replied after 33 packets
[  3497.599241s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link initialization completed
[  3497.606322s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] destination is up
[  3497.612452s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] buffer space is 128
```
Satellite UART log shows no errors before rebooting automatically at around t=3480s above, after which the system was functional again.
```
[  2629.468080s]  INFO(satman): resetting RTIO                
                                                                                                             
 __  __ _ ____         ____                                                                                  
|  \/  (_) ___|  ___  / ___|                                                                                 
| |\/| | \___ \ / _ \| |                                                                                     
| |  | | |___) | (_) | |___                                                                                  
|_|  |_|_|____/ \___/ \____|                                                                                 
                                                                                                             
MiSoC Bootloader                              
Copyright (c) 2017-2021 M-Labs Limited        
                                              
Bootloader CRC passed
Gateware ident 7.0.dev+101.ga8579bb7;satellite
Initializing SDRAM...
Read leveling scan:
Module 1:
00000000000011111111111100000000              
Module 0:
00000000000011111111111100000000
Read leveling: 17+-6 17+-6 done 
SDRAM initialized     
Memory test passed    

Booting from flash... 
Starting firmware.    
[     0.000004s]  INFO(satman): ARTIQ satellite manager starting...
[     0.005615s]  INFO(satman): software ident 7.0.dev+101.ga8579bb7;satellite
[     0.012505s]  INFO(satman): gateware ident 7.0.dev+101.ga8579bb7;satellite
[     0.290649s]  INFO(board_artiq::si5324): waiting for Si5324 lock...       
[     2.272202s]  INFO(board_artiq::si5324):   ...locked                      
[     2.456857s]  INFO(satman): uplink is up, switching to recovered clock    
[     2.489985s]  INFO(board_artiq::si5324): waiting for Si5324 lock...       
[     4.253701s]  INFO(board_artiq::si5324):   ...locked                      
[     8.686028s]  INFO(board_artiq::si5324::siphaser): calibration successful, lead: 395, width: 437 (351deg)
[     9.020074s]  INFO(satman): TSC loaded from uplink
[     9.129203s]  INFO(satman): rank: 1
[     9.131309s]  INFO(satman): routing table: RoutingTable { 0: 0; 1: 1 0; 2: 2 0; 3: 3 0; } 
```
### Second Example
Master UART log:
```
[   208.915283s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link is down
[   208.920617s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] destination is down
[   291.788829s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link RX became up, pinging
[   296.227846s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] remote replied after 23 packets
[   296.350955s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link initialization completed
[   296.358144s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] destination is up
[   296.364164s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] buffer space is 128
[   296.571944s] ERROR(runtime::rtio_mgt::drtio): [LINK#0] error(s) found (0x07):
[   296.577842s] ERROR(runtime::rtio_mgt::drtio): [LINK#0] received packet of an unknown type
[   296.586022s] ERROR(runtime::rtio_mgt::drtio): [LINK#0] received truncated packet
[   296.593421s] ERROR(runtime::rtio_mgt::drtio): [LINK#0] timeout attempting to get remote buffer space
```
where the Satellite again shows no ERROR and just rebooted automatically around t=280s above.

### Further incidents
The Satellite shows no ERRORs and reboots automatically after a while. The master loses connection seemingly out of the blue. The connection is restored once the Satellite rebooted, sometimes with all three DRTIO error flags set (second example above), but most times without any.

### Your System

* Version of the gateware and runtime loaded in the core device: firmware and relevant gateware cores up-to-date with b8ab5f260743eb3ab937e189bfd88bed24aa6162
* Hardware involved: 
    - Kasli v2.0.1
    - Kasli v1.1
    - SFP fibre link (tried both known-working and brand-new hardware components)

pmldrmota, 2021-07-26:
We'll post more information as we continue to investigate this issue, but any thoughts and independent reports on this hardware combination would be very helpful.

pathfinder49, 2021-08-10:
I've seen something similar with a v1.0 master and a v2.0 satellite. Connection was established and dropped at random intervals. In my case this was resolved by using a different pair of SFP fiber adapters.

jbqubit, 2021-08-10:
Which SFP didn't work? Which one did work? See https://github.com/sinara-hw/meta/wiki/SFP

pathfinder49, 2021-08-10:
> 
> 
> Which SFP didn't work? Which one did work? See https://github.com/sinara-hw/meta/wiki/SFP

I don't recall the exact model. They were the same part, though the pair that worked came from a different batch.

sbourdeauducq, 2021-11-10:
We cannot reproduce this and it's likely a hardware issue.

closed 2021-11-10 by sbourdeauducq

HarryMakes, 2021-11-10:
As supplementary info, we tested a DRTIO setup on ARTIQ-6 (d9b01ed81ad217d8c9d578432dbd2ebab45dc342) between a Kasli v1.1 and v2.0, with DAC cable connection. The UART output was monitored.

Kasli v1.1 was the master and the v2.0 was the satellite at first. The FPGA reload was triggered (with `artiq_flash start`) 9 times on the master, and then the master was power-cycled. This was repeated on the satellite. No timeout error was ever observed on the log. Finally, `artiq_sinara_tester` was run without power-cycling or reloading, and no errors were found.

Then, these two Kasli controller switched roles, and the same set of test were performed. Similarly, no related errors were found, i.e. when Kasli v2.0 was master and v1.1 was satellite.

I have not tested the same setup on ARTIQ-7.