m-labs/artiq #1732: openocd build 198533 for nix is attempting to build locally and failing

filipmazurek, 2021-07-28:
# Bug Report

## One-Line Summary

Attempting to open a nix shell which includes `artiq-full.openocd` will attempt to locally build `openocd` and fail.

## Issue Details

### Steps to Reproduce

1. Have a `shell.nix` file with the package `artiq-full.openocd`
2. Update nix channels to use build 14481 of artiq-full (2021-07-27 19:51:54)
3. Attempt to open a nix shell using `nix-shell`.

### Expected Behavior

The nix shell should open and fetch any newly built packages.

### Actual (undesired) Behavior

Nix attempts to build openocd locally and fails.

The issue appears to be with this build of openocd: https://nixbld.m-labs.hk/build/198533

```
these derivations will be built:
  /nix/store/b93nq33hslblxi5x61rqqnv7741yvwgh-python3-3.8.8-env.drv
  /nix/store/g6andxxh9xqkyc3yny2i9msbxx7nvcpf-openocd-0.10.0.drv
  /nix/store/g3qpdvyj13q4naxawn3f4080hbimv6zq-openocd-bscanspi.drv
these paths will be fetched (0.00 MiB download, 9.48 MiB unpacked):

...

building '/nix/store/g6andxxh9xqkyc3yny2i9msbxx7nvcpf-openocd-0.10.0.drv'...
unpacking sources
unpacking source archive /nix/store/jjwhrgbflld1aqzwrap2ffrm9slbcfw0-openocd-0.10.0.tar.bz2
source root is openocd-0.10.0
setting SOURCE_DATE_EPOCH to timestamp 1485117625 of file openocd-0.10.0/tcl/target/tmpa900.cfg
patching sources
applying patch /nix/store/r1linwxgsgsk3ii4vi7xb9w9jiwqxdbz-fix-sheeva.patch
patching file tcl/interface/ftdi/sheevaplug.cfg
applying patch /nix/store/f69qsskda83l24p7l3yqmwja7i9mf8kl-openocd-jtagspi.nix
patching file src/flash/nor/jtagspi.c
Hunk #1 FAILED at 82.
Hunk #2 FAILED at 102.
Hunk #3 FAILED at 118.
Hunk #4 succeeded at 260 with fuzz 2 (offset -44 lines).
Hunk #5 succeeded at 341 (offset -36 lines).
3 out of 5 hunks FAILED -- saving rejects to file src/flash/nor/jtagspi.c.rej
patching file src/flash/nor/spi.h
Hunk #1 FAILED at 87.
1 out of 1 hunk FAILED -- saving rejects to file src/flash/nor/spi.h.rej
builder for '/nix/store/g6andxxh9xqkyc3yny2i9msbxx7nvcpf-openocd-0.10.0.drv' failed with exit code 1
cannot build derivation '/nix/store/g3qpdvyj13q4naxawn3f4080hbimv6zq-openocd-bscanspi.drv': 1 dependencies couldn't be built
error: build of '/nix/store/b93nq33hslblxi5x61rqqnv7741yvwgh-python3-3.8.8-env.drv', '/nix/store/g3qpdvyj13q4naxawn3f4080hbimv6zq-openocd-bscanspi.drv' failed
```

### Your System (omit irrelevant parts)

* Operating System: Ubuntu 20.04.1

filipmazurek, 2021-07-28:
@sbourdeauducq We are running into issues with the most recent build of artiq-full with the openocd package. Any attempt to open a nix shell attempts to build openocd locally and fails.

sbourdeauducq, 2021-07-28:
Update nixpkgs to 21.05
```text
$ nix-channel --remove nixpkgs
$ nix-channel --add https://nixos.org/channels/nixos-21.05 nixpkgs
$ nix-channel --update
```

closed 2021-07-28 by sbourdeauducq