m-labs/artiq #1734: Conda environment libllvmlite.so not found

lriesebos, 2021-07-29:
# Bug Report

## One-Line Summary

`OSError: libllvmlite.so: cannot open shared object file: No such file or directory`

## Issue Details

### Steps to Reproduce

1. `conda create -n artiq_test artiq`
2. `conda activate artiq_test`
3. `artiq_run test.py`

test.py:
```python
from artiq.experiment import *

class E(EnvExperiment):
    def run(self):
        pass
```

### Expected Behavior

File runs without problems.

### Actual (undesired) Behavior

`OSError: libllvmlite.so: cannot open shared object file: No such file or directory`

See output below. Adding the package `llvm-or1k` does not seem to help.

This issue was originally found by our [DAX CI job](https://gitlab.com/duke-artiq/dax/-/jobs/1462987603).

```
leon@ubuntu-2004:~/$ conda create -n artiq_test artiq
Collecting package metadata (current_repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /home/leon/miniconda3/envs/artiq_test

  added / updated specs:
    - artiq

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    cffi-1.14.6                |   py38h400218f_0         224 KB
    glib-2.69.0                |       h5202010_0         1.7 MB
    libgit2-1.1.1              |       hee63804_1         647 KB  conda-forge
    pcre2-10.37                |       h032f7d1_0         1.1 MB  conda-forge
    ------------------------------------------------------------
                                           Total:         3.7 MB

The following NEW packages will be INSTALLED:

  _libgcc_mutex      pkgs/main/linux-64::_libgcc_mutex-0.1-main
  _openmp_mutex      pkgs/main/linux-64::_openmp_mutex-4.5-1_gnu
  artiq              artiq/noarch::artiq-6.7617.19daf912-0
  binutils-or1k-lin~ artiq/linux-64::binutils-or1k-linux-2.30-0
  blas               pkgs/main/linux-64::blas-1.0-mkl
  bzip2              pkgs/main/linux-64::bzip2-1.0.8-h7b6447c_0
  ca-certificates    pkgs/main/linux-64::ca-certificates-2021.7.5-h06a4308_1
  cached-property    pkgs/main/noarch::cached-property-1.5.2-py_0
  certifi            pkgs/main/linux-64::certifi-2021.5.30-py38h06a4308_0
  cffi               pkgs/main/linux-64::cffi-1.14.6-py38h400218f_0
  dbus               pkgs/main/linux-64::dbus-1.13.18-hb2f20db_0
  expat              pkgs/main/linux-64::expat-2.4.1-h2531618_2
  fontconfig         pkgs/main/linux-64::fontconfig-2.13.1-h6c09931_0
  freetype           pkgs/main/linux-64::freetype-2.10.4-h5ab3b9f_0
  glib               pkgs/main/linux-64::glib-2.69.0-h5202010_0
  gst-plugins-base   pkgs/main/linux-64::gst-plugins-base-1.14.0-h8213a91_2
  gstreamer          pkgs/main/linux-64::gstreamer-1.14.0-h28cd5cc_2
  h5py               pkgs/main/linux-64::h5py-3.2.1-py38h6c542dc_0
  hdf5               pkgs/main/linux-64::hdf5-1.10.6-hb1b8bf9_0
  icu                pkgs/main/linux-64::icu-58.2-he6710b0_3
  importlib-metadata pkgs/main/linux-64::importlib-metadata-3.10.0-py38h06a4308_0
  intel-openmp       pkgs/main/linux-64::intel-openmp-2021.3.0-h06a4308_3350
  jpeg               pkgs/main/linux-64::jpeg-9b-h024ee3a_2
  ld_impl_linux-64   pkgs/main/linux-64::ld_impl_linux-64-2.35.1-h7274673_9
  libffi             pkgs/main/linux-64::libffi-3.3-he6710b0_2
  libgcc-ng          pkgs/main/linux-64::libgcc-ng-9.3.0-h5101ec6_17
  libgfortran-ng     pkgs/main/linux-64::libgfortran-ng-7.5.0-ha8ba4b0_17
  libgfortran4       pkgs/main/linux-64::libgfortran4-7.5.0-ha8ba4b0_17
  libgit2            conda-forge/linux-64::libgit2-1.1.1-hee63804_1
  libgomp            pkgs/main/linux-64::libgomp-9.3.0-h5101ec6_17
  libiconv           pkgs/main/linux-64::libiconv-1.15-h63c8f33_5
  libpng             pkgs/main/linux-64::libpng-1.6.37-hbc83047_0
  libssh2            pkgs/main/linux-64::libssh2-1.9.0-h1ba5d50_1
  libstdcxx-ng       pkgs/main/linux-64::libstdcxx-ng-9.3.0-hd4cf53a_17
  libuuid            pkgs/main/linux-64::libuuid-1.0.3-h1bed415_2
  libxcb             pkgs/main/linux-64::libxcb-1.14-h7b6447c_0
  libxml2            pkgs/main/linux-64::libxml2-2.9.12-h03d6c58_0
  llvmlite-artiq     artiq/linux-64::llvmlite-artiq-0.23.0.dev-0
  mkl                pkgs/main/linux-64::mkl-2021.3.0-h06a4308_520
  mkl-service        pkgs/main/linux-64::mkl-service-2.4.0-py38h7f8727e_0
  mkl_fft            pkgs/main/linux-64::mkl_fft-1.3.0-py38h42c9631_2
  mkl_random         pkgs/main/linux-64::mkl_random-1.2.2-py38h51133e4_0
  ncurses            pkgs/main/linux-64::ncurses-6.2-he6710b0_1
  numpy              pkgs/main/linux-64::numpy-1.20.3-py38hf144106_0
  numpy-base         pkgs/main/linux-64::numpy-base-1.20.3-py38h74d4b33_0
  openssl            pkgs/main/linux-64::openssl-1.1.1k-h27cfd23_0
  pcre               pkgs/main/linux-64::pcre-8.45-h295c915_0
  pcre2              conda-forge/linux-64::pcre2-10.37-h032f7d1_0
  pip                pkgs/main/linux-64::pip-21.1.3-py38h06a4308_0
  prettytable        conda-forge/noarch::prettytable-2.1.0-pyhd8ed1ab_0
  pycparser          pkgs/main/noarch::pycparser-2.20-py_2
  pygit2             conda-forge/linux-64::pygit2-1.6.1-py38h497a2fe_0
  pyqt               pkgs/main/linux-64::pyqt-5.9.2-py38h05f1152_4
  pyqtgraph          pkgs/main/noarch::pyqtgraph-0.11.0-py_0
  python             pkgs/main/linux-64::python-3.8.10-h12debd9_8
  python-dateutil    pkgs/main/noarch::python-dateutil-2.8.2-pyhd3eb1b0_0
  python-levenshtein pkgs/main/linux-64::python-levenshtein-0.12.2-py38h27cfd23_0
  python_abi         conda-forge/linux-64::python_abi-3.8-2_cp38
  pythonparser       artiq/noarch::pythonparser-1.3-0
  qasync             artiq/noarch::qasync-0.10.0-0
  qt                 pkgs/main/linux-64::qt-5.9.7-h5867ecd_1
  readline           pkgs/main/linux-64::readline-8.1-h27cfd23_0
  regex              pkgs/main/linux-64::regex-2021.7.6-py38h7f8727e_0
  scipy              pkgs/main/linux-64::scipy-1.6.2-py38had2a1c9_1
  setuptools         pkgs/main/linux-64::setuptools-52.0.0-py38h06a4308_0
  sip                pkgs/main/linux-64::sip-4.19.13-py38he6710b0_0
  sipyco             artiq/noarch::sipyco-1.2-0
  six                pkgs/main/noarch::six-1.16.0-pyhd3eb1b0_0
  sqlite             pkgs/main/linux-64::sqlite-3.36.0-hc218d9a_0
  tk                 pkgs/main/linux-64::tk-8.6.10-hbc83047_0
  wcwidth            pkgs/main/noarch::wcwidth-0.2.5-py_0
  wheel              pkgs/main/noarch::wheel-0.36.2-pyhd3eb1b0_0
  xz                 pkgs/main/linux-64::xz-5.2.5-h7b6447c_0
  zipp               pkgs/main/noarch::zipp-3.5.0-pyhd3eb1b0_0
  zlib               pkgs/main/linux-64::zlib-1.2.11-h7b6447c_3

leon@ubuntu-2004:~/$ conda activate artiq_test
(artiq_test) leon@ubuntu-2004:~/$ artiq_run test.py 
Traceback (most recent call last):
  File "/home/leon/miniconda3/envs/artiq_test/lib/python3.8/site-packages/llvmlite_artiq/binding/ffi.py", line 119, in <module>
    lib = ctypes.CDLL(os.path.join(_lib_dir, _lib_name))
  File "/home/leon/miniconda3/envs/artiq_test/lib/python3.8/ctypes/__init__.py", line 381, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /home/leon/miniconda3/envs/artiq_test/lib/python3.8/site-packages/llvmlite_artiq/binding/libllvmlite.so)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/leon/miniconda3/envs/artiq_test/bin/artiq_run", line 6, in <module>
    from artiq.frontend.artiq_run import main
  File "/home/leon/miniconda3/envs/artiq_test/lib/python3.8/site-packages/artiq/frontend/artiq_run.py", line 13, in <module>
    from llvmlite_artiq import binding as llvm
  File "/home/leon/miniconda3/envs/artiq_test/lib/python3.8/site-packages/llvmlite_artiq/binding/__init__.py", line 6, in <module>
    from .dylib import *
  File "/home/leon/miniconda3/envs/artiq_test/lib/python3.8/site-packages/llvmlite_artiq/binding/dylib.py", line 4, in <module>
    from . import ffi
  File "/home/leon/miniconda3/envs/artiq_test/lib/python3.8/site-packages/llvmlite_artiq/binding/ffi.py", line 124, in <module>
    lib = ctypes.CDLL(_lib_name)
  File "/home/leon/miniconda3/envs/artiq_test/lib/python3.8/ctypes/__init__.py", line 381, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libllvmlite.so: cannot open shared object file: No such file or directory
```

### Your System (omit irrelevant parts)

* Operating System: Ubuntu 20.04 with Conda
* ARTIQ version: Latest ARTIQ 6

sbourdeauducq, 2021-07-29:
The error is actually this:

> OSError: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /home/leon/miniconda3/envs/artiq_test/lib/python3.8/site-packages/llvmlite_artiq/binding/libllvmlite.so)

Conda and conda packages have implicit dependencies on various packages from your distro, which are out of control and can break, as is happening now.

I guess you can upgrade the glibc on your system (2.32 is already one year old...), or downgrade the glibc used for building conda packages in nix-scripts.

There are apparently also some hacks such as this (this is a common conda problem and affecting other projects as well):
https://gist.github.com/michaelchughes/85287f1c6f6440c060c3d86b4e7d764b

lriesebos, 2021-07-30:
You are right, I did look over the glibc version error. For CI, we moved from an Ubuntu 20.04 LTS image (ubuntu:stable docker image) to an Ubuntu 21.04 image (ubuntu:rolling) and [the CI job passes now](https://gitlab.com/duke-artiq/dax/-/jobs/1464096401).

closed 2021-07-30 by lriesebos