m-labs/artiq #1735: clock input specification improvements

sbourdeauducq, 2021-07-30:
Currently the the external clock selection on ARTIQ boards (KC705, ZC706, Kasli, Kasli-SoC) is inconsistent, limited in features, and difficult to use with firmware recompilations and reflashing required to change Si5324 settings on Kasli.

There are currently two clock settings, the ``rtio_clock`` config key used by KC705, ZC706 and Kasli, and the JSON options used by Kasli.

I propose replacing them in ARTIQ-7 with a single ``rtio_clock`` config key that can take values such as:
* ``ext0_synth0_10to125``: 10MHz clock input to 125MHz RTIO clock, using a PLL (e.g. Si5324).
* ``ext0_synth0_100to125``: same as above with 100MHz input.
* ``ext0_synth0_125to125``: 125MHz clock input to 125MHz RTIO clock, through a PLL, which can filter jitter. The properties of the PLL would be board-dependent.
* ``ext0_bypass_125``: 125MHz clock input to 125MHz RTIO clock, any PLL bypassed
* ``int_125``: 125MHz RTIO clock, internally generated

If we ever have a board with several clock inputs, the input could be selected with ``ext0``/``ext1``/``ext2``...
Synthesizers (e.g. WRPLL vs. Si5324) could similarly be selected using ``synth0``/``synth1``...

There can be board-dependent extensions for these strings. For example, PLL loop bandwidth could be selected using ``ext0_synth125to125_bw606`` / ``ext0_synth125to125_bw200``.

Any given board would only support its own set of strings.

dhslichter, 2021-08-02:
Would it be better to have a series of key, value pairs indicating: clock input, synthesizer (incl. bypass), input frequency, output frequency, loop bandwidth?  I just worry about the strings being fragile and unwieldy.

sbourdeauducq, 2021-08-04:
>  Or is the idea that only a small subset of combinations will actually be supported, so making a small list of possible strings allows users to choose from among the only available combinations?

Yes, that's the idea. While the strings share a common format for easier understanding, each board would only support a limited set of clocking configurations and corresponding strings. The strings would make it easier to see what is supported and what is not.