m-labs/artiq #1738: Adding delay before reading in Sampler

han94ros, 2021-08-05:
# ARTIQ Feature Request

## Problem this request addresses

A clear and concise description of what the problem is.

## Describe the solution you'd like

A clear and concise description of what you want to happen.

## Additional context

Add any other context about the feature request here.

sbourdeauducq, 2021-08-05:
I don't think a gateware rebuild is required for that. Why do you think it is?

sbourdeauducq, 2021-08-05:
And it seems you basically have to split ``sample_mu()`` into 1) the part that does ``self.bus_adc.write`` and 2) the part that does ``self.bus_adc.read``. Can you prepare and carefully test a PR?

han94ros, 2021-08-05:
I 

> I don't think a gateware rebuild is required for that. Why do you think it is?

I thought a rebuild was required because I changed values in the function for example self.cnv.pulse(30*ns) to self.cnv.pulse(300*ns) and when running sampler the underflow error still referenced the original line of  self.cnv.pulse(30*ns).

I will split the sample_mu function into two sections and prepare a PR.

sbourdeauducq, 2021-08-05:
> self.cnv.pulse(30_ns) to self.cnv.pulse(300_ns)

The sampler coredevice driver is made of regular ARTIQ-Python kernels.

han94ros, 2021-08-05:
> > self.cnv.pulse(30_ns) to self.cnv.pulse(300_ns)
> 
> The sampler coredevice driver is made of regular ARTIQ-Python kernels.

Ok I will make the changes in sampler.py and prepare a PR.
Do you believe this device driver change is necessary to add the sampler to my sequence of stages which I have posted in https://forum.m-labs.hk/d/227-help-request-with-scheduling-stages/39, as a recap I am making use of DMA to save the output signals but inclusion of the sampler is obviously causing RTIO underflow issues.

han94ros, 2021-08-05:
Ok I will make the changes in sampler.py and prepare a PR.
Do you believe this sampler coredevice driver change is necessary to add the sampler to my sequence of stages which I have posted in https://forum.m-labs.hk/d/227-help-request-with-scheduling-stages/39, as a recap I am making use of DMA to save the output signals but inclusion of the sampler is obviously causing RTIO underflow issues. Am I overcomplicating a simple issue?

han94ros, 2021-08-05:
> > self.cnv.pulse(30_ns) to self.cnv.pulse(300_ns)
> 
> The sampler coredevice driver is made of regular ARTIQ-Python kernels.

Seb .... I am now able to run several stages 3 out of 7 with this modification, all I need to do is increase the input FIFO of the ADC SPI RTIO channel to avoid the RTIO input overflow errors. Thanks for pointing me in the right direction.

han94ros, 2021-08-05:
> Seb .... I am now able to run several stages 3 out of 7 with this modification, all I need to do is increase the input FIFO of the ADC SPI RTIO channel to avoid the RTIO input overflow errors. [...]

han94ros, 2021-08-05:
I am able to run until a stage which is 300us in its duration however after that I am getting the following error `RTIOOverflow: RTIO input overflow on channel 20` . Channel 20 corresponds to the ADC SPI channel and I have tried to increase the input FIFO depth by entering `self.rtio_channels[20].ififo_depth = 1024 ` in line 642 in gateware/targets/kasli.py. This has not solved the issue, what would you recommend.

han94ros, 2021-08-06:
I have tested the functions and prepared a PR, please see #1742.

sbourdeauducq, 2021-09-12:
@han94ros Is this issue still relevant and are you still working on your PR, or can we abandon them?

han94ros, 2021-09-13:
> @han94ros Is this issue still relevant and are you still working on your PR, or can we abandon them?

I will finish the PR this week. Can you please see the DM I have sent you on the M-labs Chat forum about implementing a migen PID control loop into ARTIQ software.

closed 2021-12-06 by sbourdeauducq

hanros1994, 2021-12-06:
Hi Seb

Is there any reason why this was closed?

sbourdeauducq, 2021-12-06:
No progress and you said you had resolved your initial issue.