m-labs/artiq #1748: SUServo simulation fails because async is a reserved keyword since python 3.7

marmeladapk, 2021-09-01:
# Bug Report

## One-Line Summary

SUServo simulation (artiq/gateware/tests/suservo/test_servo.py) fails because async is a reserved keyword since python 3.7

### Steps to Reproduce

Run test (artiq/gateware/tests/suservo/test_servo.py).

If you change async to anything else (e.g. async2) at these places the issue goes away:

- https://github.com/m-labs/artiq/blob/master/artiq/gateware/test/suservo/test_servo.py#L94
- https://github.com/m-labs/artiq/blob/master/artiq/gateware/test/suservo/test_adc.py#L46
- https://github.com/m-labs/artiq/blob/master/artiq/gateware/test/suservo/test_adc.py#L65
- https://github.com/m-labs/artiq/blob/master/artiq/gateware/test/suservo/test_adc.py#L98

### Expected Behavior

Test runs.

### Actual (undesired) Behavior

```
File "(...)/test_adc.py", line 46
self.sync.async += [
              ^
SyntaxError: invalid syntax
```

### Your System (omit irrelevant parts)

* ARTIQ version: commit 11790c6

dnadlinger, 2021-09-01:
Do you want to propose a change (submit a PR) renaming the clock domain? Should just be a case of find/replace.

marmeladapk, 2021-09-02:
Sure, I didn't do that because I wasn't sure what the proper name would be. For my use case I just renamed it to `async2` but it won't fly in official repo. ;)

marmeladapk, 2021-09-03:
I just learned that single trailing underscore is used by convention to avoid conflicts with Python keyword. Would that be okay?

jordens, 2021-09-03:
Sure. We're doing that elsewhere as well (`pow_` in `suservo`).

sbourdeauducq, 2022-01-10:
@pmldrmota The tests fail with your proposed change (I don't know when the problem was introduced...)
```
======================================================================
FAIL: test_run (test_dds.DDSTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/sb/artiq/artiq/gateware/test/suservo/test_dds.py", line 90, in test_run
    main()
  File "/home/sb/artiq/artiq/gateware/test/suservo/test_dds.py", line 84, in main
    assert data[-1][1] == [[0xe, 0x40 | i, 0x30 | i, 0x20 | i] for i in
AssertionError

----------------------------------------------------------------------
Ran 4 tests in 1.652s

```

pmldrmota, 2022-01-10:
Probably the commit just now (changing the command from `0xe` to `0x0e + DEFAULT_PROFILE`)?

occheung, 2022-01-11:
9eee0e5 updated the profile check in test (to 0x15). This fail should be gone now.

commit 4eee49f8 (2022-01-11): gateware.test.suservo: Fix tests for python >=3.7

Closes #1748
commit 4f875315 (2022-01-11): gateware.test.suservo: Fix tests for python >=3.7

Closes #1748
commit ee0eb238 (2022-01-11): gateware.test.suservo: Fix tests for python >=3.7

Closes #1748
closed 2022-01-11 by sbourdeauducq