m-labs/artiq #1751: llvmlite version number to allow for uploading wheel to PyPI

lochsh, 2021-09-06:
# ARTIQ Feature Request
Please add a version number (e.g. via a git tag) to the llvmlite-artiq repo that artiq is dependent on.

## Problem this request addresses
ARTIQ's dependency, llvmlite-artiq, requires having LLVM 6.0 installed, and the environment variable LLVM_CONFIG pointing to the appropriate file. For projects using tools like poetry to manage their Python dependencies, this creates an inconvenient extra step in the installation process.

## Describe the solution you'd like
If llvmlite-artiq had wheels available on PyPI, then this would avoid the need for the manual installation. I am happy to build and upload wheels for cp38 linux x86_64 to PyPI. However, the https://github.com/m-labs/llvmlite/tree/artiq-6.0 latest commit on the repo (which I assume is what's needed for ARTIQ) does not have a version tag.

I don't want to upload a wheel to PyPI with a made-up version number. If M-Labs could pick a suitable version number and tag the appropriate commit, then I'm happy to build and upload a wheel for my target architecture.

I'm not sure what version number to suggest -- it looks like numba stopped using git tags for their llvmlite versions at 0.6, but the M-Labs fork was made after version 0.23. So currently, if you install the llvmlite_artiq package (whether from conda forge or as a git dependency) it has version 0.6.0+600.g1d167be. Note that PyPI doesn't accept "local" version numbers like these.

## Additional context
Issues aren't enabled on the M-Labs llvmlite repo, hence making this issue here. Thanks.

sbourdeauducq, 2021-09-12:
>  the https://github.com/m-labs/llvmlite/tree/artiq-6.0 latest commit on the repo (which I assume is what's needed for ARTIQ) does not have a version tag.

The version used by ARTIQ is defined here:
https://git.m-labs.hk/M-Labs/nix-scripts/src/branch/master/artiq-fast/pkgs/llvmlite-artiq.nix
The conda packages for Linux and Windows are also built from this single reference.

Are you happy with a 0.23.0.dev tag on that commit? Note that ARTIQ-7 does not use llvmlite-artiq anymore.

lochsh, 2021-11-01:
Apologies for not responding to this -- we are now looking to upgrade and use the RISC-V Kasli softcore, removing the dependency on the `llvmlite-artiq` fork. Thanks again.

pca006132, 2021-11-01:
> Apologies for not responding to this -- we are now looking to upgrade and use the RISC-V Kasli softcore, removing the dependency on the `llvmlite-artiq` fork. Thanks again.

Note that upgrading to a newer version of LLVM may trigger some bugs due to incorrect codegen: #1599