m-labs/artiq #1752: artiq_flash error, artiq_81ll_wqc_top.bit contents differ

jbqubit, 2021-09-09:
I get an error when flashing kasli using nix openocd from nix-scripts master -- see below for error. I see the same error across several Kasli v1.1 known to work.Details are in the Forum [here](https://forum.m-labs.hk/d/282-artiq-flash-error-artiq-81ll-wqc-topbit-contents-differ). 

On branch [988d411946](https://git.m-labs.hk/M-Labs/nix-scripts/commit/988d41194619c0b95477927b2a819004742ef56f) the open-ocd version is `Open On-Chip Debugger 0.10.0-snapshot (2021-07-26-23:34)`. And `artiq_flash` succeeds. 

Probably nix-scripts needs to revert to using the m-labs 0.10.0 version of openocd.  

----------
```
[nix-shell:~/artiq-test]$ artiq_flash --srcbuild -d artiq_kasli -t kasli -V brittonlab-phaserfun-master
Design: top;UserID=FFFFFFFF;COMPRESS=TRUE;Version=2020.1
Part name: 7a100tfgg484
Date: 2021/09/07
Time: 21:40:37
Bitstream payload length: 0x2c9aec
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 25000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
TEMP 39.34 C
VCCINT 0.992 V
VCCAUX 1.780 V
VCCBRAM 0.994 V
VPVN 0.000 V
VREFP 0.000 V
VREFN 0.000 V
VCCPINT 0.000 V
VCCPAUX 0.000 V
VCCODDR 0.000 V
Info : Found flash device 'sp s25fl256s' (ID 0x00190201)
Warn : device needs paging or 4-byte addresses - not implemented
Info : Found flash device 'sp s25fl256s' (ID 0x00190201)
Warn : device needs paging or 4-byte addresses - not implemented
Info : sector 0 took 1 ms
Info : sector 1 took 1 ms
Info : sector 2 took 0 ms
...
Info : sector 43 took 0 ms
Info : sector 44 took 1 ms
Info : Found flash device 'sp s25fl256s' (ID 0x00190201)
Warn : device needs paging or 4-byte addresses - not implemented
Info : Found flash device 'sp s25fl256s' (ID 0x00190201)
Warn : device needs paging or 4-byte addresses - not implemented
read 2923244 bytes from file /run/user/1000/artiq_81ll_wqc_top.bit and flash bank 1 at offset 0x00000000 in 1.758357s (1623.521 KiB/s)
contents differ
diff 0 address 0x0000007c. Was 0x20 instead of 0x3c
diff 1 address 0x0000007e. Was 0x48 instead of 0x5a
diff 2 address 0x0000007f. Was 0x10 instead of 0x1d
diff 3 address 0x00000102. Was 0x40 instead of 0x41
diff 4 address 0x00000103. Was 0x04 instead of 0x94
...
diff 123 address 0x000008d3. Was 0x00 instead of 0x04
diff 124 address 0x000008e4. Was 0x00 instead of 0x30
diff 125 address 0x000008e6. Was 0x00 instead of 0x20
diff 126 address 0x000008e7. Was 0x00 instead of 0x01
diff 127 address 0x000008eb. Was 0x00 instead of 0x90
More than 128 errors, the rest are not printed.
Traceback (most recent call last):
File "/nix/store/apmi6cjxwcwir2d52djs6yavmp44bzy6-python3.8-artiq-6.7621.d9b01ed8/bin/.artiq_flash-wrapped", line 9, in <module>
sys.exit(main())
File "/nix/store/5vg1apvd0zzbx7lv5mfh015n5zcml5if-python3-3.8.9-env/lib/python3.8/site-packages/artiq/frontend/artiq_flash.py", line 474, in main
programmer.run()
File "/nix/store/5vg1apvd0zzbx7lv5mfh015n5zcml5if-python3-3.8.9-env/lib/python3.8/site-packages/artiq/frontend/artiq_flash.py", line 207, in run
self._client.run_command(cmdline)
File "/nix/store/5vg1apvd0zzbx7lv5mfh015n5zcml5if-python3-3.8.9-env/lib/python3.8/site-packages/artiq/remoting.py", line 57, in run_command
subprocess.check_call([arg.format(tmp=self._tmp, **kws) for arg in cmd])
File "/nix/store/q6gfck5czr67090pwm53xrdyhpg6bx67-python3-3.8.9/lib/python3.8/subprocess.py", line 364, in check_call
raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['openocd', '-s', '/nix/store/24793jlzrfswi1cnjanwpb4xyh7n0zk4-openocd-bscanspi/share/openocd/scripts', '-c', 'source [find board/kasli.cfg]; target create xc7.spi0.proxy testee -chain-position xc7.tap; flash bank spi0 jtagspi 0 0 0 0 xc7.spi0.proxy 0x2; gdb_port disabled; tcl_port disabled; telnet_port disabled; init; xadc_report xc7.tap; pld load 0 {/nix/store/24793jlzrfswi1cnjanwpb4xyh7n0zk4-openocd-bscanspi/share/bscan-spi-bitstreams/bscan_spi_xc7a100t.bit}; flash probe spi0; flash erase_sector spi0 0 44; flash write_bank spi0 {/run/user/1000/artiq_81ll_wqc_top.bit} 0x0; flash verify_bank spi0 {/run/user/1000/artiq_81ll_wqc_top.bit} 0x0; flash probe spi0; flash erase_sector spi0 64 66; flash write_bank spi0 {artiq_kasli/brittonlab-phaserfun-master/software/bootloader/bootloader.bin} 0x400000; flash verify_bank spi0 {artiq_kasli/brittonlab-phaserfun-master/software/bootloader/bootloader.bin} 0x400000; flash probe spi0; flash erase_sector spi0 69 79; flash write_bank spi0 {artiq_kasli/brittonlab-phaserfun-master/software/runtime/runtime.fbi} 0x450000; flash verify_bank spi0 {artiq_kasli/brittonlab-phaserfun-master/software/runtime/runtime.fbi} 0x450000; xc7_program xc7.tap; exit']' returned non-zero exit status 1.
```

sbourdeauducq, 2021-09-09:
> Probably nix-scripts needs to revert to using the m-labs 0.10.0 version of openocd.

We've been using the patched 0.11 to flash a substantial number of Kasli of various versions, so it's not that simple...

jbqubit, 2021-09-09:
I'm using Kasli v1.1. Which according to the [schematic](https://github.com/sinara-hw/Kasli/releases/tag/v1.1.1) has S25FL128S flash. It looks like the openocd probe reports a different flash chipset: [Cypress s25fl256s flash](https://www.cypress.com/documentation/datasheets/s25fl128ss25fl256s-128-mb-16-mb256-mb-32-mb-30v-spi-flash-memory). 

The current openocd [patch](https://git.m-labs.hk/M-Labs/nix-scripts/src/branch/master/artiq-fast/pkgs/openocd-jtagspi.diff) doesn't reference either version of the Cypress flash.

jbqubit, 2021-09-09:
Confirmed that the Kasli v1.1 PCB in my lab is stuffed with s25fl256s. 
[image]

sbourdeauducq, 2021-09-20:
Found a board here with the same flash that exhibits the issue. https://git.m-labs.hk/M-Labs/nix-scripts/commit/575ef05cd554c239e4cc8cb97ae4611db458a80d

closed 2021-09-20 by sbourdeauducq