m-labs/artiq #1754: Startup kernel aborted error

ljstephenson, 2021-09-21:
# Bug Report

## One-Line Summary

Startup kernel aborts due to "not a shared library for current architecture"

## Issue Details

### Steps to Reproduce

1. startup.py:
```
from artiq.experiment import *

class Startup(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        pass
```
2. Compile and write to kasli with artiq_compile/artiq_coremgmt
3. Restart kasli

### Expected Behavior

Startup kernel runs (and in this case does nothing)

### Actual (undesired) Behavior

Following error message in UART log
```
[     6.676657s]  INFO(runtime::session): running startup kernel
[     6.711239s] ERROR(runtime::session): startup kernel aborted: cannot load kernel: parse error: not a shared library for current architecture
```

### Your System (omit irrelevant parts)

This isn't the cleanest setup at the moment - we're using gateware built from our own fork, last upstream commit cca654bd475bf7401bfce76afdcb6bd62c1cad24 (with the rest of artiq also at this state). I tried using artiq_compile in a fresh environment from the stable channel, and this also failed.

* ARTIQ version: tried with ARTIQ v6.7621.d9b01ed8/ARTIQ v7.unknown.beta
* Hardware involved: kasli v1.1

Is this linked to the move away from or1k and I'm in some sort of broken in-between state, having built gateware from an intermediate commit?

Conda list for custom environment: [artiq-ndscan.txt](https://github.com/m-labs/artiq/files/7206652/artiq-ndscan.txt)
Conda list for artiq 6 environment: [artiq-6.txt](https://github.com/m-labs/artiq/files/7206655/artiq-6.txt)

sbourdeauducq, 2021-09-22:
You need to compile the startup kernel with the same ARTIQ major version used to compile the gateware. ARTIQ-7 (since recently) uses RISC-V whereas ARTIQ-6 still uses OpenRISC (and will not move to RISC-V).

ljstephenson, 2021-09-22:
I get the same error using artiq_compile from the same commit referenced above, but I'll pull all in the upstream changes and try again.

closed 2021-11-08 by sbourdeauducq