m-labs/artiq #1755: Developing ARTIQ assertion Error

artiqsam, 2021-09-24:
The development is  on machine with Ubuntu 20.04 LTS  is able to  complete with unmodified nix-scripts, with the ARTIQ channel configured  with $ nix-shell "<artiq-full/fast/shell-dev.nix>" and then running     $ python -m artiq.gateware.targets.kasli. However, when trying to Run with $ nix-shell -I artiqSrc=path_to_artiq_sources shell-dev.nix getting the following error.

**_@artiq:~/artiq/nix-scripts/artiq-fast$ nix-shell -I artiqSrc=~/artiq/artiq/ shell-dev.nix 
trace: For ARTIQ 7+, use 'nix develop' on the flake instead.
error: assertion (((pkgs).lib.asserts.assertMsg  (! (((pkgs).lib.strings.versionAtLeast  (artiqpkgs).artiq.version)  "7.0")))  "For ARTIQ 7+, use 'nix develop' on the flake instead.") failed at /home/artiq/artiq/nix-scripts/artiq-fast/shell-dev.nix:7:3_**

Please suggest a workaround for the assertion error

## Category: FILL_IN

## Description

Question text

sbourdeauducq, 2021-09-24:
Do what the message says.

closed 2021-09-24 by sbourdeauducq

artiqsam, 2021-09-24:
I am new at the nix environment. It would be helpful if you could give a sample command procedure for nix develop on flake to get the ARTIQ assertion cleared. Thank you for your help.

sbourdeauducq, 2021-09-24:
nix-env -iA nixUnstable
git clone artiq
cd artiq
nix develop

You don't need nix-scripts anymore.

jbqubit, 2021-09-24:
@sbourdeauducq Please update the developing docs now that ARTIQ is using flakes. https://m-labs.hk/artiq/manual/developing.html

sbourdeauducq, 2021-09-25:
I have not decided whether to keep using flakes, it's just a beta-test. Note that the legacy and stable branches do not have flakes and the development shell can be entered as before.