m-labs/artiq #1757: Phaser init underflows on RISC-V

airwoodix, 2021-09-27:
# Bug Report

## One-Line Summary

Phaser `init()` underflows on RISC-V.

## Issue Details

The JSON file below describes a simple system with some TTLs and a Phaser module on a Kasli v2.0 coredevice. Similar observations are valid for Kasli 1.1. When built against the latest master, Phaser `init()` underflows (in the `artiq_sinara_tester` and in very simple experiments doing a core reset, a seconds-long delay, and a phaser init).

- Such crate configurations are known to work on OpenRISC-1000 (e.g. against e7a46ec767219d80e2a0ae73d462bc8af553e9df)
- Configurations with e.g. Urukul instead of Phaser work fine as well
- The following searches do not hit:

```bash
grep -Pe "Timing constraints are not met\." artiq_kasli/variant/gateware/vivado.log
grep -Pe "\d+ constraint not met\." artiq_kasli/variant/gateware/vivado.log
```

@jordens Can you reproduce this? Could it be a case of "Vivado non-reproducibility" addressed e.g. at [sinara-systems/52f15e1](https://git.m-labs.hk/M-Labs/sinara-systems/commit/52f15e1eb9270f8f6dcc0bf90c9df6d1b9d45923)? (I tried to change the Phaser EEM port but it doesn't help).

@sbourdeauducq Can this be related to the RISC-V cores? or is this pure coincidence?

### Steps to Reproduce

1. `nix develop github:m-labs/artiq/4bfd010f0306819ec1624f1af6f14e59bfc26196`
2. `python -m artiq.gateware.targets.kasli_generic variant.json`

With `variant.json`:

```json
{
    "target": "kasli",
    "variant": "variant",
    "hw_rev": "v2.0",
    "base": "standalone",
    "peripherals": [
        {
            "type": "dio",
            "board": "DIO_SMA",
            "hw_rev": "v1.1",
            "ports": [0],
            "edge_counter": true,
            "bank_direction_low": "input",
            "bank_direction_high": "output"
        },
        {
            "type": "phaser",
            "hw_rev": "v1.1",
            "ports": [1]
        }
    ]
}

```
3. `artiq_ddb_template variant.json > device_db.py`
4. `artiq_flash -V variant -d artiq_kasli --srcbuild`
5. `artiq_sinara_tester -o phasers`

### Expected Behavior

`artiq_sinara_tester` executes properly.

### Actual (undesired) Behavior

`artiq_sinara_tester` fails with an RTIO underflow on the phaser channel:

```
artiq.coredevice.exceptions.RTIOUnderflow: RTIO underflow at 26425349352 mu, channel 12, slack -10392 mu
```

### Your System (omit irrelevant parts)

* ARTIQ version: 4bfd010f0306819ec1624f1af6f14e59bfc26196
* Vivado: v2020.1 (64-bit)

sbourdeauducq, 2021-09-29:
Try increasing some of the delays?

jordens, 2021-09-29:
@airwoodix Gateware issues like that Vivado non-reproducibility show up as spurious failures, not as a reproducible RTIOUnderflow. That regression feels like compiler/cpu architecture/cpu implementation area. I'd propose two things: (1) work around it for now by adding a bit more slack in phaser init or in the tester script (happy to accept PRs) and (2) identify/reduce what code pattern exhibits that regression. vexrisc and mor1kx are quite different, not least in their endianess.

commit 3c1cbf47 (2021-10-10): phaser: add more slack during init. Closes #1757
closed 2021-10-10 by sbourdeauducq