m-labs/artiq #1759: Frequent connection loss with Kasli

steine01, 2021-10-01:
# Bug Report

## One-Line Summary

Within days to weeks after startup, the connection to ARTIQ is lost with the Error `WinError 10060` and cannot be restored, other than by power cycling Kasli.

## Issue Details

### Steps to Reproduce

1. I have not found a way to reliably reproduce the initial occurence of the error yet.
2. Submit the experiment 
```
class ResetCore(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        
    @kernel
    def run(self):
        self.core.reset()
```
3. Ping the device IP.

### Expected Behavior

The experiment returns quickly and successfully. The device responds to the ping.

### Actual (undesired) Behavior

* Sometimes the device responds to the ping within the expected delay, sometimes `The destination host is unreachable`.
* The submitted experiment hangs in the dashboard as `running` for about a minute. Termination requests do not work, only deleting the experiment is possible. After about a minute the error below is raised.
* Restarting the master, control manager and dashboard does not resolve the issue.
* The error itself would not be such a big problem, if it could be resolved without power cycling the system. This causes issues in the experiment such as losing trapped ions due to the missing RF signal.

```
root:Terminating with exception (TimeoutError: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond)
Traceback (most recent call last):
  File "C:\Users\ybion443\.conda\envs\artiq-6-2\lib\site-packages\artiq\master\worker_impl.py", line 300, in main
    exp_inst.run()
  File "C:\Users\ybion443\.conda\envs\artiq-6-2\lib\site-packages\artiq\language\core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Users\ybion443\.conda\envs\artiq-6-2\lib\site-packages\artiq\coredevice\core.py", line 132, in run
    self.comm.check_system_info()
  File "C:\Users\ybion443\.conda\envs\artiq-6-2\lib\site-packages\artiq\coredevice\comm_kernel.py", line 342, in check_system_info
    self._write_empty(Request.SystemInfo)
  File "C:\Users\ybion443\.conda\envs\artiq-6-2\lib\site-packages\artiq\coredevice\comm_kernel.py", line 310, in _write_empty
    self._write_header(ty)
  File "C:\Users\ybion443\.conda\envs\artiq-6-2\lib\site-packages\artiq\coredevice\comm_kernel.py", line 302, in _write_header
    self.open()
  File "C:\Users\ybion443\.conda\envs\artiq-6-2\lib\site-packages\artiq\coredevice\comm_kernel.py", line 188, in open
    self.socket = initialize_connection(self.host, self.port)
  File "C:\Users\ybion443\.conda\envs\artiq-6-2\lib\site-packages\artiq\coredevice\comm.py", line 25, in initialize_connection
    sock = socket.create_connection((host, port))
  File "C:\Users\ybion443\.conda\envs\artiq-6-2\lib\socket.py", line 728, in create_connection
    raise err
  File "C:\Users\ybion443\.conda\envs\artiq-6-2\lib\socket.py", line 716, in create_connection
    sock.connect(sa)
TimeoutError: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond
```

### Your System (omit irrelevant parts)

* Operating System: Windows 10
* ARTIQ version: ARTIQ v6.7621.d9b01ed8
* Version of the gateware and runtime loaded in the core device: I would need to power cycle the device to gain that information, since a similiar error to the one above is produced in the current state. I updated the firmware a couple of days ago though. This problem has persisted over multiple releases.

[conda_list.txt](https://github.com/m-labs/artiq/files/7265990/conda_list.txt)

sbourdeauducq, 2021-10-01:
Can you record the serial output of Kasli and see if any error message or panic/crash is reported?

steine01, 2021-10-01:
I used Wireshark + USBPcap to monitor the connection to the host from restarting the master to seeing the error message and I got this on the serial USB connection (Quad RS232-HS): 

No.	Time	Source	Destination	Protocol	Length	Info
1	0.000000	host	1.6.0	USB	36	GET DESCRIPTOR Request DEVICE
2	0.000000	1.6.0	host	USB	46	GET DESCRIPTOR Response DEVICE
3	0.000000	host	1.6.0	USB	36	GET DESCRIPTOR Request CONFIGURATION
4	0.000000	1.6.0	host	USB	129	GET DESCRIPTOR Response CONFIGURATION
5	0.000000	host	1.6.0	USB	36	SET CONFIGURATION Request
6	0.000000	1.6.0	host	USB	28	SET CONFIGURATION Response

I let it run for about an hour afterwards and did not receive any additional lines.

sbourdeauducq, 2021-10-01:
Don't use wireshark, just look at the serial port normally.

Petroglyph94, 2021-10-01:
What do you mean by that? I am not very familiar with the serial connection of ARTIQ. Is there a dedicated built-in terminal? When I search the documentation for "serial", I find `flterm`, which I cannot use I think (because I am on Windows). I initially tried to use hterm, but there are no serial connections listed there. Also, there is the `SPIMaster` device, do you mean that? I have not used it yet, do I just need to use its `read`method to get the logs?

steine01, 2021-10-01:
What do you mean by that? I am not very familiar with the serial connection of ARTIQ. Is there a dedicated built-in terminal? When I search the documentation for "serial", I find `flterm`, which I cannot use I think (because I am on Windows). I initially tried to use hterm, but there are no serial connections listed there. Also, there is the `SPIMaster` device, do you mean that? I have not used it yet, do I just need to use its `read`method to get the logs?

Also I do not get any output when I try to get the firmware version with artiq_coremgmt as described in the form for bugs. The four dots are supposed to be the IP address of the device, right?

hartytp, 2021-10-05:
@steine01 

> I am not very familiar with the serial connection of ARTIQ. Is there a dedicated built-in terminal? When I search the documentation for "serial", I find flterm, which I cannot use I think (because I am on Windows). I initially tried to use hterm, but there are no serial connections listed there.

Any serial terminal will do (e.g. [`python -m serial.tools.miniterm`](https://pyserial.readthedocs.io/en/latest/tools.html#module-serial.tools.miniterm) ) -- just make sure you're connecting to the right port and using the correct baud rate.

If you want more help with that, the issue tracker is probably not the best place (M-Labs forum?).

> Also, there is the SPIMaster device, do you mean that? I have not used it yet, do I just need to use its readmethod to get the logs?

No, that's different.

steine01, 2021-10-06:
@hartytp 
Thank you for the help, I will open a question in the forum about it, because something does not seem to be correct on my end.

steine01, 2022-02-23:
I have since changed the network architecture between the host PC and Kasli, because other groups reported improvements for similiar issues after doing that. Instead of connecting the host and Kasli to a common switch, I now use an USB<-> Ethernet adapter to connect the host directly to the Kasli.

But now that I ran multi-hour long experiments again, I still see the same error (TimeoutError) and ARTIQ does not respond to requests from the dashboard anymore until it is power-cycled. Also I still do not know what kind of serial connection I should use to get a more complete error message, there were barely any response to my question on the forum [https://forum.m-labs.hk/d/303-serial-connection-between-host-and-kasli](url).

What can I do to solve this problem? From communication with our IT department last year, it seemed as if the Kasli just stops responding if a package is sent to the host and it takes a long amount of time for the host to respond. Specifically, there seemed to have been a 2 second delay between a request sent by the Kasli and the response of the host in one instance.

An assumption was, that a congested network might be the cause of the long delay, so I exchanged the connection over a switch to the direct connection between Kasli and host. This did not solve the issue though, because I still get the error. Also, that Kasli does not just time out and cancel the experiment, if there is a significant delay in communication, seems like an issue to me.

jordens, 2022-02-23:
Hmm. For this now wireshark trace would be very helpful. That would show why and when which connection breaks.

steine01, 2022-02-23:
In the current state, where I have not yet power-cycled after the error, I only see broadcasts from the host (192.168.1.2) to the Kasli (192.168.1.3), but no response. At time 0, I submitted reset core and I submitted it again at time 132. 192.168.1.1 is the gateway IP.

Should I power-cycle and run wireshark while my long experiment is running? Or are there other diagnostics I can perform in this state?

[image]

jordens, 2022-02-23:
Not much. The wireshark trace should cover some time before the problem start and then a few more minutes after the problem has occurred.

steine01, 2022-02-28:
This weekend it happened again with the same setup, but I logged the network activity for the connection. I attached a couple of minutes of the log before and after the crash at ~11:38 as a CSV file. For the remaining log, there were no further connection attempts with the Kasli after this point. If I attempt to run any experiments, I find the same issue as before.

It looks as if at 11:38:26 three Keep-Alive packets are sent by the host, but no response from Kasli arrives. Afterwards, there are a couple of ARP requests, both to the MAC address of the Kasli and general broadcasts. Then, no broadcasts for the Kasli occur anymore (probably, because the experiment terminates with the error in this issue?)
[wireshark_log_artiq_crash.csv](https://github.com/m-labs/artiq/files/8151756/wireshark_log_artiq_crash.csv)
.

jordens, 2022-02-28:
Doesn't look like an obvious networking issue, more like a crash. Are you still using the exact same experiment that you mentioned at the top and ARTIQ versions on device and your computer?
Could you have a look at the console output around the issue?
If you ran zadig on all four serial interfaces, then you won't see any. Revert the console interface back to standard serial port. The console is interface 2 (the third) that is being exposed over the USB connection.

steine01, 2022-02-28:
To test, yes I am using the exact experiment I mentioned above. The sequence during which the crash happens has been modified between then and now, but uses the same components (mainly Urukul and TTL). I just checked the version, it is the same on the computer and should be the same on the device, since we did not flash a new version onto the device. But we also had this issue for several versions and at least two different Kasli over time.

For the console: I changed the Interface 2 driver to USB Serial (CDC) and checked the serial connection with `python -m serial.tools.miniterm`. I then see the port in the list, but cannot access it with the following error message:

```
(artiq-6-2) C:\Users\ybion443>python -m serial.tools.miniterm

--- Available ports:
---  1: COM3                 'Quad RS232-HS (Interface 2) (COM3)'
--- Enter port index or full name: 1
could not open port 'COM3': Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)

```
Do I need to configure the other interfaces of the Quad-USB interface as well? Or is there something else wrong?

jordens, 2022-02-28:
That looks a bit like [Jonas' issue](https://github.com/m-labs/artiq/issues/1487#issuecomment-660995196) which seems to be resolved but unknown (to me) how.
Could you quickly try hyperterm/putty/whatever from another (without zadig modifications) computer?

steine01, 2022-02-28:
I checked the serial connection with hterm on another PC, that did not have Zadiq installed. There I see the connection on COM4, and I can connect to it. But the received data is empty. Should I restart the Kasli and check again? Is a continous log feasible here as well, or is there too much data being written on the serial bus?

jordens, 2022-02-28:
Use the device manager to figure out what port number windows assigned to interface 2. Connect to that. There is only logging output if something is happening (booting, running new kernels).

steine01, 2022-02-28:
Well, this is the only serial port I see in the device manager and hterm, no port is labeled as Interface 2 (or any interface). If I disconnect the USB cable on the Kasli, this port disappears. After reconnecting I get a byte `00` and nothing else. Submitting an experiment also causes no output and results after some time in the error mentioned in the beginning.

jordens, 2022-02-28:
And this is with the correct baudrate/stopbits/parity settings?

steine01, 2022-02-28:
This is with 115200 baud 1 stopbit , 8 data, no parity as sb10q mentioned in the forum.

jordens, 2022-02-28:
Just to check: are you trying to get logging output while the device is in this "crashed" state? Does the kernel actually run?
Is there output when the device boots?

steine01, 2022-02-28:
So far I have not power-cycled the Kasli, so it is in the crashed state. I speicifically asked earlier, if I should reboot it and said that experiments do not run currently. I do this, because I do not know, if you expect some messages on the serial connection in this state or not. I can power-cycle it, if you say that this is now the right moment.

jordens, 2022-02-28:
With "console output around the issue" (above) I mean a time interval from well before and well after the issue occurs.
With "There is only logging output if something is happening (booting, running new kernels)." I mean there is no output if the system isn't booting or (successfully) executing new kernels.
Yes. Please procure these logs.

steine01, 2022-02-28:
Ok, then I will power cycle and let the system run while logging the serial output. Since I so far cannot reproduce the crash, it might take some time, until it happens again.

jordens, 2022-02-28:
But to confirm, you can connect to the console on that other computer now and you see the boot log output?

steine01, 2022-02-28:
I connected to the port with hterm and after I power-cycled and reconnected, I saw many zeros written on the received data. After booting has finished though, I could not get any additional output so far. If I submit an experiment, it runs well and terminates, but I see no change on the serial connection. 

Also if I raise an RTIOUnderflow on purpose (which arrives in the dashboard), I see no output on the serial terminal. Is this expecteed? Is there some test I can do to check, if the serial connection works as intended? I checked the documentation and issues but found nothing specific there. [In the forum](https://forum.m-labs.hk/d/172-debugging-and-profiling-with-artiq/4), there is a mention of `artiq_coremgmt --device-db <path_to_device_db.py> debug allocator`, that should print memory info on the serial output, but I see nothing there.

jordens, 2022-02-28:
Do not power-cycle and then reconnect. Just remain connected and power cycle.
Do one thing at a time. Let's get some messages when booting first. The log should look as described elsewhere, e.g. https://github.com/m-labs/artiq/issues/1776

steine01, 2022-02-28:
When I just do that, I only see many zeros on the terminal monitor, it does not look like the example you provided:
[image]

Maybe it is easier to comunicate in Mattermost?

jordens, 2022-02-28:
I'm not often on mattermost but usually at https://matrix.to/#/#quartiq:matrix.org Let's go there.

steine01, 2022-05-01:
After installing a satellite and running the experiment for a longer amount of time, the connection loss happened again and this time I also checked the serial connection. This is the complete output:

```
panic at src/libcore/slice/mod.rs:1965:5: index 32 out of range for slice of length 20<\n>
backtrace for software version 6.7659.c6a7b8a8;ptb3master:<\n>
0x40032180<\n>
0x4002341c<\n>
0x40023b88<\n>
0x400279e8<\n>
0x40030200<\n>
0x400318c0<\n>
halting.<\n>
use `artiq_coremgmt config write -s panic_reset 1` to restart instead<\n>
```

The packets seen by Wireshark are similiar as before:
[image]

sbourdeauducq, 2022-05-02:
Can you decode the backtrace with or1k-elf-addr2line?

hartytp, 2022-05-02:
Most of the time when I've hit panics like this it's turned out to be due to a compiler issue like https://github.com/m-labs/artiq/issues/1677

steine01, 2022-05-02:
I hope I did this correctly, I never used this tool before. This is the command I used and the ouput corresponding to the adresses from the serial monitor:

```
or1k-linux-addr2line @C:\Users\ybion443\Desktop\addresses.txt -e C:\Users\ybion443\.conda\envs\artiq-6-3\Lib\site-packages\artiq\board-support\kasli-ptb3master\runtime.elf
/nix/store/4k66f67dvw6akyzdkfn0a3849czsf3yz-python3.8-artiq-6.7659.c6a7b8a8/lib/python3.8/site-packages/artiq/firmware/libunwind_backtrace/lib.rs:42
/build/source/src/libcore/panicking.rs:92
/build/source/src/libcore/slice/mod.rs:1965
/build/source/src/libcore/slice/mod.rs:2142
/nix/store/nibcfajv0yswvxvd0dffylxgcvqn23bh-artiq-firmware-cargo-vendored/registry/smoltcp/src/iface/ethernet.rs:595
/build/source/src/libcore/ops/function.rs:223
```

hartytp, 2022-05-02:
https://github.com/smoltcp-rs/smoltcp/issues/527

steine01, 2022-05-03:
@hartytp Thank you for the link to this issue, that seems to fit pretty well to what I see. I will try the suggestion at the end for now, but I don't think this is a good long-term solution. 

What is the status of the 0.8 update of smoltcp for ARTIQ 6 (which is suggested in the issue as a possible fix)? In my Cargo.tml file, version 0.6 is still included, while in the repository version 0.8 is included. The latest build of conda-artiq (120931) still references version 0.6 though (I just downloaded it and checked).

jordens, 2022-05-03:
Use the beta builds for your systems to try that.

steine01, 2022-05-03:
Unless I am mistaken, the earliest available beta build on Hydra (https://nixbld.m-labs.hk/jobset/artiq/main-beta) is an ARTIQ 7 build (3632). When I tried an earlier ARTIQ 7 build, most of my experiments broke because of typing issues (e.g. `ad9910.cfg_sw` now needs a `bool` instead of an `int`), so using that would require that I rewrite many of my experiments. While this may be necessary in the future, I think it is a bit much just to fix this bug.

jordens, 2022-05-03:
The alternatives are likely quite a bit more involved. Rewriting your experiments is not required now. Just verifying that there is a fix. It would be great if you could test it, maybe on a system that isn't used in production but where you can still reproduce it.

steine01, 2022-05-03:
Ok I will have to see if I can setup a system that provokes the error. Maybe also interesting: I cannot restart the Kasli with the artiq_coremgmt command suggested on the serial terminal. Instead I receive the error of the issue [WinError 10060].

jordens, 2022-05-03:
Yes. If it's crashed you won't be able to restart it using `coremgmt`.

steine01, 2022-10-21:
Maybe an update of this issue for our lab: I have commented out the line `smoltcp::phy::EthernetTracer::new(net_device, net_trace_fn)` in `~\artiq\firmware\runtime\main.rs ` for two of our systems and have so far not observed another network error over the last few months. I don't quite understand how this fixed the problem though, since I would have expected, that some kind of recompilation is necessary for the change to have an effect. Unless this part is compiled again for each experiment?