m-labs/artiq #1768: unexpected underflow with Zotino and RISC-V

sbourdeauducq, 2021-10-16:
artiq_sinara_tester on the master branch currently underflows during the Zotino test.

This works around it:
```
--- a/artiq/frontend/artiq_sinara_tester.py
+++ b/artiq/frontend/artiq_sinara_tester.py
@@ -417,6 +417,7 @@ class SinaraTester(EnvExperiment):
         delay(200*us)
         i = 0
         for voltage in voltages:
+            delay(100*us)
             zotino.write_dac(i, voltage)
             delay(100*us)
             i += 1
```

Increasing the second delay (below ``write_dac``) does NOT make the underflow disappear, which suggests a possible compiler issue.

dnadlinger, 2021-10-16:
Just looking at the snippet you posted, couldn't it also be the first iteration?

Spaqin, 2022-04-27:
No underflow on current version of ARTIQ-7 and NAC3. Also tested on a mid-November version and could not reproduce.

closed 2022-04-27 by sbourdeauducq