m-labs/artiq #1776: Unable to use Kasli kernel after reflashing gateware

clayton-ho, 2021-10-30:
# Bug Report

## One-Line Summary

ARTIQ experiments are unable to use kernel after reflashing gateware.

## Issue Details

### Steps to Reproduce

1. Enter nix shell using nix-shell "<artiq-full/fast/shell-dev.nix>"
2. Build gateware using python -m artiq.gateware.targets.kasli
3. Flash gateware using artiq_flash --srcbuild -d artiq_kasli -V ucla
4. Submit an experiment with run as a kernel method and call self.core.reset()

### Expected Behavior

Experiment should run all the way through and finish.

### Actual (undesired) Behavior

* The experiment stops whenever we call a function that uses the hardware (e.g. self.core.reset) or a function with the kernel decorator
* After sufficiently long time, we get a worker timeout (the following message uses a kernel function in the prepare method since the timeout happens more quickly than if we use a kernel function in the run method, in which case the timeout takes over an hour to happen):

[image]

* However, we can still connect to the Kasli core via USB and see the startup messages. It seems like the core still recognizes that we have submitted an experiment to the master.

[image]

* TTLs are also still accessible through the dashboard

* Sometimes, flashing the gateware hangs halfway through, nearly always after sector 43. In this case, we get the following warning repeatedly for longer and longer times: "Warn : Haven't made progress in mpsse_flush() for 4109ms"

### Your System (omit irrelevant parts)

* Operating System: Windows 10 (for using hardware), Ubuntu 18.04.6 in VirtualBox (for flashing gateware)
* ARTIQ version: 6.7621.d9b01ed8
* Version of the gateware and runtime loaded in the core device: 6.7621.d9b01ed8
* Hardware involved: Kasli v1.1
* Xilinx version: 2020.1

sbourdeauducq, 2021-10-30:
>     * Sometimes, flashing the gateware hangs halfway through, nearly always after sector 43. In this case, we get the following warning repeatedly for longer and longer times: "Warn : Haven't made progress in mpsse_flush() for 4109ms"

Probably a bad USB cable.

clayton-ho, 2021-10-31:
> > ```
> > * Sometimes, flashing the gateware hangs halfway through, nearly always after sector 43. In this case, we get the following warning repeatedly for longer and longer times: "Warn : Haven't made progress in mpsse_flush() for 4109ms"
> > ```
> 
> Probably a bad USB cable.

I tried replacing the cable, using a different USB port, unplugging other USB devices, all without any success.

sbourdeauducq, 2021-10-31:
It really looks like some USB problem, look at the errors in the UART output as well (missing characters). Try another computer and/or without the VM.

sbourdeauducq, 2021-10-31:
There are Linux live-CDs and live-USBs that you can use if you don't want to wipe your Windoze installation.

clayton-ho, 2021-11-02:
dual-booting ubuntu and reflashing the gateware from there fixed the issue.

thanks!

closed 2021-11-02 by clayton-ho

sbourdeauducq, 2021-11-03:
If your hardware and BIOS support it (it's hit or miss with consumer PCs), you can also try to redirect the USB controller to the VM at the PCIe level using the IOMMU. This works around most VM USB redirection related bugs in my experience.