m-labs/artiq #1779: Aux packet timeout/CRC errors for certain packet types

HarryMakes, 2021-11-12:
# Bug Report

## One-Line Summary

As of 80115fcc022400e2f0e914481a1f13dd6d33e70b, aux packet errors will be emitted when some types of DRTIO aux packets are transmitted.

## Issue Details

In attempting transmission of an aux packet for the following actions, aux packet errors will occur (NB this list might be incomplete, and does not specify the exact `drtioaux_proto::Packet` type since these types of packets have yet to be tested separately):
* MonInj operations on a device located at a satellite crate
* JESD204B DAC operations (e.g. AD9154 init, sync) on a Sayma satellite issued from a Metlino master

However, it is shown that certain actions as follows does not trigger any packet errors:
* I2C operations on the PCA9548 I2C switch located at the satellite Kasli

The full list of available DRTIO aux packet types is in [artiq/firmware/libproto_artiq/drtioaux_proto.rs](https://github.com/m-labs/artiq/blob/80115fcc022400e2f0e914481a1f13dd6d33e70b/artiq/firmware/libproto_artiq/drtioaux_proto.rs#L18-L59)).

### Steps to Reproduce

Example: triggering errors for MonInj actions:

1. Prepare a simple DRTIO setup with one Kasli as master and one Kasli as satellite. Define an 8-channel TTL card on the satellite description, where it has 4 inputs and 4 outputs.
2. Generate device_db.py with artiq_ddb_template.
3. Rename one `TTLInOut` (input) as `loop_in` and one `TTLOut` (output) as `loop_out`.
4. Connect a corresponding TTL card to the satellite and connect those two channels together (e.g. with SMA cable)
5. Set environment variable `ARTIQ_ROOT` to the directory containing device_db.py.
6. Run: `python -m unittest -v artiq.test.coredevice.test_moninj.MonInjTest`

### Expected Behavior

No aux packet errors will be emitted.

### Actual (undesired) Behavior

* The output from running the unit test reports failure.
<details>
<summary>Unit test output:</summary>

```
test_moninj (artiq.test.coredevice.test_moninj.MonInjTest) ... FAIL

======================================================================
FAIL: test_moninj (artiq.test.coredevice.test_moninj.MonInjTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/nix/store/zrvp7n909s4ppbhqdjyhx32h5p1gr7yb-python3-3.8.11-env/lib/python3.8/site-packages/artiq/test/coredevice/test_moninj.py", line 58, in test_moninj
    self.assertEqual(notifications, [
AssertionError: Lists differ: [(65536, 0, 0)] != [(65536, 0, 0), (65536, 0, 1), (65536, 0, 0)]

Second list contains 2 additional elements.
First extra element 1:
(65536, 0, 1)

- [(65536, 0, 0)]
+ [(65536, 0, 0), (65536, 0, 1), (65536, 0, 0)]

----------------------------------------------------------------------
Ran 1 test in 2.016s
```
</details>

* Kasli master will generate aux packet **timeout** errors.
<details>
<summary>Kasli master log excerpt:</summary>

```
[    68.005968s]  INFO(runtime::moninj): new connection from 192.168.1.116:38542
[    68.214317s] ERROR(runtime::moninj::remote_moninj): aux packet error (timeout)
[    68.623256s] ERROR(runtime::moninj::remote_moninj): aux packet error (timeout)
[    68.830753s] ERROR(runtime::moninj::remote_moninj): aux packet error (timeout)
[    69.219720s] ERROR(runtime::moninj::remote_moninj): aux packet error (timeout)
[    69.427243s] ERROR(runtime::moninj::remote_moninj): aux packet error (timeout)
[    69.634744s] ERROR(runtime::moninj::remote_moninj): aux packet error (timeout)
[    70.042755s] ERROR(runtime::moninj::remote_moninj): aux packet error (timeout)
[    70.250278s] ERROR(runtime::moninj::remote_moninj): aux packet error (timeout)
[    72.516016s]  INFO(runtime::mgmt): new connection from 192.168.1.116:40546
```  
</details>

* Kasli satellite will generate aux packet **CRC** errors.
<details>
<summary>Kasli satellite log excerpt:</summary>

```
[    68.542509s]  WARN(satman): aux packet error (packet CRC failed)
[    68.951069s]  WARN(satman): aux packet error (packet CRC failed)
[    69.158713s]  WARN(satman): aux packet error (packet CRC failed)
[    69.366804s]  WARN(satman): aux packet error (packet CRC failed)
[    69.371445s]  WARN(satman): aux packet error (gateware reported error)
[    69.377981s]  WARN(satman): aux packet error (packet CRC failed)
[    69.547567s]  WARN(satman): aux packet error (packet CRC failed)
[    69.755164s]  WARN(satman): aux packet error (packet CRC failed)
[    69.962726s]  WARN(satman): aux packet error (packet CRC failed)
[    70.170582s]  WARN(satman): aux packet error (packet CRC failed)
[    70.175230s]  WARN(satman): aux packet error (packet CRC failed)
[    70.370694s]  WARN(satman): aux packet error (packet CRC failed)
[    70.578244s]  WARN(satman): aux packet error (packet CRC failed)
[    70.786112s]  WARN(satman): aux packet error (packet CRC failed)
```  
</details>

## Suggestions

It is possible that the shift in CPU endianness from big to little in ARTIQ-7 is related to this issue. With reference to possibly relevant changes to DMA (051a14abf22395cd7d80defa5d0eb4c799921b54) and analyser (a833974b5067d66ab9fdf61fb928ac3f4e8ba991), I've tried out some solutions that consistently remove aux packet errors (tested between Metlino and Sayma):

* Byte-swapping on the memory port for the aux controller
* Reversing bytes inside the `Converter` of the aux transmitter and receiver

Although not yet verified, I cannot seem to replicate the aux errors for some aux packet types with data of types smaller than a 32-bit. An example is the I2C request packet types. I have been working out a clear explanation for such a difference, but it will take some time.

occheung, 2021-11-18:
Replicated using Kasli v1.1 as master, Kasli v2.0 as satellite.
```
Recv buffer: [81, 1, 0, 0, 0, 0, 0, 0, 133, 39, 217, 90, 0, 0, 0, 0]
[    26.098722s]  WARN(satman): aux packet error (packet CRC failed)
```
`[133, 39, 217, 90]` should be the CRC32 of `[81, 1, 0, 0, 0, 0, 0, 0]` in little-endian bytes, but the bytes got padded to 64-bits alignment so the CRC32 was read as 0.
Seems like another case of CRC32 being in the wrong place: https://github.com/m-labs/misoc/pull/124/commits/df249b5079fa4678625d1984510841466b49ab98

occheung, 2021-11-19:
This is caused by the bit alignment assumption in DRTIO gateware, where tx/rx length must be a multiple of data width.

Inserting the CRC at byte offset 4 (8 bytes alignment) would solve the issue, like the following:
```
        let padding = (12 - (writer.position() % 8)) % 8;
        for _ in 0..padding {
            writer.write_u8(0)?;
        }
```
For example, an monitor request would look like this in the DRTIO buffer.
```drtio recv: [65, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 48, 226, 157, 17]```

There could still be checksum issue if the DRTIO system involves older versions/Zynq. It can be solved by either updating their DRTIO message protocol, or perform checksum verification on both offset 0 and offset 4.

```
        let checksum_at = buffer.len() - 8;
        let checksum = crc::crc32::checksum_ieee(&reader.get_ref()[0..checksum_at]);
        reader.set_position(checksum_at);
        if reader.read_u32()? != checksum {
            let padded_checksum = crc::crc32::update(
                checksum, &crc::crc32::IEEE_TABLE, &reader.get_ref()[checksum_at..checksum_at + 4]
            );
            reader.set_position(checksum_at + 4);
            if reader.read_u32()? != padded_checksum {
                return Err(Error::CorruptedPacket)
            }
        }
```

sbourdeauducq, 2021-11-19:
Let's just change the DRTIO protocol on 7.

PR #1786 (2021-11-24): Fix DRTIO message alignment
# ARTIQ Pull Request

## Description of Changes
DRTIO messages are now 64-bits aligned. See discussion in #1779.

## Tests
Tested with Kasli v1.1 and Kasli v2.0 using DRTIO.

### Related Issue
Closes #1779

## Type of Changes

|   | Type |
| ------------- | ------------- |
| ✓  | :bug: Bug fix  |

## Steps (Choose relevant, delete irrelevant before submitting)

### All Pull Requests

- [x] Use correct spelling and grammar.
- [x] Update [RELEASE_NOTES.rst](../RELEASE_NOTES.rst) if there are noteworthy changes, especially if there are changes to existing APIs.
- [x] C [... cut]
closed 2021-11-24 by sbourdeauducq