m-labs/artiq #1781: how to initialize a list of hardware channels

SadFdW, 2021-11-17:
# Question
how to initialize a list of hardware channels with a loop

## Category: FILL_IN

## Description

I am a newbie of python, artiq and github. I have a question about how to initialize a list of hardware channels. For example, I have a channel list chnlist = ["ttl0", "ttl1", "ttl2"], how to set all the three channels to output mode with a loop in the kernel?

'''
from artiq.experiment import *

chnlist = ['ttl0', 'ttl1', 'ttl2']

class test(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        for chn in chnlist:
            self.setattr_device(chn)

    @kernel
    def run(self):
        self.core.reset()
        self.ttl0.output()
        self.ttl1.output()
        self.ttl2.output()
        delay(200*us)
        self.ttl0.pulse(100 * us)
        self.ttl1.pulse(100 * us)
        self.ttl2.pulse(100 * us)
'''
how to write the "@kernel run(self)" function with a loop?

hartytp, 2021-11-17:
I think what you're looking for is `getattr_device`. So in pseudocode something like
```python

def build(self):
    self.ttls = [self.getattr_device(f'ttl{n}' for n in range (10, 12)]

@kernel
def run(self):
    for ttl in self.ttls:
        ttl.output()
    delay(200*us)

    for ttl in self.ttls:
        ttl.pulse(100*us)
```

dnadlinger, 2021-11-17:
`get_device`?

closed 2021-11-18 by sbourdeauducq

SadFdW, 2021-11-18:
Hartytp and dnadlinger, thx a lot! The code can work if I replace the *getattr_device* with *get_device*.

hartytp, 2021-11-18:
Thanks @dnadlinger! Should have checked the api rather than posting from memory.