m-labs/artiq #1784: with parallel: time error with loop

SadFdW, 2021-11-18:
# Question
with parallel: time error with loop

## Category: FILL_IN

## Description
I have a question about the *with parallel*. If I use some loop function, the output time sequence is incorrect, while it is corrent for the "regular experssion".  How to solve this problem when using a loop function?
Please see the code below
```
from artiq.experiment import *

chnlist = ['ttl0', 'ttl1', 'ttl2']

class test(EnvExperiment):
    def build(self):
        self.setattr_device('core')
        for ele in chnlist:
            self.setattr_device(ele)
        self.ttls = [self.get_device(f'{ele}') for ele in chnlist]

    @kernel
    def run(self):
        self.core.reset()
        for ele in self.ttls:
            ele.output()

        delay(1000 * us)
        with parallel:
            for ele in self.ttls:
                ele.pulse(100 * us)
            # self.ttl0.pulse(100 * us)
            # self.ttl1.pulse(100 * us)
            # self.ttl2.pulse(100 * us)
```
the loop expression is 
```
        with parallel:
            for ele in self.ttls:
                ele.pulse(100 * us)
```
the result of which is 
[image]
 the regular express is
```
        with parallel:
            self.ttl0.pulse(100 * us)
            self.ttl1.pulse(100 * us)
            self.ttl2.pulse(100 * us)

```
[image]

sbourdeauducq, 2021-11-18:
Please use the forum for this type of question https://forum.m-labs.hk/

sbourdeauducq, 2021-11-18:
In a parallel block, each statement is inside an implicit ``sequential`` block (this should be in the manual). This includes the for loop statement.
This behavior will be changed in NAC3 ARTIQ (most likely ARTIQ-8) where only function calls in a ``parallel`` block have an implicit ``sequential``, and it will do what you want.

closed 2021-11-18 by sbourdeauducq

SadFdW, 2021-11-18:
> In a parallel block, each statement is inside an implicit `sequential` block (this should be in the manual). This includes the for loop statement. This behavior will be changed in NAC3 ARTIQ (most [...]

Thx a lot!

dhslichter, 2021-11-18:
> This behavior will be changed in NAC3 ARTIQ (most likely ARTIQ-8) where only function calls in a parallel block have an implicit sequential, and it will do what you want.

@sbourdeauducq this appears to be a breaking change to the `parallel` context as discussed in #1555, where I strongly advocated against such breaking changes because of the impact on legacy code.  Why is this occurring?  The compromise discussed in #1555 was to make a separate context (e.g. `deep_parallel` or something).  Can we discuss? I propose discussion be carried out in #1555 and not here.