m-labs/artiq #1785: ARTIQ WhiteRabbit on Kasli v2 discussion

jbqubit, 2021-11-23:
Heads up about a [new forum](https://chat.m-labs.hk/m-labs/channels/whiterabbit) for discussing the ARTIQ implementation of White Rabbit for Kasli v2.0. Current status is that the PLLs don't lock. While sorting out implementation details we've elected to use the M-Labs Forum. Once things are working well enough that it's possible to define discrete bugs, error reporting will revert to github Issue tracking. 

Note that if you're wanting to learn more about WR the best place to start is the [thesis](https://white-rabbit.web.cern.ch/documents/Precise_time_and_frequency_transfer_in_a_White_Rabbit_network.pdf) of Tomasz Włostowski (Warsaw University of Technology, 2011; advised by @gkasprow and others). 

WR, DDMTD, DMTD, WhiteRabbit, Kasli v2, Kasli v2.0

closed 2021-11-23 by jbqubit

gkasprow, 2021-11-23:
Actually, I"m supervising one student whose task is to make WR-PLL running on Kasli. He has experience with digital PLLs.

sbourdeauducq, 2021-11-23:
How's his progress?

gkasprow, 2021-11-23:
He just started a few weeks ago. I gave him all the docs. I pinged him.

sbourdeauducq, 2021-11-23:
Please give him that chat.m-labs.hk link.

jbqubit, 2022-01-15:
@gkasprow When you say you gave your  student (@d_sudowski) "all the [wrpll] docs" what did you have in mind? Here's the list of salient discussions that are on my radar. Posting here for posterity -- please include additions or documentation that we should collectively have in mind. 

GOAL: Pull together list of discussions on Sinara hardware and ARTIQ support related to DRTIO clock synchronization and WRPLL development. 

Locations where discussion took place.
* https://github.com/sinara-hw/meta/issues
* https://github.com/m-labs/artiq
* https://github.com/sinara-hw/sinara
* https://github.com/sinara-hw/kasli
* https://chat.m-labs.hk/m-labs OLD IRC LOGS

__Initial discussions__ of clock recovery and WRPLL development in the Sinara project
* [#515 DRTIO clock recovery/Si5324 in Sinara](https://github.com/sinara-hw/sinara/issues/515) -- broad discussion of clock recovery (2018)

__Kasli WRPLL hardware development__
* [#15 Clock recovery in Sinara/ARTIQ](https://github.com/sinara-hw/meta/issues/15) -- discussion planning WRPLL for Kasli 2.0 (2018)
	* @hartytp details his [Sinara-WR proposal in full](https://github.com/sinara-hw/meta/issues/15#issuecomment-445266610) 
	* @hartytp [BD of the WR test setup we used on the KC705](https://github.com/sinara-hw/meta/issues/15#issuecomment-454840325) -- nice diagram
	* @weidazhang [phase noise plot for Oxford tests of WRPLL on pair of KC705](https://github.com/sinara-hw/meta/issues/15#issuecomment-458743002) 
* [#19 WR PLL/ clock recovery](https://github.com/sinara-hw/Kasli/issues/19) @hartytp -- initial discussion (2018)
* [#25 Kasli 1.2 WR support](https://github.com/sinara-hw/Kasli/issues/25)  -- path leading v2.0 hardware on Kasli 
* [#39 WR oscillator Si549 characterization](https://github.com/sinara-hw/meta/issues/39) (2019)

__Sayma__ WRPLL hardware development
* [#36 AMC WR options](https://github.com/sinara-hw/Sayma_AMC/issues/36)   (2019)

__ARTIQ WRPLL development__
* Weida's WRPLL ARTIQ implementation [diagram](https://github.com/m-labs/artiq/blob/master/doc/wrpll_diagram.png) [posted](https://github.com/m-labs/artiq/commit/f7a5df8d81ff369cfb7d52c4864f0a6e9097573c) (12/2019)
    * above is surely outdated now... needs to be updated 
* [#1534 WRPLL on Kasli 2.0](https://github.com/m-labs/artiq/issues/1534) development status as of autumn 2020
	 * [phase unwrapping and loop filter kicks](https://github.com/m-labs/artiq/issues/1534#issuecomment-712378768) (10/2020)
	* wrpll support for Kasli v2 is part of [ARTIQ 7.0 release plan](https://github.com/m-labs/artiq/milestone/24)
* Collector inline [documentation](https://github.com/m-labs/artiq/blob/735cd1eb3e5c069481ff2ca10de6e91e96692980/artiq/gateware/drtio/wrpll/ddmtd.py#L112) 
* [#1785 ARTIQ WhiteRabbit on Kasli v2 discussion](https://github.com/m-labs/artiq/issues/1785) (2021)

sudowski27, 2022-01-19:
Hi. I have some ideas about modeling programmable oscilator, and maybe somebody have better or easier to implement idea. Firstly I have question. 

Will the oscillator(549bacb001937abg)  continuously change its frequency during operation, or will it only have one frequency at initiation? 

If have only one frequency the oscillator model can be static, and its easer to code in Python or Matlab.

If oscillator model have to be dynamic its qustion how model a delay of sending the i2c frame and writing the bits to the oscillator registers.

Any ideas or suggestions ?

gkasprow, 2022-01-22:
In WR we use two oscillators. One of them, the helper oscillator frequency does not change much. The other one is continuously adjusted with very small steps and its latency matters. The datasheet specifies:
[image]

gkasprow, 2022-03-15:
Use frequency detector instead of phase detector