m-labs/artiq #1788: Urukul gives proto_rev mismatch error

clayton-ho, 2021-11-27:
# Bug Report

## One-Line Summary

Get proto_rev mismatch error when using Urukul

## Issue Details

Scripts/experiments involving our Urukul cards give a proto_rev mismatch.
We originally bought a crate from m-labs with our Urukul cards connected to the Sample in an SU-Servo configuration.
To access the Urukuls individually, we tried to uncouple them by re-flashing the gateware with a new json file, and it started to give the proto_rev mismatch error.
We reflashed the Urukuls, making sure we used the v1.4 gateware to ensure compatibility with our Urukul hardware version (v1.5). However, we still get the proto_rev mismatch error.

### Steps to Reproduce

1. Reflashed Kasli using the fast development shell (nix-shell "<artiq-full/fast/shell-dev.nix>") and the attached json file
2. Flashed Urukul using a JTAG HS2 cable, Digilent Adept, and v1.4 of the Urukul firmware
3. Tried to run a script initializing and setting the Urukul

### Expected Behavior

Ideally we wouldn't get this error.

### Actual (undesired) Behavior

* The error:
[image]
* The new json file we used to flash Kasli:
[image]

### Your System (omit irrelevant parts)

* Operating System: Windows 10 Education 21H1
* ARTIQ version: ARTIQ v6.7604.040aa6f
* Version of the gateware and runtime loaded in the core device: 6.7621.d9b01ed8
* Hardware involved:
    * Kasli v2
    * Urukul v1.5

sbourdeauducq, 2021-11-27:
> We reflashed the Urukuls,

Which ESD precautions did you take?
Did you hotplug any EEM cables?

dnadlinger, 2021-11-27:
Did you change the Urukul DIP switches? For regular use, only the first one should be on, for SUServo, the first two.

clayton-ho, 2021-11-29:
> Did you change the Urukul DIP switches? For regular use, only the first one should be on, for SUServo, the first two.

Thanks! I didn't realize I had to do that. Did it and it fixed the issue for most of our Urukuls.

closed 2021-11-29 by sbourdeauducq

clayton-ho, 2021-11-29:
> > We reflashed the Urukuls,
> 
> Which ESD precautions did you take? Did you hotplug any EEM cables?

We made sure to keep the cards on an ESD mat and wore ESD wrist straps. I tried David's fix and it worked all of our cards except for one, which still gives the proto_rev mismatch error.

dnadlinger, 2021-11-29:
Another thing that might be useful to check would be the actual value of the incorrect proto_rev – whether it's e.g. all zeros or random noise might tell you something about the state of the electrical connections.

jbqubit, 2021-11-29:
https://github.com/sinara-hw/Urukul/issues/63