m-labs/artiq #1789: Automatically specify v1.x Kasli softcore architecture

lochsh, 2021-11-29:
# ARTIQ Feature Request

## Problem this request addresses

In artiq/coredevice/core.py, `"rv32g"` is the default argument for the `target` kwarg to `Core.__init__`.

https://github.com/m-labs/artiq/blob/c6039479e47e56743657355184c5f249e5137743/artiq/coredevice/core.py#L74

Is the expectation that users using core devices with v1.x hardware revision should add something along these lines to their `device_db.py`?

```python
device_db["core"]["arguments"]["target"] = "rv32ima"
```

This seems like it's the wrong level for the user to be specifying.

This isn't an issue I have, but I also wonder if this means you could not have a v2.x and v1.x Kasli in the same system?

## Describe the solution you'd like

It seems like it would be more convenient for the user for this to be derived from the hardware revision instead -- though I'm afraid I don't know exactly how that data flow would look.

## Additional context
`artiq.gateware` code takes into account the hardware revision correctly in order to decide whether to build rv32ima or rv32g gateware for the Kaslis. The issue is the host-side software expecting an rv32g CPU when it should be expecting an rv32ima CPU. Hence we are hitting this error in the firmware when running kernel code https://github.com/m-labs/artiq/blob/master/artiq/firmware/libdyld/lib.rs#L229

dnadlinger, 2021-11-29:
Note aside: Do you also have -3 speed grade v1.x Kaslis at Oxford Ionics? In that case, you might just want to build with the FPU instead – the issue still remains for people with the "default" -2 speed grade v1.x Kaslis, of course.

lochsh, 2021-11-29:
@hartytp was saying today that we do indeed have the -3 speed grade and that we can indeed build the rv32g gateware instead. That's probably sensible for us to do -- I'll have to think exactly how best to make it happen. Thanks.

As you say, the issue still remains.

airwoodix, 2021-11-29:
Shouldn't the target be properly set in the [`artiq_ddb_template`](https://github.com/m-labs/artiq/blob/c6039479e47e56743657355184c5f249e5137743/artiq/frontend/artiq_ddb_template.py#L15-L18) (since 750b0ce)?

sbourdeauducq, 2021-11-30:
> Shouldn't the target be properly set in the artiq_ddb_template (since 750b0ce)?

Yes it is correctly set.

> This isn't an issue I have, but I also wonder if this means you could not have a v2.x and v1.x Kasli in the same system?

You can mix Kasli 1 and Kasli 2 in ARTIQ-7 DRTIO systems.

closed 2021-11-30 by sbourdeauducq

lochsh, 2021-11-30:
I couldn't find documentation for how to use the artiq_ddb_template -- is there an issue I can follow for that being written, or a link to some existing documentation I have missed? I appreciate there hasn't been a release including this yet.