m-labs/artiq #1790: llvmlite dependence not updated when updating artiq through conda

fanmingyu212, 2021-11-30:
# Bug Report

## One-Line Summary

llvmlite package version is `0.37.0` after updating `artiq` through conda. However, llvmlite >=`0.38` is needed for calling `IRBuilder.call` with `arg_attrs` keyword argument (introduced in #1778).

## Issue Details

This bug appears after updating `artiq` and its dependences using `conda install artiq --update-deps`. llvmlite 0.37 does not support `arg_attrs` (https://github.com/numba/llvmlite/blob/release0.37/llvmlite/ir/builder.py#L874-L884).

It prevents experiments using kernel to run, as simple as

```python
from artiq.experiment import *

class Test(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        self.core.reset()
```

The error message when running the above experiment is:

```python
root:Terminating with exception (TypeError: call() got an unexpected keyword argument 'arg_attrs')
Traceback (most recent call last):
  File "C:\Users\scientist\code\artiq\artiq\master\worker_impl.py", line 300, in main
    exp_inst.run()
  File "C:\Users\scientist\code\artiq\artiq\language\core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Users\scientist\code\artiq\artiq\coredevice\core.py", line 131, in run
    self.compile(function, args, kwargs, set_result)
  File "C:\Users\scientist\code\artiq\artiq\coredevice\core.py", line 114, in compile
    library = target.compile_and_link([module])
  File "C:\Users\scientist\code\artiq\artiq\compiler\targets.py", line 199, in compile_and_link
    return self.link([self.assemble(self.compile(module)) for module in modules])
  File "C:\Users\scientist\code\artiq\artiq\compiler\targets.py", line 199, in <listcomp>
    return self.link([self.assemble(self.compile(module)) for module in modules])
  File "C:\Users\scientist\code\artiq\artiq\compiler\targets.py", line 151, in compile
    llmod = module.build_llvm_ir(self)
  File "C:\Users\scientist\code\artiq\artiq\compiler\module.py", line 92, in build_llvm_ir
    return llvm_ir_generator.process(self.artiq_ir,
  File "C:\Users\scientist\code\artiq\artiq\compiler\transforms\llvm_ir_generator.py", line 433, in process
    self.process_function(func)
  File "C:\Users\scientist\code\artiq\artiq\compiler\transforms\llvm_ir_generator.py", line 586, in process_function
    llinsn = getattr(self, "process_" + type(insn).__name__)(insn)
  File "C:\Users\scientist\code\artiq\artiq\compiler\transforms\llvm_ir_generator.py", line 1366, in process_Call
    llresult = self.llbuilder.call(llfun, llargs, name=insn.name,
TypeError: call() got an unexpected keyword argument 'arg_attrs'
```

A temporary workaround is to install llvmlite using `conda install -c numba llvmlite` which installs llvmlite-0.38.0rc1. I am not sure how this affects future updates.

### Your System (omit irrelevant parts)

* Operating System: Windows 10
* ARTIQ version: Latest master https://github.com/m-labs/artiq/commit/c6039479e47e56743657355184c5f249e5137743
* Version of the gateware and runtime loaded in the core device: N/A
* If using Conda, output of `conda list`: [conda_list.txt](https://github.com/m-labs/artiq/files/7623003/conda_list.txt)
* Hardware involved: N/A

sbourdeauducq, 2021-11-30:
The ARTIQ-7 branch currently does not support Windoze and will not do until the llvmlite 0.38 release and publication of upstream packages, which thankfully should happen next month.

I'm sick to death of a constant stream of Windoze problems such as:
* CMake not finding Visual Studio
* CMake freezing in the middle of a build with no error message
* Parallel builds not working due to idiotic Windoze filesystem design, resulting in long rebuild times, compounding all the other issues
* CMake/Visual Studio randomly not finding libraries
* Conda dependency handler bugs constantly breaking builds with insane/nonsensical error messages
* Conda taking 10 minutes to install a few packages on a Ryzen 9 machine
* Various Visual Studio bugs e.g. the command-line compiler not working if you have not opened the GUI
* Various adware/spyware throughout the entire process, constant spamming for creation of Microsoft accounts, tons of bloatware, etc.
* etc. etc.

Please consider installing an operating system and kicking out Microsoft filthware out of your lab.

NB: MinGW/MSYS2 may be able to create a small bubble of relative sanity on Windoze (https://github.com/m-labs/artiq/issues/1471).

closed 2021-11-30 by sbourdeauducq

dhslichter, 2021-11-30:
@sbourdeauducq for all its many and well-documented failings -- and I hear and understand your frustration here -- Windows is the operating system for roughly half of ARTIQ installations, and many users do not have the easy option of using Linux because of various other constraints (including institutional IT policies, which are very hard to change).  NIST has a service contract with M-Labs to pay for Windows-related yak shaving (among other things) to ensure that ARTIQ runs and is supported on Windows with conda installation.  If there are major new problems with ARTIQ on Windows that require much more involved work, please reach out so we can discuss appropriate avenues to have them addressed.  For better or worse, it is critical to maintain Windows support for ARTIQ.  

Relatedly, we are running the Zynq branch of ARTIQ 7 on Windows and at last check it was still working.  Does this issue mean there will be challenges upstreaming this into the main ARTIQ repo and master branch?

sbourdeauducq, 2021-12-01:
The llvmlite patches are all merged upstream and will be available in the next release, which is planned soon along with conda packages (in conda-forge and numba channels).
Considering this situation and the problems above which flared up again when attempting to update LLVM or use an updated pre-built LLVM to compile llvmlite, it doesn't seem like a good use of our time to persist in attempting to rebuild llvmlite ourselves.
Eventually the plan is to have NAC3 cross-compiled from Linux using mingw, so we won't have to deal with any of this Visual Studio nonsense anymore. I'm not sure if it could be loaded into official Python and conda Python which are compiled with VS, but with the MSYS2 Python (which they patched to build without VS) it works fine. MSYS2/pacman is a superior package management solution anyway.

sbourdeauducq, 2021-12-02:
Proof-of-concept: https://nixbld.m-labs.hk/build/72484 - entirely cross-compiled using Nix flakes and without Visual Studio or other particularly low-quality software.
Using it currently requires manual fiddling with libstdc++/libffi/zlib/mcfgthreads DLLs (copying them from the Nix store into where Windows will find them), but otherwise it seems to work.

sbourdeauducq, 2021-12-27:
@fanmingyu212 it works now, but you need to use the ``numba`` conda channel to get the latest llvmlite (0.38rc).
In case the conda dependency "solver" craps out on you, these lists of packages work:
https://git.m-labs.hk/M-Labs/nix-scripts/src/branch/master/artiq-fast/windows/conda_noarch_packages-7.nix
https://git.m-labs.hk/M-Labs/nix-scripts/src/branch/master/artiq-fast/windows/conda_win-64_packages-7.nix