m-labs/artiq #1799: RPC issue causing kernel to stuck in running state under certain conditions

stevefan1999, 2021-12-14:
# Bug Report 

## Issue Details

This bug is seen plaguing the build [#75623](https://nixbld.m-labs.hk/build/75623) and build [#73252](https://nixbld.m-labs.hk/build/73252) previously, and it is seemingly repeatable under specific conditions. The bug lead us to here, which is where the unit test stopped in both Hydra executions:
https://github.com/m-labs/artiq/blob/4f723e19a6bc9e84c0eb627461fe625007307504/artiq/test/coredevice/test_embedding.py#L90

Indeed, after commenting out that line of code, the test pipeline continues smoothly. But after some close inspections, it seems like this probably isn't the entire story. I have written a Minimal Reproducible Code that replicated this issue pretty consistently.

<details>
<summary> Minimal reproducible code: </summary>

```python3
import numpy
from artiq.experiment import *

class KernelRoundtripStuck(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def roundtrip(self, obj, fn):
        fn(obj)

    def run(self):
        lam = lambda _: None
        for i in range(1, 5):
            self.roundtrip(numpy.array(["a", "b", "c"]), lam)
            self.roundtrip(numpy.array([[1, 2], [3, 4]], dtype=numpy.int32), lam)
        print("passed")
```
</details>

### Expected Behavior

It should pass no matter what

### Actual (undesired) Behavior

It stuck like this in the log below.

<details>
<summary>Click here to see the UART log</summary>

```
[ 36104.746180s]  INFO(runtime::session): no connection, starting idle kernel
[ 36104.751942s]  INFO(runtime::session): no idle kernel found
[ 36128.716513s]  INFO(runtime::session): new connection from <REDACTED>:47884
[ 36128.723274s] DEBUG(runtime::session): comm<-host SystemInfo
[ 36128.727924s] DEBUG(runtime::session): comm->host SystemInfo { ident: "7.7901.4f723e19.beta;nist_clock", finished_cleanly: true }
[ 36128.741635s] DEBUG(runtime::session): comm<-host LoadLibrary(...)
[ 36128.770242s] DEBUG(runtime::session): comm->kern LoadRequest(...)
[ 36128.775442s] DEBUG(runtime::session): comm<-kern LoadReply(Ok(()))
[ 36128.781197s] DEBUG(runtime::session): comm->host LoadCompleted
[ 36128.788080s] DEBUG(runtime::session): comm<-host RunKernel
[ 36128.792818s] DEBUG(runtime::session): comm<-kern RpcSend { async: false, service: 2, tag: [102, 58, 110], data: 0x4ffffeb4 }
[ 36128.803688s] DEBUG(runtime::session): comm->host RpcRequest { async: false }
[ 36128.810740s] DEBUG(proto_artiq::rpc_proto): send<2>(Float64)->None
[ 36129.818758s] DEBUG(runtime::session): comm<-host RpcReply { tag: [110] }
[ 36129.824387s] DEBUG(runtime::session): comm<-kern RpcRecvRequest(0x4ffffec0)
[ 36129.831094s] DEBUG(proto_artiq::rpc_proto): recv ...->None
[ 36129.836569s] DEBUG(runtime::session): comm->kern RpcRecvReply(Ok(0))
[ 36129.843004s] DEBUG(runtime::session): comm<-kern (async RPC)
[ 36129.848546s] DEBUG(runtime::session): comm->host RpcRequest { async: true }
[ 36129.855515s] DEBUG(runtime::session): [1, 0, 0, 0, 110, 0, 1, 0, 0, 0, 110]
[ 36129.862691s] DEBUG(runtime::session): comm<-kern RpcFlush
[ 36129.868170s] DEBUG(runtime::session): comm<-kern RunFinished
[ 36129.873476s] DEBUG(runtime::session): comm->host KernelFinished { async_errors: 0 }
[ 36129.924269s] DEBUG(runtime::session): comm<-host LoadLibrary(...)
[ 36129.952773s] DEBUG(runtime::session): comm->kern LoadRequest(...)
[... 139 lines cut ...]
[ 36132.588147s] DEBUG(runtime::session): comm<-kern RpcFlush
[ 36132.593470s] DEBUG(runtime::session): comm<-kern RunFinished
[ 36132.598931s] DEBUG(runtime::session): comm->host KernelFinished { async_errors: 0 }
[ 36132.695840s] DEBUG(runtime::session): comm<-host LoadLibrary(...)
[ 36132.724339s] DEBUG(runtime::session): comm->kern LoadRequest(...)
[ 36132.729603s] DEBUG(runtime::session): comm<-kern LoadReply(Ok(()))
[ 36132.735294s] DEBUG(runtime::session): comm->host LoadCompleted
[ 36132.742351s] DEBUG(runtime::session): comm<-host RunKernel
```
</details>

It seems like this issue is from numpy.

### Your System (omit irrelevant parts)

* Operating System: `NixOS 21.11`
* ARTIQ version: `ARTIQ v7.0-dev`
* Version of the gateware and runtime loaded in the core device:
```
[     0.003826s]  INFO(runtime): software ident 7.7901.4f723e19.beta;nist_clock
[     0.010781s]  INFO(runtime): gateware ident 7.7901.4f723e19.beta;nist_clock
```
* Hardware involved: KC705 (attached to `rpi-1`)

It is also reported there is some hardware fluke on KC705. Maybe we should take this to other hardwares for sure.

sbourdeauducq, 2021-12-14:
> It is also reported there is some hardware fluke on KC705.

Other than the problems due to the IP settings?

stevefan1999, 2021-12-16:
Somehow this is related to #1798, and I can pass the MRC building with PR #1802, but I should try more rounds of testing first.

stevefan1999, 2021-12-20:
Case resolved with #1802

closed 2021-12-20 by stevefan1999