m-labs/artiq #1808: Scheme of data saved with artiq_influxdb - Tag keys as field keys

steine01, 2021-12-21:
# Bug Report

## One-Line Summary

artiq_influxdb writes data to the database in a format not well compatible with the recommendations in Influx documentation.

## Issue Details

### Steps to Reproduce

1. Add  the following device to `device_db.py` (where ... is the Influx configuration)
```
"influx": {
        "type": "controller",
        "host": "::1",
        "port": 3248,
        "command": "artiq_influxdb ...
    }
```
2. Add the string "+servo.base.y" to the Influx patterns.
3. Check the format with which the data is stored in Influx.

### Expected Behavior

The data can be accessed with `SELECT mean("servo.base.y") FROM "table" WHERE time >= t1 AND time <= t2 GROUP BY time("10s") fill("previous")`.

### Actual (undesired) Behavior

* The data can only be accessed with  `SELECT mean("float") FROM "table" WHERE "dataset='servo.base.y'" AND time >= t1 AND time <= t2 GROUP BY time("10s") fill("previous")`
* The key of the data stored in the field is a tag and the field key is more or less meaningless.
* According to the documentation (https://docs.influxdata.com/influxdb/v1.8/concepts/schema_and_data_layout/), numeric data should be stored in fields and grouping metadata in tags. 
* For many datasets, this results in many queries with different tags for connected data (i.e. several servos and excitations of the transitions of a single clock) instead of one query with several fields.

### Your System (omit irrelevant parts)

* Operating System: Windows 10
* ARTIQ version: ARTIQ v6.7621.d9b01ed8
* Hardware involved: InfluxDB

jordens, 2021-12-21:
It's unclear to me what your problem is, what solution you are actually proposing and what advantage it would bring. I also  don't think your analysis and understanding of good and bad influxdb schemata or even the query language is fully correct.
Numerical data is in fact stored in fields and grouped by metadata in tags. Increasing the number of fields (to the does not scale and is explicitly advised against. Querying multiple datasets can easily and directly be done in a single query. Just group by dataset and expand the where clause. We've been doing this for a long time without problem. The cardinality of the dataset tag is exactly right in common use cases.
The field name is extremely meaningful as fields need to be type-homogeneous. Having just one field does not work. Having one field per dataset does not scale. Having one measurement per dataset doesnt scale well either.

steine01, 2021-12-22:
I made the report short on purpose, maybe let me explain a bit more what the issue is:

1. I propose, that the structure of the way data is saved from Artiq to Influx can be configured. Currently, only the tag-based identifiers are possible, which in my opinion do not fit well to the way data should be stored in Influx. The current behaviour can of course stay as default to not break compability. I opt for a configuration, because I think the optimal way to structure the data in the database can be very different for different experiments or groups.

2. Yes the data itself is stored in the fields. but I do not see why the grouping should be done to this extent in the tags. I would definitely use a tag for e.g. the experiment or all servo data and filter my data for that tag in the query. But within that I would expect to have different fields corresponding to the different data that is recorded. As an example:

We have datasets: `servo_1.y, servo_2.y, servo_1.exc, servo_2.exc` from the experiment `clock_a` in the table `clock_data`
I would expect to access these with `SELECT mean("servo_1.y"), mean("servo_2.y"), mean("servo_1.exc"), mean("servo_2.exc") FROM "clock_data" WHERE "experiment" = "clock_a" AND time >= t1 AND time <= t2 GROUP BY time("10s") fill("previous")`. The averaged values of all four datasets are then present in the data returned by the database.

3. Why does increasing the number of fields not scale? Can you point me to the points in the documentation where this is discussed? I admit, that I am definitely no expert for Influx, so I may have overlooked something. But from https://docs.influxdata.com/influxdb/v1.8/troubleshooting/frequently-asked-questions/#why-does-series-cardinality-matter it seems to me, that the number of series and thus tags should be kept to a minimum. Currently a new tag is created for each dataset, so I would expect that a growing number of datasets poses much more of a problem if stored in tags instead of if they are stored in fields.

4. I do not understand what you mean by "Just group by dataset and expand the where clause". I checked the documentation for `GROUP BY` again (https://docs.influxdata.com/influxdb/v1.8/query_language/explore-data/#the-group-by-clause) and found this snippet for querying by more than one tag `SELECT MEAN("index") FROM "h2o_quality" `GROUP BY` "location","randtag"`. This works of course, but in this case all data with the same field key is averaged from the datasets specified by the tags. This is not what I want, I expect a seperate time series for each dataset. In the documentation of WHERE I did not find something that would address this issue.

5. Yes of course fields need to be type-homogeneous. But where is the problem? The only issue I see is, if one would change the data type of the dataset in Artiq and then there would be a conflict between old and new data. But is that not a good thing? If there is no easy conversion possible (e.g. from int to float), I would want to have some kind of warning, that I throw together data that does not fit together.

6. I agree with most of the last part, but I do not see, why one field per dataset does not scale. This is the option I would go for and as I mentioned above, I did not find a problem with it yet. But if you can provide some reference and/or discussion, I would be happy to learn!

jordens, 2021-12-22:
If you feel there is a better way give it a try and check how usable it is and measure how it scales. The influxdb controller code is just a couple lines. Note that this design has been in use for years showing very good scalability over large databases and few limitations. And I'm not aware of any problems with it so far.
I don't see the reason or any advantage in blowing up the number of fields. You don't actually say why you think the current design is problematic or why the alternative would be better or why it whether it would scale at all. You only say that you expect it. But that's not an argument.
Re 3 and 4 I would refer you to the influxdb documentation, their forum or doing the comparison and query design yourself. It's incorrect to say that when you group by a tag, the aggregation spans different groups. The contrary is true. Please try it.
The problem with type homogeneous fields and one field per dataset is that it imposes a constraint that ARTIQ does not have, does not enforce and does not need to have. It's pointless. Changing datatype is more common than you think. Also you would only notice very late that you have broken a constraint. This would be a terrible design. Try it.
Another downside of one field per dataset is that you have to be explicit about the field keys but you can for example regexp on the tag values.

jordens, 2025-03-05:
For anyone landing here: this pivoting presumably can be done with telegraf.