m-labs/artiq #1809: AD9910 get/set default behavior out of sync

b-bondurant, 2021-12-24:
# Bug Report

## One-Line Summary

As of #1584, AD9910 `get`/`get_mu` and `set`/`set_mu` behavior is not aligned.

## Issue Details

At a minimum, the default profile used for the `get` methods should be updated to reflect `set`. But there are also some significant changes to `set` in the presence of the (new) `ram_destination` parameter that should perhaps also be reflected in `get`.

For simplicity, and to avoid this problem in the future (at least for default arguments), it might make sense to have some module/class variables for default profiles, e.g.:
```
_DEFAULT_PROFILE = 7
_DEFAULT_PROFILE_RAM = 0
```

b-bondurant, 2021-12-24:
@lriesebos

sbourdeauducq, 2021-12-25:
@occheung

dnadlinger, 2021-12-30:
@pmldrmota

occheung, 2022-01-10:
#1818 made the default behavior consistent.

> But there are also some significant changes to set in the presence of the (new) ram_destination parameter that should perhaps also be reflected in get.
Please open another issue if you want this change.

closed 2022-01-10 by occheung