m-labs/artiq #1810: RPC tuple size calculation is wrong

pca006132, 2021-12-30:
# Bug Report

## One-Line Summary

Tuple size calculation for RPC is wrong and can lead to invalid memory access.

## Issue Details

### Steps to Reproduce
```python
from artiq.experiment import *

class TestListTuple(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @rpc
    def fine(self) -> TList(TTuple([TFloat, TFloat])):
        return [(123.456, 123.456), (123.456, 123.456)]

    @rpc
    def error(self) -> TList(TTuple([TBool, TFloat, TBool])):
        return [(True, 123.456, True), (True, 123.456, True)]

    @kernel
    def run(self):
        print(self.fine()[0]) # OK
        print(self.error()[0]) # Error
        return 0
```

The error case would not finish, as it is accessing an incorrect memory address.

Affects artiq-zynq and riscv. Not tested for or1k but I expect the same issue.

### Cause
https://github.com/m-labs/artiq/blob/bbac4770926048b3c0b434aede371a63d198ecd5/artiq/firmware/libproto_artiq/rpc_proto.rs#L398-L406

The size calculation does not take into account of padding and alignment requirements. The case I use is specifically tailored to trigger this bug, as the alignment of TBool should be 1 byte and the alignment of TFloat should be 4/8 bytes, there should be padding in between.

dnadlinger, 2021-12-30:
As a note aside, verifying the two sides of the RPC protocol implementation seems like a problem perfectly suited to fuzzing.

pca006132, 2021-12-31:
Alignment should also be wrong, as the tuple should have the same alignment as its largest element, and we do not calculate alignment for tuples in the current RPC code.

pca006132, 2022-01-04:
See https://git.m-labs.hk/M-Labs/artiq-zynq/pulls/160 for a fix on zynq.

PR #1841 (2022-01-28): Backport of "fixes alignment and size problem" from artiq-zynq
# ARTIQ Pull Request

## Description of Changes

Backporting of https://git.m-labs.hk/M-Labs/artiq-zynq/pulls/160 to master

### Related Issue
Closes #1810

## Type of Changes

|   | Type |
| ------------- | ------------- |
| ✓  | :bug: Bug fix  |

### Licensing

See [copyright & licensing for more info](https://github.com/m-labs/artiq/blob/master/CONTRIBUTING.rst#copyright-and-sign-off).
ARTIQ files that do not contain a license header are copyrighted by M-Labs Limited and are licensed under LGPLv3+.
closed 2022-01-28 by sbourdeauducq