m-labs/artiq #1811: NumberValue incorrect behaviour without default

hartytp, 2021-12-30:
# Bug Report

## Issue Details

Consider the following simple experiment

```python
import artiq.language as aq

class TestIntParam(aq.EnvExperiment):
    def build(self):
        self.setattr_argument('int_param', aq.NumberValue(type="int",
                                                          ndecimals=0,
                                                          scale=1,
                                                          step=1,
                                                          min=1)
                             )

    def run(self):
        print(self.int_param)

```

### Expected Behavior

1. `int_param` doesn't have a default specified so should default to 1
2. The GUI should show a value of 1
3. The experiment should print "1"

### Actual (undesired) Behavior

The GUI shows 1, but the experiment prints 0. After clicking on the spin button to increase `int_param` to 2 and then decrease it back to 1, the experiment correctly prints "1". Recomputing arguments then restores the original (incorrect) behaviour, with the GUI showing "1" but the experiment printing "0". 

This behaviour is not exhibited if "1" is explicitly given as a default value for `int_param`

```python
import artiq.language as aq

class TestIntParam(aq.EnvExperiment):
    def build(self):
        self.setattr_argument('int_param', aq.NumberValue(default=1,
                                                                                          type="int",
                                                                                          ndecimals=0,
                                                                                          scale=1,
                                                                                          step=1,
                                                                                          min=1)
                             )

    def run(self):
        print(self.int_param)

```

ciciwu213, 2022-01-18:
Hi, it seems that there was something wrong when I reproduced the issue. When I ran the code, the GUI showed a value of 0 and the experiment showed that ``DefaultMissing`` error was raised. Could you please provide some suggestions about reproducing it?

hartytp, 2022-01-18:
@ciciwu213 thanks for looking into this.

If you're unable to reproduce this then it sounds most likely that the issue is with the artiq setup I'm using. I'm not familiar enough with this part of the code to track this down quickly, but it may be related to one of the changes in https://github.com/OxfordIonTrapGroup/ndscan/issues/1

sbourdeauducq, 2022-01-18:
I can reproduce it. @ciciwu213 please follow the reproduction instructions carefully.

sbourdeauducq, 2022-01-18:
[image]

stevefan1999, 2022-01-27:
The default value of NumberValue is indeed not respecting the min bound (because it defaults to NoDefault if you do not specifiy it), so a rather simple fix is to passthrough min if default is none, and back to NoDefault if there is still no min.

stevefan1999, 2022-01-27:
https://github.com/m-labs/artiq/blob/bdaaf3c1d7fe362b9c741f4e8c13bca1e52d8874/artiq/gui/entries.py#L103

yup, this is why the default value is 0 is not specified, we can also tackle this by using min here

ciciwu213, 2022-01-27:
> The default value of NumberValue is indeed not respecting the min bound (because it defaults to NoDefault if you do not specify it, and the dashboard converts it to 0 internally), so a rather simple fix is to passthrough min if default is none, and back to NoDefault if there is still no min.
> 
> However, what if the min value is negative? Should we set this to the min, or 0?
> 
> @ciciwu213 you are probably using artiq_run and not dashboard before, right?

I used both artiq_run and master, and if artiq_run was used, ``DefaultMissing`` error was raised. When using artiq_master, it showed 0

PR #1839 (2022-01-27): prioritize min as part of default value resolution in dashboard
# ARTIQ Pull Request

## Description of Changes

This solves an issue where the dashboard value will appear to go out of bound when the default initial value on dashboard refresh is less than the minimum value of the argument (not rendered correctly). This could be solved with a forced re-render, but it is much better to handle it by setting the default value to min if min is positive, and 0 if min is negative

### Related Issue

Closes #1811 

## Type of Changes

|   | Type |
| ------------- | ------------- |
| ✓  | :bug: Bug fix [... cut]
closed 2022-01-27 by sbourdeauducq