m-labs/artiq #1812: artiq-beta package versions on conda are not up to date

fanmingyu212, 2022-01-01:
# Bug Report

## One-Line Summary

`artiq-beta` channel finds `7.7871.f0c50c80.beta` instead of the hydra build (`python3.8-artiq-7.7907.bbac4770.beta`).

## Issue Details

It seems that the `artiq-beta` package index https://conda.m-labs.hk/artiq-beta/index.html is not up to date. All the boards are not up to date either. The stable channel package index https://conda.m-labs.hk/artiq/index.html seems to be up to date.

### Expected Behavior

Installs latest artiq build (bbac4770) when running `conda update artiq`.

### Actual (undesired) Behavior

Installs f0c50c80.

### Your System (omit irrelevant parts)

* Operating System: Windows 10
* ARTIQ version: (with recent versions of ARTIQ, run ``artiq_client --version``)
* Version of the gateware and runtime loaded in the core device: (in the output of ``artiq_coremgmt -D .... log``)
* If using Conda, output of `conda list` (please submit as a file attachment, as this tends to be long)
* Hardware involved:

sbourdeauducq, 2022-01-01:
That's because the conda-artiq-board-kasli-unb build consistently fails due to FPGA timing problems (Cc @occheung).
At some point we should overhaul the bitstream build system which has scalability issues.
@fanmingyu212 in the meantime you can just install the old version and then overwrite packages with individual builds downloaded manually from Hydra.

fanmingyu212, 2022-01-01:
@sbourdeauducq thank you for the fast reply and helpful workaround!

fanmingyu212, 2022-01-12:
I think the original issue is solved (`artiq-beta` channel in conda is up to date now). This is probably solved by removing building of the unb variant starting from evaluation 3250 of Hydra though (2022/01/05), so the issue described in the title is unresolved.

sbourdeauducq, 2022-03-20:
Vivado update fixes this.

closed 2022-03-20 by sbourdeauducq