m-labs/artiq #1813: Implement Exception Allocation

pca006132, 2022-01-02:
# ARTIQ Feature Request

## Problem this request addresses

Exceptions can cause memory corruption now (#943, #1491) as they can contain pointers to stack memory and we only keep one copy of the exception object at a time, which cannot handle multiple in-flight exceptions.

## Describe the solution you'd like

Implement exception allocation and deallocation, similar to `__cxa_allocate_exception` and `__cxa_free_exception`. We may want to allocate them in the kernel CPU heap, or provide a special exception heap for exceptions (and terminate the kernel immediately when we cannot allocate more exception objects). We should also perform escape analysis to avoid users from using the exception object outside of the catch block, as we should free it when we exit the catch block.

## Additional context

We may also implement #1587 together with this.

stevefan1999, 2022-01-13:
[This](https://github.com/ApexAI/static_exception) could be a good source of inspiration for implementing it

sbourdeauducq, 2022-01-14:
Adding to ARTIQ-7 milestone as we want to keep runtime compatibility with NAC3.

closed 2022-01-25 by sbourdeauducq