m-labs/artiq #1816: Better Support for Custom EEMs

drewrisinger, 2022-01-06:
# ARTIQ Feature Request

## Problem this request addresses

There should be a simpler method for adding custom EEM modules, without requiring acceptance into ARTIQ mainline (e.g. plugins for EEMs). Sort of a follow-up to #1403, which was accepted but had a regression in ARTIQ 6 with the requirement of a JSON file to match the schema.

### Use-Case (actual case at Duke)

Group X creates a custom EEM module to interface with a custom HW device Y. Several internal groups want to use the same software, but Y hasn't been open-sourced, so it's not appropriate for ARTIQ. The JSON board description makes sense, b/c most of the hardware is standard Kasli EEMs, but for the Y EEM. Such an issue leads to PRs like #1800, which have the downside that they are slow to be integrated, require outside approval for something that's slightly experiment-specific, and typically require waiting for a new ARTIQ version before being able to be used.

## Describe the solution you'd like

The main issue is that there's not a clear path for adding out-of-tree EEM support, and it feels that adding out-of-tree EEM support is made extra difficult by the requirement that JSON files match the standard ARTIQ Kasli schema.

Proposed solutions (some/any combination of these would be great):
1. Allow disabling checking the JSON schema against file (maybe via environment variable)
2. Allow changing the JSON schema to a different schema (again, via environment variable/option flag to ``kasli_generic``?)
3. Document how/what needs changed to accommodate outside EEMs.

## Additional context

As a partial attempt to do the documentation suggested above:
* Add custom gateware module (optional, depends on use case)
* Add custom EEM class (sub-classing ``EEM``). Signature:
```python
class CustomEEM(artiq.gateware.eem.EEM):
    @staticmethod
    def io(*args, **kwargs, iostandard: migen.IOStandard = artiq.gateware.eem.default_iostandard) -> Sequence["Pad Assignments"]:
         """Define names for EEM IO pins to add to the available pins that the Kasli has available."""

    @classmethod
    def add_std(cls, target, *args, **kwargs) -> None:
        """Add the ARTIQ RTIO PHYs for your (custom) gateware to the target module."""
        cls.add_extension(target, *args, **kwargs) # calls CustomEEM.io(*args, **kwargs) internally
        # request IO pins that were added in CustomEEM.io()
        # add submodules & RTIO channels for each PHY that you're adding
        
```
* Add custom peripheral interpretation function to add the ``CustomEEM`` for a given set of EEM slots. Example
```python
def peripheral_custom(module, peripheral: Dict[str, Any]) -> None:
     # interpret peripheral arguments
     # add CustomEEM to ARTIQ SOC
     CustomEEM.add_std(module, *args, **kwargs)
```
* Add custom peripheral interpretation to kasli EEM peripheral module lookup dictionary: ``artiq.gateware.eem_7series.peripheral_processors["custom_eem"] = peripheral_custom``
* Write JSON schema for your custom peripheral.
* Modify JSON schema used by ``kasli_generic.py`` to validate the Kasli system description file. Example:
```python
import mergedeep
import artiq.coredevice.jsondesc as artiq_jsondesc

mergedeep.merge(artiq_jsondesc.schema, custom_eem_schema, strategy=mergedeep.Strategy.TYPESAFE_ADDITIVE)
```

All of the above can be seen (less well-organized) in https://github.com/drewrisinger/entangler-core/blob/634ed112649dd1e26b48d92700a07321231aa86c/entangler/kasli_generic.py if users would like an example.

sbourdeauducq, 2022-01-06:
What about adding a special EEM type to the JSON schema where you'd specify an external plugin (Python file/module)?

drewrisinger, 2022-01-06:
> What about adding a special EEM type to the JSON schema where you'd specify an external plugin (Python file/module)?

Yes, I mistakenly left any sort of plugin infrastructure out in my suggested solutions above. That sort of solution would be great. I assume the ``peripheral_processors[...] = peripheral_custom`` modification would still be needed in that case, so it's not an all-in-one solution? But otherwise sounds great. Edited suggested solutions above to add plugin suggestion

sbourdeauducq, 2022-01-07:
> That sort of solution would be great. I assume the `peripheral_processors[...] = peripheral_custom` modification would still be needed in that case,

I haven't looked into implementation details and how much needs to be changed/refactored. But the plugin solution should be doable.