m-labs/artiq #1826: Add support for DHCP

mbirtwell, 2022-01-12:
# ARTIQ Feature Request

I'm looking at adding DHCP support for core devices and I wanted to see if there were any thoughts on this upstream. 

## Problem this request addresses

We want to be able to centralise management of Kasli IP addresses.

## Describe the solution you'd like

I'm planning to upgrade smoltcp. Then make use of smoltcp's DHCPv4 client in the firmware runtime. I'm not planning to add DHCP support to the bootloader or DHCPv6 support. I think probably it's worth adding support for a config option, probably just on/off. Although we could always run DHCP if an IP address isn't present in the config.

stevefan1999, 2022-01-13:
@occheung label this as `area:smoltcp`, `type:feature` and `complexity:invasive`*

*: There are lots of breaking changes in [smoltcp 0.8.0](https://github.com/smoltcp-rs/smoltcp/blob/master/CHANGELOG.md#080---2021-12-11) with a MSRV of 1.56 when we are still using `1.51.0-nightly-2021-01-28`. For artiq-zynq the upgrade should be relatively less painful.

sbourdeauducq, 2022-01-13:
@stevefan1999 Did you actually try it or is this just an uninformed guess? I just tried compiling the firmware with 1.57 and most of it seems fine.

stevefan1999, 2022-01-13:
@sbourdeauducq Tried but need to take a closer look, changes like InterfaceBuilder changes are easy, but changes like https://github.com/smoltcp-rs/smoltcp/pull/557/commits/ef213fa7720d013dd57736540daa1adb55e67f91 and https://github.com/smoltcp-rs/smoltcp/pull/571/commits/4e365ce9ba4641dff5d88a46a3257eb2582c8bba seems complex.

mbirtwell, 2022-01-13:
There are a few breaking changes, but I think I have them all under control. I have everything building with smoltcp 0.8 and some basic testing looks good. I was going to test it more thoroughly and try adding the DHCP stuff before creating a PR, but I'm happy to show you what I have so far.

occheung, 2022-01-13:
#1085 (just for the record, not much stuff in there)
Apparently, there is already an open issue exactly for this. At least now there is a dedicated DHCP socket available in 0.8.

> There are a few breaking changes, but I think I have them all under control. I have everything building with smoltcp 0.8 and some basic testing looks good.

Good to hear that.

sbourdeauducq, 2022-01-13:
Right. Let's continue this discussion in #1085.

closed 2022-01-13 by sbourdeauducq