m-labs/artiq #1827: TypeError on running artiq_dashboard

cjbe, 2022-01-12:
# Bug Report

## One-Line Summary

The dashboard does not run on current master 3f812c4c2c1c322b28a68b3e985006631db872cd. 

## Issue Details

### Steps to Reproduce

Run `artiq_dashboard` - the process terminates with the error:
```
>python -m artiq.frontend.artiq_dashboard -s dgm
Traceback (most recent call last):
  File "c:\scratch\artiq\artiq\gui\models.py", line 413, in data
    return convert(key, self.backing_store[key],
  File "c:\scratch\artiq\artiq\dashboard\datasets.py", line 156, in convert
    return "Y" if v["persist"] else "N"
TypeError: tuple indices must be integers or slices, not str
```

From a bisection, this problem was introduced by d8b1e59538822b3377a30ba21306073f1367f3d7

cjbe, 2022-01-12:
@airwoodix I assume this is working for you, and that you are using the Artiq dashboard? Do you have some divergence in your environment from upstream Artiq?

airwoodix, 2022-01-12:
This looks like you have persistent datasets that are not stored in the new format. Maybe [this](https://github.com/m-labs/artiq/pull/1544#issuecomment-863123458) helps?

hartytp, 2022-01-12:
Indeed!

The release notes seem a bit misleading here

> The internal dataset representation was changed to support tracking HDF5 options like e.g. a compression method. This requires changes to code reading the dataset persistence file (dataset_db.pyon) and to custom applets.

To my mind, what this does not say is that e.g. the artiq_master is no longer compatible with artiq 6 `datset_db.pyon`s. It also doesn't give any guidance as to what a user has to do to their `dataset_db` to make it compatible.

airwoodix, 2022-01-13:
> The release notes seem a bit misleading here

Agreed, sorry for the inconvenience! Does #1828 look better to you?

closed 2022-01-18 by hartytp