m-labs/artiq #1836: poor tracebacks from re-raised exceptions

hartytp, 2022-01-20:
Consider the following simple experiment

```python
import artiq.experiment as aq
from artiq.experiment import kernel, RTIOUnderflow

class TestRaise(aq.EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl0")

    @kernel
    def underflow(self):
        at_mu(now_mu() - 1000000)
        self.ttl0.off()

    @kernel
    def run(self):
        try:
            self.underflow()
        except RTIOUnderflow:
            print("oh no!")
            raise
```

I would expect this to give a clear traceback pointing to the original `self.ttl0.off()`. Instead it gives a rather unhelpful error
```
root:Terminating with exception (RTIOUnderflow: RTIO underflow at 3178166213081343 mu, channel 65536, slack -171408241 mu)
Core Device Traceback (most recent call last):
  File ".../scratch/test_ignored_underflow_backtrace.py", line 0, in artiq_worker_test_ignored_underflow_backtrace.TestRaise.run(..., ...) (RA=+0x258)
    <unknown>
  File "ksupport/rtio.rs", line 69, column 14, in (Rust function)
    <unknown>
                 ^
artiq.coredevice.exceptions.RTIOUnderflow(0): RTIO underflow at 3178166213081343 mu, channel 65536, slack -171408241 mu
Traceback (most recent call last):
  File ".../artiq/master/worker_impl.py", line 376, in main
    exp_inst.run()
  File ".../artiq_tmp/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File ".../artiq/coredevice/core.py", line 139, in run
    self.comm.serve(embedding_map, symbolizer, demangler)
  File ".../artiq/coredevice/comm_kernel.py", line 645, in serve
    self._serve_exception(embedding_map, symbolizer, demangler)
  File ".../artiq/coredevice/comm_kernel.py", line 637, in _serve_exception
    raise python_exn
artiq.coredevice.exceptions.RTIOUnderflow: RTIO underflow at 3178166213081343 mu, channel 65536, slack -171408241 mu
```

If the exception is not caught, I get a helpful traceback.

If I replace the `self.ttl.off()` with a simple `raise RuntimeError` the situation is a bit better

```
root:Terminating with exception (RuntimeError: RuntimeError)
Core Device Traceback (most recent call last):
  File ".../scratch/test_ignored_underflow_backtrace.py", line 0, in artiq_worker_test_ignored_underflow_backtrace.TestRaise.run(..., ...) (RA=+0x278)
    <unknown>
  File ".../scratch/test_ignored_underflow_backtrace.py", line 14, column 9, in ... artiq_worker_test_ignored_underflow_backtrace.TestRaise.underflow<artiq_worker_test_ignored_underflow_backtrace.TestRaise>(...)
    raise RuntimeError
    ^
...
```

hartytp, 2022-01-20:
PS if I replace the `at_mu(now_mu() - 1000000)` with `at_mu(0)` I don't get an underflow at all. Is that expected behaviour?

pca006132, 2022-01-20:
I guess there are a few issues here:

1. It seems that re-raising overwrites the backtrace. I will look into this issue.
2. Our backtrace does not handle inlined function very well (I get `rtio_output(self.target_o, 1 if o else 0)` in the backtrace when I don't do re-raise, which is not very helpful), we could look into how to fix this so it can point to the `self.ttl0.off()`.

Thanks for opening this issue, I am fixing exception related bugs and working on nac3 exception code generation recently, so we can know what are the existing issues and how we can fix them.

> PS if I replace the `at_mu(now_mu() - 1000000)` with `at_mu(0)` I don't get an underflow at all. Is that expected behaviour?

I guess you need to call `self.core.break_realtime()`?

dnadlinger, 2022-01-20:
> I guess you need to call self.core.break_realtime()?

If I understand Tom correctly, passing `0`, which should only be further in the past, doesn't produce an underflow for him.

pca006132, 2022-01-20:
I guess this is because when the exception is caught, it is considered as handled, and re-raise would call `raise` which starts a new unwinding process with the re-raise as the starting point. Will fix in the exception branch together with nested exception and exception allocation support.

closed 2022-01-25 by sbourdeauducq