m-labs/artiq #1837: Errors when scanning the repo

hartytp, 2022-01-23:
@airwoodix I'm looking into merging the new dataset db changes into our artiq fork at the moment.

When I scan the repo I see a large number of errors related to `self.get_dataset("foo")` returning a dictionary. This is the first time I've looked at this part of artiq so I might be wrong, but I think roughly the following is happening...

You've changed `DatasetManager.get` to return `dataset["value"]`. I guess this is the code path taken when experiments run normally?

When the master inspects the code it looks like a different path is taken that goes via:
- [`ExamineDeviceMgr.get`](https://github.com/m-labs/artiq/blob/f58aa3bdf6e2bb53544688aa042c791e5d319ccc/artiq/master/worker_impl.py#L144)
- [`ParentDatasetDB.get`](https://github.com/m-labs/artiq/blob/f58aa3bdf6e2bb53544688aa042c791e5d319ccc/artiq/master/worker_impl.py#L72)
- [`make_parent_action`](https://github.com/m-labs/artiq/blob/f58aa3bdf6e2bb53544688aa042c791e5d319ccc/artiq/master/worker_impl.py#L49)
- IPC
- [`Worker._handle_worker_requests`](https://github.com/m-labs/artiq/blob/f58aa3bdf6e2bb53544688aa042c791e5d319ccc/artiq/master/worker.py#L227)
- [`dataset_db.get`](https://github.com/m-labs/artiq/blob/f58aa3bdf6e2bb53544688aa042c791e5d319ccc/artiq/frontend/artiq_master.py#L119)
- [`DatasetDB.get`](https://github.com/m-labs/artiq/blob/f58aa3bdf6e2bb53544688aa042c791e5d319ccc/artiq/master/databases.py#L84)
- ...which returns the raw data

hartytp, 2022-01-24:
I'm keen to unbreak master so we can continue testing the latest patches. How do we want to do that? 

Add a lambda in [frontend.master](https://github.com/m-labs/artiq/blob/f58aa3bdf6e2bb53544688aa042c791e5d319ccc/artiq/frontend/artiq_master.py#L119) to return the value from datasets rather than the whole structure?

Tell the [`ExamineDatasetManager`](https://github.com/m-labs/artiq/blob/f58aa3bdf6e2bb53544688aa042c791e5d319ccc/artiq/master/worker_impl.py#L151) to return the value?

Something else?

Taking a step back, is this definitely the right way of serving the underlying use-case? As I understand it (from reading the discussion in the [original PR](https://github.com/m-labs/artiq/pull/1544)) the motivation here is mainly around saving the occasional large dataset, like camera images. Questions:
- do we ever want these large datasets to persist (in the sense of being stored within the `dataset_db`)? If this is for data like camera images, presumably the answer is "no"
-  is it ever desirable for these large datasets to be exposed via the dashboard? Presumably no
- more generally, are there any cases where the metadata is actually useful in the persistent datasets, or is it just a biproduct of needing it for the large datasets (which don't persist)? Same for the dashboard
- do we want to broadcast these datasets (e.g. to a camera viewer applet)? If so, is `set_dataset(..., broadcast=True)` the right way of doing it (given that the `Notifier`/`Subscriber` interface is deliberately not efficient) or would it be better done e.g. using a simple controller with an efficient binary representation and maybe something like ZMQ to handle queues / slow VPN connections

Putting it another way, is the goal with this PR really just giving the user a way of getting large compressed datasets into the run HD5 file? If so, is this really the right way of achieving that?

Going back to @jordens comment on the original PR

> Alternatively, as you indicated, expose the hdf5/group handle to the user early in the experiment. That would enable #505 and give them full transparent access to all the hdf5 functionality (instead of having to wrap it through the dataset api).

That feels like more the right approach to me.

Anyway, I don't fully understand the use cases here, but I wonder if we should revert the PR merge to get upstream master back into a working state and discuss options a bit further? Otherwise, if we really feel that this PR is worth having, let's settle on a way of unbreaking things.

airwoodix, 2022-01-24:
> When I scan the repo I see a large number of errors related to self.get_dataset("foo") returning a dictionary.

Do you have a reproduction for this issue? I tried the following `repository/exp.py` on f58aa3bdf6e2bb53544688aa042c791e5d319ccc:

```python
from artiq.experiment import *
import numpy as np

class Exp1(EnvExperiment):
    def run(self):
        self.set_dataset(
            "data",
            np.arange(10),
            hdf5_options={"compression": "gzip"},
            persist=True,
        )

class Exp2(EnvExperiment):
    def build(self):
        self.setattr_dataset("data", np.zeros(10))

    def run(self):
        self.set_dataset("new_data", self.data * 2)

    def analyze(self):
        assert np.flatnonzero(self.data).size > 0
        assert (self.get_dataset("new_data") == self.data * 2).all()
```

Starting from an clean working directory (no `dataset_db.pyon`, no `last_rid.pyon`), running `artiq_master` and:

- `artiq_client scan-repository` (just for testing)
- `artiq_client submit repository/exp.py -c Exp1`
- `artiq_client submit repository/exp.py -c Exp2`

does not produce errors.

<details>
<summary>The HDF5 archives has the expected datasets/filters</summary>

```
$ h5dump -Hpd "datasets/data" results/2022-01-24/23/000000000-Exp1.h5
HDF5 "results/2022-01-24/23/000000000-Exp1.h5" {
DATASET "datasets/data" {
   DATATYPE  H5T_STD_I64LE
   DATASPACE  SIMPLE { ( 10 ) / ( 10 ) }
   STORAGE_LAYOUT {
      CHUNKED ( 10 )
      SIZE 34 (2.353:1 COMPRESSION)
   }
   FILTERS {
      COMPRESSION DEFLATE { LEVEL 4 }
   }
   FILLVALUE {
      FILL_TIME H5D_FILL_TIME_ALLOC
      VALUE  H5D_FILL_VALUE_DEFAULT
   }
   ALLOCATION_TIME {
      H5D_ALLOC_TIME_INCR
   }
}
}
```

```
$ h5dump -Hpd "datasets/new_data" results/2022-01-24/23/000000001-Exp2.h5
HDF5 "results/2022-01-24/23/000000001-Exp2.h5" {
DATASET "datasets/new_data" {
   DATATYPE  H5T_STD_I64LE
   DATASPACE  SIMPLE { ( 10 ) / ( 10 ) }
   STORAGE_LAYOUT {
      CONTIGUOUS
      SIZE 80
      OFFSET 2432
   }
   FILTERS {
      NONE
   }
   FILLVALUE {
      FILL_TIME H5D_FILL_TIME_IFSET
      VALUE  H5D_FILL_VALUE_DEFAULT
   }
   ALLOCATION_TIME {
      H5D_ALLOC_TIME_LATE
   }
}
}
```
</details>

How does this case differ from your use case? (as said on the M-Labs chat, I don't use this code myself in production and it's been a while since I last used the ARTIQ management system).

That said, I agree that squeezing this functionality (HDF5 dataset-level storage customization) into the dataset API is suboptimal. Nevertheless, it seems way too low-level to me to purposely expose the full `h5py`-group API directly (at least, it would require extra wrappers to not *have* to learn it). IIRC, early access to the hdf5 group also requires significant modifications to the code in [worker_impl.py](https://github.com/m-labs/artiq/blob/f58aa3bdf6e2bb53544688aa042c791e5d319ccc/artiq/master/worker_impl.py#L266-L322). The proposed implementation is intuitive and aims at being seamless to integrate with existing code bases (pending automatic porting of `dataset_db.pyon` e.g. in #1828).

I'm happy to discuss better implementations (maybe in #1545?). While they surely pose some challenges, I don't think that images should be treated differently than other datasets (how does one decide what is large enough to get a dedicated controller for synchronization and something else than `pyon` as serializer?).

closed 2022-01-25 by sbourdeauducq

hartytp, 2022-01-25:
@airwoodix I don't have a minimal reproduction I can post I'm afraid. From a quick skim, the main difference from what you haev there is that I was calling `self.get_dataset` inside `build`. If we want breaking changes like this, someone needs to take charge of testing in a production system and I'm afraid I'm out of time to look at this feature further though for now.

> The proposed implementation is intuitive and aims at being seamless to integrate with existing code bases (pending automatic porting of dataset_db.pyon e.g. in #1828).

I'm not sure the seamless integration is a good thing -- the use-case here is quite distinct from normal dataset usage, and the problems stem from trying to force different workflows into the same API. Repeating myself, but AFAICT:
- you don't want camera images persisted
- you don't want them in the dashboard
- you probably don't want them broadcast using the normal `Notifier` / `Subscriber` model
- all you want is an ergonomic way of getting custom data added to the HD5 file
This is not a good fit with the standard usage of datasets.

> (how does one decide what is large enough to get a dedicated controller for synchronization and something else than pyon as serializer?).

That's the kind of question which I think ARTIQ should take a position on; ARTIQ should provide the basic tools to support a variety of workflows and leave decisions like that to the user.

jordens, 2022-01-25:
Camera images (like flat field or dark field) absolutely need to persist. The should also be in the dashboard and available to applets and hence need to be broadcast. It's not distinct from normal dataset usage.

hartytp, 2022-01-25:
> The should also be in the dashboard and available to applets and hence need to be broadcast

Our preference has been to handle the broadcast of camera images to applets outside of the artiq datasets. Using a more efficient way of representing them and something to handle queues makes life much nicer (e.g. when using a slow VPN connection).

What's the motivation for having the camera images visible in the dashboard?

> Camera images (like flat field or dark field) absolutely need to persist.

Does every piece of calibration data need to live in the dataset db? It's not clear to me that this is obviously the right answer for large things like camera images compared.

jordens, 2022-01-25:
I can't comment on your choices and whether they are good or whether they work for others. If you want to run things over a slow VPN connection, then sure, some other data exchange mechanism might be required. ARTIQ datasets can be used for images and we have had several examples showcasing that since the early days, certainly with some caveats and limitations. The motivation for having images visible in the dashboard (through applets) should be clear. It's the same reason as for any other piece of data available to or generated by an experiment and hence the question is why one should not use ARTIQ datasets for it. Yes you can always use some other mechanism and nobody said that every piece of data needs to live in the dataset db.

hartytp, 2022-01-25:
>  The motivation for having images visible in the dashboard (through applets) should be clear

My language was imprecise here -- I was thinking specifically of the dataset explorer in the dashboard, not applets.

airwoodix, 2022-01-25:
> I don't have a minimal reproduction I can post I'm afraid. From a quick skim, the main difference from what you have there is that I was calling self.get_dataset inside build.

In `Exp2` above, `setattr_dataset()` [calls `get_dataset()`](https://github.com/m-labs/artiq/blob/f58aa3bdf6e2bb53544688aa042c791e5d319ccc/artiq/language/environment.py#L409-L412). Even explicitly, the following works on f58aa3b (of course triggers a `KeyError` upon `artiq_master` start if no `data` dataset is persisted, but that's also the case after #1838, and can be circumvented using the `default` argument):

```python
from artiq.experiment import *

class NewExp(EnvExperiment):
    def build(self):
        self.data = self.get_dataset("data")

    def run(self):
        pass

    def analyze(self):
        print(self.data)
```

I agree that there are issues linked to #1544 (manual migration, unit tests on Windows) but I don't yet see how it's obviously broken.

Conceptually, there's no difference between images and smaller datasets (they can be primary or secondary data, can be re-used raw or processed by further experiments, are useful to be seen live by the persons operating the experiment, may or not end up in the HDF5 archive). I would therefore rather improve the overall performance of the dataset system than special-casing images.

hartytp, 2022-01-25:
Can you print `self.data` in the build method and post the logs for artiq_master scanning the repo. Otherwise, can you point out which code path you expect to be taken when artiq examines the repo please? If it is the codepath I posted above then we would expect `get_dataset` to return a dictionary since the `ExamineDatasetMgr` accesses the dataset db `get` method directly

airwoodix, 2022-01-26:
When scanning the repository, `get_dataset()` called in `build()` indeed returns a dictionary:

```
$ artiq_master
ARTIQ master is now ready.
INFO:worker(scan,newexp.py):print:build: {'persist': True, 'value': array([0, 1, 2, 3, 4, 5, 6, 7, 8, 9]), 'hdf5_options': {'compression': 'gzip'}}
```

I believe that your analysis is correct. I was however not aware of this changing behavior and got confused because I only tested dataset manipulations in `prepare()`, `run()`, and `analyze()` where the value is returned correctly. Thanks and sorry for the extra round.

The mistake is indeed in `DatasetDB.get()` that was not migrated consistently with the code prior to #1544 ([diff](https://github.com/m-labs/artiq/pull/1544/files#diff-49d6148de5bccfe9df890904b5a554b635765110579bda05c24506f2b927f06cL62)). `self.data.raw_view[key][1]` should have read `self.data.raw_view[key]["value"]` and not `self.data.raw_view[key]`.

Looking at this, there's essentially no difference between storing the dataset as a dictionary vs. a tuple in the notifier and returning the value only (and not the persistence metadata). Already with only the two fields `persist` and `value`, the dictionary is much more readable than the tuple.

As already said, I'm not against alternative implementations of the HDF5 storage customization but the short-notice reversion of #1544 is now practically a regression and it doesn't realistically look like the feature's going to be back soon (diverging interests, lack of time, etc.).

PR #1838 (2022-01-25): Revert "Merge pull request #1544 from airwoodix/dataset-compression"
closes #1837 

DRAFT (needs check before merge)