m-labs/artiq #1857: support PCA9547 (Kasli-SoC assembly option) in i2c.py

sbourdeauducq, 2022-02-23:
Suggestion: expose the I2C switch functions from the runtime to the kernels, and use them in i2c.py instead of reimplementing them as is currently done. The artiq-zynq runtime already supports PCA9547/PCA9548 autodetection.
Needs support in DRTIO aux as well.

Spaqin, 2022-02-24:
Should it encompass only PCA954X series of i2c switches, or ``TCA6424A`` too, replacing all the i2c switch classes with just one generic one? Autodetect would be probably limited to PCAs on Kasli and Kasli-SoC only as well, since only they have a i2c reset pin - so should the type be overrideable?

sbourdeauducq, 2022-02-24:
TCA6424A is not a I2C switch and not relevant here.

PR #1860 (2022-03-01): [WIP] PCA9547 support

# ARTIQ Pull Request

## Description of Changes

As #1857 mentions - Kasli SoC may use PCA9547 I2C switch. Suggested solution was to expose the select function (which is already done in firmware) to i2c.py, with a more generic API that would be suited to both available PCA9547 and PCA9548. This PR aims to do that. That includes also exposing the function with DRTIO.

Most importantly, this should fix Kasli-SoC behavior with cards that communicate over I2C.

Not well tested yet - Kasli firmware compiles, but haven't tested on a crate yet - thus marked as WIP - to gather i [... cut]
closed 2022-03-01 by sbourdeauducq