m-labs/artiq #1858: DAC widget for non-Zotino AD53xx

jniedermeyer, 2022-02-25:
# Question
Is the DAC monitor compatible with DACs that use the general AD53xx driver? 

## Category: moninj

## Description

We use AD5370 DACs in our experiment. In our recent upgrade from ARTIQ 3.7 to the latest stable release of ARTIQ 6, we tried to get the new DAC widget working in the dashboard. 

We found one typo in line 229 of moninj.py that tried to scan the device_db for devices with a class of "AD53XX", when the actual class name is "AD53xx" (I can submit a PR for this if desired). Fixing this allowed moninj to find our DAC, but it doesn't seem like the monitor is working. Is the DAC widget compatible with non-Zotino AD53xx DACs?

sbourdeauducq, 2022-02-26:
> We found one typo in line 229 of moninj.py that tried to scan the device_db for devices with a class of "AD53XX", when the actual class name is "AD53xx" (I can submit a PR for this if desired). 

Thanks, fixed.

> Fixing this allowed moninj to find our DAC, but it doesn't seem like the monitor is working. Is the DAC widget compatible with non-Zotino AD53xx DACs?

I guess not then.

closed 2022-02-26 by sbourdeauducq

jniedermeyer, 2022-02-28:
Thanks for fixing that. Since we are using the KC705 NIST_QC2 variant and not an EEM, it doesn't look like there is an easy way to implement the ad53xx_monitor module in the gateware. For now, we'll just rely on our external monitors.