m-labs/artiq #1867: Would you be willing to accept a PR that will add flake support for branch release-6

doronbehar, 2022-03-08:
# ARTIQ Feature Request

## Problem this request addresses

I'd like to use Artiq version 6.0 in a flake based environment and I noticed there's no `flake.nix` in that branch.

## Describe the solution you'd like

Perhaps copying most of the `flake.nix` at branch master right to branch `release-6`? Is there anything else that will need consideration?

## Additional context

I also noticed there are flake `inputs` that are not flakes, and that there are many derivations defined in the flake that are built with `src`s that are not locked in the `flake.lock`. Would you be willing to also accept a PR that will do something like this:

```patch
diff --git i/flake.nix w/flake.nix
index a43321d0..f1981c4e 100644
--- i/flake.nix
+++ w/flake.nix
@@ -9,7 +9,12 @@
   inputs.src-migen = { url = github:m-labs/migen; flake = false; };
   inputs.src-misoc = { type = "git"; url = "https://github.com/m-labs/misoc.git"; submodules = true; flake = false; };
 
-  outputs = { self, nixpkgs, mozilla-overlay, src-sipyco, src-pythonparser, src-migen, src-misoc }:
+  inputs.src-qasync = {
+    url = "github:CabbageDevelopment/qasync?ref=v0.10.0";
+    flake = false;
+  };
+
+  outputs = { self, nixpkgs, mozilla-overlay, src-sipyco, src-pythonparser, src-migen, src-misoc, src-qasync }:
     let
       pkgs = import nixpkgs { system = "x86_64-linux"; overlays = [ (import mozilla-overlay) ]; };
       rustManifest = pkgs.fetchurl {
@@ -59,21 +64,25 @@
         propagatedBuildInputs = with pkgs.python3Packages; [ regex ];
       };
 
-      qasync = pkgs.python3Packages.buildPythonPackage rec {
+      qasync =(let
+        lock = builtins.fromJSON (builtins.readFile ./flake.lock);
+      in pkgs.python3Packages.buildPythonPackage rec {
         pname = "qasync";
-        version = "0.10.0";
-        src = pkgs.fetchFromGitHub {
-          owner = "CabbageDevelopment";
-          repo = "qasync";
+        version = pkgs.lib.removePrefix "v" lock.nodes.src-qasync.original.ref;
+        src =  pkgs.fetchFromGitHub {
+          inherit (lock.nodes.src-qasync.original)
+            owner
+            repo
+          ;
           rev = "v${version}";
-          sha256 = "1zga8s6dr7gk6awmxkh4pf25gbg8n6dv1j4b0by7y0fhi949qakq";
+          sha256 = lock.nodes.src-qasync.locked.narHash;
         };
         propagatedBuildInputs = [ pkgs.python3Packages.pyqt5 ];
         checkInputs = [ pkgs.python3Packages.pytest ];
         checkPhase = ''
           pytest -k 'test_qthreadexec.py' # the others cause the test execution to be aborted, I think because of asyncio
         '';
-      };
+      });
 
       outputcheck = pkgs.python3Packages.buildPythonApplication rec {
         pname = "outputcheck";

```

Imagine the same change would be done for all other packages have an `src = fetchFromGitHub {...}`.

sbourdeauducq, 2022-03-08:
Generally we do not add features to release branches, only fix bugs. But you are welcome to provide your own flake.nix for ARTIQ-6 somewhere else.

> Is there anything else that will need consideration?

The Rust compiler, LLVM and llvmlite need to be changed, since ARTIQ-6 uses OR1k and a modified toolchain, whereas 7 uses upstream and RISC-V.

As for reading the version from the flake inputs, ``qasync.sourceInfo`` should be the way to go - I'm not sure if reading the lockfile is a good idea.

doronbehar, 2022-03-08:
> Generally we do not add features to release branches, only fix bugs. But you are welcome to provide your own flake.nix for ARTIQ-6 somewhere else.

Perhaps you would consider adding this flake never the less? It's not a new feature per say IMHO, it's just making it a bit easier to use this package with tools (nix flakes) that were not present at the time it was developed. I was thinking about maintaining my own fork for this task, but only as a last resort.

> As for reading the version from the flake inputs, `qasync.sourceInfo` should be the way to go - I'm not sure if reading the lockfile is a good idea.

Similar nix code is written in https://github.com/edolstra/flake-compat/blob/master/default.nix so I think it's not such a bad idea it it came from the main developer of Nix. What exactly do you mean by `qasync.sourceInfo`? The `src-qasync` doesn't have such an attribute as `sourceInfo`.

sbourdeauducq, 2022-03-08:
You don't need a fork, you can simply put the flake in a repository on its own and reference the ARTIQ repos as a source.

BTW I suspect you underestimate the difficulty of getting or1k rust to work, maybe this flake isn't really worth the effort. Why not use 7 if you want flakes?

Seems for non-flakes input the information is at the top level:
```
nix-repl> :lf /home/sb/artiq
warning: Git tree '/home/sb/artiq' is dirty
Added 13 variables.

nix-repl> sourceInfo
{ lastModified = 1646562782; lastModifiedDate = "20220306103302"; narHash = "sha256-a3lOmL0WF2VZx1H8nr1Rh9qTzXZdhtZ629uUOSFmBQo="; outPath = "/nix/store/5yg774kad9y46pdl9yw62kp2irirglnb-source"; submodules = false; }

nix-repl> inputs.src-migen.rev
"ac703010eaa06ac9b6e32f97c6fa98b15de22b31"

nix-repl> inputs.src-migen.narHash
"sha256-qpVj/yJf4hDDc99XXpVPH4EbLC8aCmEtACn5qNc3DGI="
```

In any case why not just ``src = src-qasync`` instead of duplicating ``fetchFromGitHub``?
You could also perhaps define ``qasync-version = "0.10.0"`` and use it in the source input and ``buildPythonPackage``, and avoid reading the lockfile and other complications. Maybe it can be a passthrough attribute into ``src-qasync``.

doronbehar, 2022-03-08:
> In any case why not just `src = src-qasync` instead of duplicating `fetchFromGitHub`?
> You could also perhaps define `qasync-version = "0.10.0"` and use it in the source input and [...]

Interesting ideas indeed, thanks!

> BTW I suspect you underestimate the difficulty of getting or1k rust to work, maybe this flake isn't really worth the effort. Why not use 7 if you want flakes?

I'm currently not near my artiq device, but I had trouble interacting with it when I tried using version 7 from the flake last time I checked. Plus, at our team we tend to appreciate the stability of version 6 vs 7. Perhaps it'd be possible to use version 6 with the nix scripts at https://git.m-labs.hk/M-Labs/nix-scripts ? The channels setup explained [here](https://m-labs.hk/artiq/manual/installing.html#installing-via-nix-linux) is very imperative.

sbourdeauducq, 2022-03-08:
> I'm currently not near my artiq device, but I had trouble interacting with it when I tried using version 7 from the flake last time I checked.

Did you reflash it? ARTIQ-6 firmware cannot talk to ARTIQ-7 software, as I mentioned the CPU is different.

doronbehar, 2022-03-08:
> Did you reflash it? ARTIQ-6 firmware cannot talk to ARTIQ-7 software, as I mentioned the CPU is different.

Yes, that sounds familiar, we had a warning regarding firmware version mismatched, along with Python errors filling our screen. I see now there are instructions to flash the firmware [here](https://m-labs.hk/artiq/manual/installing.html#flashing-gateware-and-firmware-into-the-core-device). I will look into it more once I reach the hardware, thanks a lot!

Lastly, as for the idea of using non flake inputs instead of raw `fetchFromGitHub`, I'm not sure I understood you thoroughly. Indeed it's much nicer to use the flake inputs as are for an `src`, but the `sourceInfo` attribute seems to be part of the artiq flake, and you imply that I can get the git `ref` or the git revision of the inputs with this attribute?

sbourdeauducq, 2022-03-08:
Yes, try just ``src-qasync.rev`` for the git revision. Seems ``ref`` is not available.

sbourdeauducq, 2022-03-08:
Using the ``sourceInfo``-like attributes is fine, reading the lockfile directly I'm not sure.

sbourdeauducq, 2022-03-10:
> Perhaps it'd be possible to use version 6 with the nix scripts at https://git.m-labs.hk/M-Labs/nix-scripts ?

Yes - just use the ``artiq-fast`` and select the ARTIQ sources with ``-I artiqSrc=...``.
I don't think there is any clear action to take in ARTIQ, so closing this Issue.

closed 2022-03-10 by sbourdeauducq