m-labs/artiq #1878: Cannot find 'dataset' tab on the artiq5 dashboard

sahrak, 2022-03-30:
# Bug Report

## One-Line Summary

I cannot find the dataset tab on the dashboard. Restarted artiq multiple times, but dataset tab not appearing.

### Expected Behavior

Dataset should be next to the shotcuts/applets/Explorer tabs?

### Your System (omit irrelevant parts)

* Operating System:
* ARTIQ version: ARTIQ v5.0+202.g4ebd956e.dirty
* Version of the gateware and runtime loaded in the core device: 
* PS C:\Users\LABadmin\Documents\artiq-master> artiq_coremgmt -D ... log
Traceback (most recent call last):
  File "C:\ProgramData\Anaconda3\envs\artiq\lib\encodings\idna.py", line 165, in encode
    raise UnicodeError("label empty or too long")
UnicodeError: label empty or too long

* If using Conda, output of `conda list` (please submit as a file attachment, as this tends to be long)
* [conda list output.txt](https://github.com/m-labs/artiq/files/8379748/conda.list.output.txt)
* Hardware involved: software issue

sbourdeauducq, 2022-03-30:
Most likely Qt and/or Windows bug, try deleting the GUI state file. https://forum.m-labs.hk/d/90-open-artiq/9

sbourdeauducq, 2022-04-07:
Did this resolve the problem?

closed 2022-04-20 by sbourdeauducq