m-labs/artiq #1887: Dashboard crashes; passing floats where PyQt expects ints

ljstephenson, 2022-05-06:
# Bug Report

## One-Line Summary

Several PyQt functions used within the artiq dashboard are being passed float arguments where they are expecting integers, causing the dashboard to crash.

## Issue Details

This appears to affect many of the Qt functions. Had a look at some old working versions; the relevant code on the artiq side hasn't changed in a long time, so I thought it might be a new PyQt version, but it's the same as in some old environments I have that appear to work. I think I'm missing something obvious but at the moment I'm beginning to work my way down the chain of errors just casting directly to integer where relevant.

### Steps to Reproduce

1. Start `artiq_master`
2. Start `artiq_dashboard`

### Expected Behavior

Dashboard starts.

### Actual (undesired) Behavior

Dashboard crashes before starting, with following error:

```
(artiq-test)>artiq_dashboard
Traceback (most recent call last):
  File "C:\Users\<user>\Anaconda3\envs\artiq-test\Scripts\artiq_dashboard-script.py", line 9, in <module>
    sys.exit(main())
  File "C:\Users\<user>\Anaconda3\envs\artiq-test\lib\site-packages\artiq\frontend\artiq_dashboard.py", line 170, in main
    d_explorer = explorer.ExplorerDock(expmgr, d_shortcuts,
  File "C:\Users\<user>\Anaconda3\envs\artiq-test\lib\site-packages\artiq\dashboard\explorer.py", line 268, in __init__
    self.waiting_panel = WaitingPanel()
  File "C:\Users\<user>\Anaconda3\envs\artiq-test\lib\site-packages\artiq\dashboard\explorer.py", line 148, in __init__
    self.waiting_spinner = QtWaitingSpinner()
  File "C:\Users\<user>\Anaconda3\envs\artiq-test\lib\site-packages\artiq\gui\waitingspinnerwidget.py", line 54, in __init__
    self.updateTimer()
  File "C:\Users\<user>\Anaconda3\envs\artiq-test\lib\site-packages\artiq\gui\waitingspinnerwidget.py", line 163, in updateTimer
    self._timer.setInterval(1000 / (self._numberOfLines * self._revolutionsPerSecond))
TypeError: setInterval(self, int): argument 1 has unexpected type 'float'
```

### Your System (omit irrelevant parts)

* Operating System: W10, conda
* ARTIQ version: (with recent versions of ARTIQ, run ``artiq_client --version``) `ARTIQ v7.8044.c440f9f.beta`
* Version of the gateware and runtime loaded in the core device: (in the output of ``artiq_coremgmt -D .... log``): (not relevant)
* If using Conda, output of `conda list` (please submit as a file attachment, as this tends to be long) [conda-list.txt](https://github.com/m-labs/artiq/files/8643872/conda-list.txt)
* Hardware involved: N/A

ljstephenson, 2022-05-06:
This appears to python=3.10 related; same issue doesn't appear (at least, not immediately, didn't press to see if it would fail opening applets/experiments) in an environment with python=3.9
[conda-list-39.txt](https://github.com/m-labs/artiq/files/8644026/conda-list-39.txt)

sbourdeauducq, 2022-05-28:
This can be reproduced on Nix with:
```
iff --git a/flake.nix b/flake.nix
index 3449f54b..8da27229 100644
--- a/flake.nix
+++ b/flake.nix
@@ -1,7 +1,7 @@
 {
   description = "A leading-edge control system for quantum information experiments";
 
-  inputs.nixpkgs.url = github:NixOS/nixpkgs/nixos-21.11;
+  inputs.nixpkgs.url = git+file:///home/sb/nixpkgs;
   inputs.mozilla-overlay = { url = github:mozilla/nixpkgs-mozilla; flake = false; };
   inputs.sipyco.url = github:m-labs/sipyco;
   inputs.sipyco.inputs.nixpkgs.follows = "nixpkgs";
@@ -331,7 +331,7 @@
             sha256 = "sha256-YgUsl4/FohfsOncM4uiz/3c6g2ZN4oZ0y5vV/2Skwqg=";
             fetchSubmodules = true;
           };
-          patches = oa.patches or [] ++ [
+          patches = [
             (pkgs.fetchurl {
               url = "https://git.m-labs.hk/M-Labs/nix-scripts/raw/commit/575ef05cd554c239e4cc8cb97ae4611db458a80d/artiq-fast/pkgs/openocd-jtagspi.diff";
               sha256 = "0g3crk8gby42gm661yxdcgapdi8sp050l5pb2d0yjfic7ns9cw81";
@@ -435,8 +435,6 @@
           packages.x86_64-linux.vivadoEnv
           packages.x86_64-linux.vivado
           packages.x86_64-linux.openocd-bscanspi
-          pkgs.python3Packages.sphinx pkgs.python3Packages.sphinx_rtd_theme
-          pkgs.python3Packages.sphinx-argparse sphinxcontrib-wavedrom latex-artiq-manual
         ];
         shellHook = ''
           export LIBARTIQ_SUPPORT=`libartiq-support`
```

nixpkgs:
```
diff --git a/pkgs/top-level/all-packages.nix b/pkgs/top-level/all-packages.nix
index 5668f65fab1..8efb752b589 100644
--- a/pkgs/top-level/all-packages.nix
+++ b/pkgs/top-level/all-packages.nix
@@ -14392,11 +14392,11 @@ with pkgs;
   # available as `pythonPackages.tkinter` and can be used as any other Python package.
   # When switching these sets, please update docs at ../../doc/languages-frameworks/python.md
   python2 = python27;
-  python3 = python39;
+  python3 = python310;
 
   # pythonPackages further below, but assigned here because they need to be in sync
   python2Packages = dontRecurseIntoAttrs python27Packages;
-  python3Packages = dontRecurseIntoAttrs python39Packages;
+  python3Packages = dontRecurseIntoAttrs python310Packages;
 
   pypy = pypy2;
   pypy2 = pypy27;

```

Tested with nixpkgs 1127044143d3813cfa7f27ffd6f0c9c14b5c5d61. Builds for that version are also currently cached on nixbld.m-labs.hk, the widespread Python dependency causes a lot of rebuilds.

commit 667f36a2 (2022-05-29): gui: fix Python 3.10 PyQt float/int issues. Closes #1887
closed 2022-05-29 by sbourdeauducq