m-labs/artiq #1888: Scheduler test failing

ljstephenson, 2022-05-06:
# Bug Report

## One-Line Summary

`python -m unittest -v artiq.test.test_scheduler.SchedulerCase.test_pending_priority` fails.

## Issue Details

Disabling the assertion and printing out the order in which events come in shows that they are only in a slightly different order; I'm not sure if behaviour has changed recently for this to be expected. The order I see is:

```
{'action': 'setitem', 'path': [], 'key': 0, 'value': {'pipeline': 'main', 'expid': {'log_level': 50, 'file': '\\\\jake.nist.gov\\688\\users\\ljs3\\scratch\\artiq\\artiq\\test\\test_scheduler.py', 'class_name': 'CheckPauseBackgroundExperiment', 'arguments': {}}, 'priority': 1, 'due_date': None, 'flush': False, 'status': 'pending', 'repo_msg': None}}
{'action': 'setitem', 'path': [], 'key': 1, 'value': {'pipeline': 'main', 'expid': {'log_level': 30, 'file': '\\\\jake.nist.gov\\688\\users\\ljs3\\scratch\\artiq\\artiq\\test\\test_scheduler.py', 'class_name': 'EmptyExperiment', 'arguments': {}}, 'priority': 3, 'due_date': 1651976855.5527163, 'flush': False, 'status': 'pending', 'repo_msg': None}}
{'action': 'setitem', 'path': [], 'key': 2, 'value': {'pipeline': 'main', 'expid': {'log_level': 30, 'file': '\\\\jake.nist.gov\\688\\users\\ljs3\\scratch\\artiq\\artiq\\test\\test_scheduler.py', 'class_name': 'EmptyExperiment', 'arguments': {}}, 'priority': 2, 'due_date': 1651876856.5527163, 'flush': False, 'status': 'pending', 'repo_msg': None}}
{'action': 'setitem', 'path': [0], 'key': 'status', 'value': 'preparing'}
{'action': 'setitem', 'path': [0], 'key': 'status', 'value': 'prepare_done'}
{'action': 'setitem', 'path': [2], 'key': 'status', 'value': 'preparing'}
{'action': 'setitem', 'path': [0], 'key': 'status', 'value': 'running'}
{'action': 'setitem', 'path': [2], 'key': 'status', 'value': 'prepare_done'}
{'action': 'setitem', 'path': [0], 'key': 'status', 'value': 'paused'}
{'action': 'setitem', 'path': [2], 'key': 'status', 'value': 'running'}
{'action': 'setitem', 'path': [2], 'key': 'status', 'value': 'run_done'}
{'action': 'setitem', 'path': [0], 'key': 'status', 'value': 'running'}
{'action': 'setitem', 'path': [2], 'key': 'status', 'value': 'analyzing'}
{'action': 'setitem', 'path': [2], 'key': 'status', 'value': 'deleting'}
{'action': 'delitem', 'path': [], 'key': 2}
```

These two items are reversed compared to the output expected by the test case:

```
{'action': 'setitem', 'path': [2], 'key': 'status', 'value': 'preparing'}
{'action': 'setitem', 'path': [0], 'key': 'status', 'value': 'running'}
```

### Steps to Reproduce

1. Run `python -m unittest -v artiq.test.test_scheduler.SchedulerCase.test_pending_priority`

### Expected Behavior

Test passes.

### Actual (undesired) Behavior

Test fails.

```
>python -m unittest -v artiq.test.test_scheduler.SchedulerCase.test_pending_priority
test_pending_priority (artiq.test.test_scheduler.SchedulerCase)
Check due dates take precedence over priorities when waiting to ... C:\Users\<user>\Anaconda3\envs\artiq-test\lib\site-packages\sipyco\asyncio_tools.py:23: DeprecationWarning: There is no current event loop
  self.task = asyncio.ensure_future(log_exceptions(self._do))
C:\Users\<user>\Anaconda3\envs\artiq-test\lib\site-packages\sipyco\asyncio_tools.py:101: DeprecationWarning: There is no current event loop
  self._loop = asyncio.get_event_loop()
Unhandled exception in TaskObject task body
Traceback (most recent call last):
  File "C:\Users\<user>\Anaconda3\envs\artiq-test\lib\site-packages\sipyco\asyncio_tools.py", line 16, in log_exceptions
    return await awaitable()
  File "C:\Users\<user>\Anaconda3\envs\artiq-test\lib\site-packages\artiq\master\scheduler.py", line 215, in _do
    run.status = RunStatus.preparing
  File "C:\Users\<user>\Anaconda3\envs\artiq-test\lib\site-packages\artiq\master\scheduler.py", line 87, in status
    self._notifier[self.rid]["status"] = self._status.name
  File "C:\Users\<user>\Anaconda3\envs\artiq-test\lib\site-packages\sipyco\sync_struct.py", line 242, in __setitem__
    self.root.publish({"action": ModAction.setitem.value,
  File "C:\Users\<user>\Anaconda3\envs\artiq-test\lib\site-packages\artiq\test\test_scheduler.py", line 266, in notify
    self.assertEqual(mod, expect[expect_idx])
  File "C:\Users\<user>\Anaconda3\envs\artiq-test\lib\unittest\case.py", line 845, in assertEqual
    assertion_func(first, second, msg=msg)
  File "C:\Users\<user>\Anaconda3\envs\artiq-test\lib\unittest\case.py", line 1144, in assertDictEqual
    self.fail(self._formatMessage(msg, standardMsg))
  File "C:\Users\<user>\Anaconda3\envs\artiq-test\lib\unittest\case.py", line 675, in fail
    raise self.failureException(msg)
AssertionError: {'action': 'setitem', 'path': [2], 'key': 'status', 'value': 'preparing'} != {'path': [0], 'action': 'setitem', 'value': 'running', 'key': 'status'}
- {'action': 'setitem', 'key': 'status', 'path': [2], 'value': 'preparing'}
?                                                 ^             - ^^^^

+ {'action': 'setitem', 'key': 'status', 'path': [0], 'value': 'running'}
?
```

### Your System (omit irrelevant parts)

* Operating System: W10, conda
* ARTIQ version: (with recent versions of ARTIQ, run ``artiq_client --version``): `ARTIQ v7.8044.c440f9f.beta`
* Version of the gateware and runtime loaded in the core device: (in the output of ``artiq_coremgmt -D .... log``) (not relevant)
* If using Conda, output of `conda list` (please submit as a file attachment, as this tends to be long) 
[conda-list.txt](https://github.com/m-labs/artiq/files/8643917/conda-list.txt)
* Hardware involved: N/A

sbourdeauducq, 2022-05-07:
Probably Python 3.10 related, like your other issue. We didn't test anything with 3.10.

ljstephenson, 2022-05-09:
Just checked - same failure in environments with 3.8 and 3.9
[conda-list-39.txt](https://github.com/m-labs/artiq/files/8655026/conda-list-39.txt)
[conda-list-38.txt](https://github.com/m-labs/artiq/files/8655042/conda-list-38.txt)

sbourdeauducq, 2022-05-09:
Strange - it's passing on the server e.g. https://nixbld.m-labs.hk/build/120442/log
This code has not changed recently.

dnadlinger, 2022-05-10:
Could this just be a timing/event ordering issue? I haven't had a look at this particular test, but some of them do rely on implicitly defining order via `sleep`s.

ljstephenson, 2022-05-12:
> Could this just be a timing/event ordering issue?

Certainly seems to be, but I don't know how that order could be defined more rigidly. The "issue" seems to be that experiment 2 is taking longer to prepare on my machine than the test server, so experiment 0 is reaching the run stage before 2 is done preparing; but the prepare stage is empty so I don't see how it can be made faster.

> Strange

Agreed. I repro'd this on a second W10 machine with python=3.9, package list generated was identical except for `zipp`. I don't think this is high priority but I do think it's bad juju for a test to pass on some machines and not others...

dnadlinger, 2022-05-12:
> Certainly seems to be, but I don't know how that order could be defined more rigidly.

I haven't looked into what this particular test would require, but for testing `Scheduler.idle` (which we haven't gotten to properly redesign and upstream yet, I had introduced some asyncio events to ensure the requisite ordering:

```py
    def _make_status_events(self, rid, previous_notify=None):
        status_events = {
            key: asyncio.Event()
            for key in ("prepare_done", "running", "paused", "run_done",
                        "deleting", "idle")
        }

        def notify(mod):
            if (mod["path"] == [rid] and mod["action"] == "setitem"
                    and mod["key"] == "status"):
                event = status_events.get(mod["value"], None)
                if event:
                    event.set()
            if previous_notify:
                previous_notify(mod)

        return status_events, notify

    def _make_scheduler_for_idle_test(self):
        class Flags:
            def __init__(self):
                self.be_idle = False
                self.take_long_in_idle = False
                self.termination_ok = False

        flags = Flags()

        def get_dataset(name):
            return getattr(flags, name)

        def check_termination(mod):
            self.assertEqual(
                mod, {
                    "action": "setitem",
                    "key": "termination_ok",
                    "value": (False, True),
                    "path": []
                })
            flags.termination_ok = True

        handlers = {
            "get_dataset": get_dataset,
            "update_dataset": check_termination
        }
        scheduler = Scheduler(_RIDCounter(0), handlers, None, None)

        status_events, notify = self._make_status_events(0)
        scheduler.notifier.publish = notify

        scheduler.start()

        expid_idle = _get_expid("IdleExperiment")
        # Suppress the SystemExit backtrace when worker process is killed.
        expid_idle["log_level"] = logging.CRITICAL
        scheduler.submit("main", expid_idle, 0, None, False)

        return scheduler, status_events, flags

    def test_close_with_idle_running(self):
        """Check scheduler exits with experiment still idle"""
        loop = self.loop
        scheduler, status_events, flags = self._make_scheduler_for_idle_test()
        loop.run_until_complete(status_events["running"].wait())
        flags.be_idle = True
        loop.run_until_complete(status_events["idle"].wait())
[... 85 lines cut ...]
        """Check that blocking is_idle_callback times out"""
        loop = self.loop
        scheduler, status_events, flags = self._make_scheduler_for_idle_test()

        # Submit empty experiment with lower priority.
        empty_status_events, notify = self._make_status_events(
            1, scheduler.notifier.publish)
        scheduler.notifier.publish = notify

        expid = _get_expid("EmptyExperiment")
        scheduler.submit("main", expid, -1, None, False)

        # Make the is_idle_callback in block; silencing the expected error
        # message in the log output.
        logging.disable(logging.ERROR)
        flags.be_idle = True
        flags.take_long_in_idle = True
        loop.run_until_complete(status_events["deleting"].wait())
        logging.disable(logging.NOTSET)

        # Make sure EmptyExperiment completes normally now.
        loop.run_until_complete(empty_status_events["run_done"].wait())

        loop.run_until_complete(scheduler.stop())
```

sbourdeauducq, 2022-05-13:
Oh, I see. In this case it is probably simpler to have the test check relevant properties of the schedule instead of comparing it exactly to an expected schedule. Synchronizing the experiments to enforce a fully deterministic schedule would need complicated IPC.

sbourdeauducq, 2022-05-13:
> some of them do rely on implicitly defining order via `sleep`s.

Which ones? Only moninj tests do that (to ensure a scan and event report by the firmware) IIRC.

dnadlinger, 2022-05-13:
Oh, just checked the Git history, and apparently I did write the offending test myself two years ago – how embarassing (in fairness, I was probably just copying `test_steps()` without thinking about what ordering guarantees the scheduler actually makes).

The above is an example on using `get_dataset` to do simple synchronisation, but yes, just testing for the expected properties might be cleaner and shorter.

> Which ones [rely on sleep]? 

It might be that the 0.2 second intervals in which pause/check_pause are called by the test experiments don't actually matter, only some scheduler internals – it's been a while.

sbourdeauducq, 2022-05-13:
Those are just to prevent 100% CPU utilization while it waits for ``check_pause()`` to return True.

ljstephenson, 2022-05-13:
Agree on checking properties; given the current overlap between this test and both `test_steps()` and `test_pause()`, it seems like the relevant check is that rid 2 goes through all stages before rid 1 does anything?

Deepskyhunter, 2022-06-13:
> it seems like the relevant check is that rid 2 goes through all stages before rid 1 does anything?

It shouldn't be the relevant check because there are other factors which have higher priority than RID

[https://m-labs.hk/artiq/manual/management_system.html#priorities-and-timed-runs](url)

marmeladapk, 2022-07-15:
I had this test fail, when my laptop cpu was forced to lowest frequency and other processes were running. With proper frequency restored this test passed.

dnadlinger, 2024-07-10:
Should be fixed (without a degradation in the behaviour for this) in https://github.com/m-labs/artiq/commit/530f67f4cd44be0e633e402d1b8d92491c264459.

https://github.com/m-labs/artiq/pull/1976 would still be good to be able to test more complex scenarios more easily.

closed 2024-07-10 by dnadlinger