m-labs/artiq #1897: Make artiq handle coredevice power cycles gracefully

ljstephenson, 2022-05-27:
# ARTIQ Feature Request
## Problem this request addresses

At the moment, power cycling the core device while running an artiq master and all the trimmings seems to require restarting the master and any dashboards, for a couple of reasons:

* moninj proxy doesn't reconnect to core device
* haven't checked but I'm assuming core logging doesn't reconnect to core device either
* even if you restart the moninj proxy manually, the dashboard won't try to reconnect to the newly started proxy

This seems like it shouldn't be necessary, and involves some annoying overheads like waiting for the master to scan the experiment directory again.

## Describe the solution you'd like

A dashboard and/or artiq_client based trigger that tells the master to reconnect moninj proxy and core logging, and sends a broadcast signal to dashboards to reconnect to the moninj proxy.

## Additional context

I'm not sure how often other users are power cycling their core device; at the moment we are running the bleeding edge to get new features so we typically have to reset at least once per day on our live system, and many times per day on the test setup used for debugging, so this amounts to a reasonable amount of time for us.

sbourdeauducq, 2022-05-28:
Are you running ctlmgr? It should restart moninj_proxy automatically when it terminates.
In any case re-scanning the repository is not necessary if you restart only the dashboard and not the master.

ljstephenson, 2022-06-06:
We are running indeed running ctlmgr - I believe moninj_proxy does not exit if it loses a connection (although I'm not sure, and don't have an easily accessible system to test remotely). By restarting manually what I did was terminate the proxy and let the ctlmgr handle restarting it. Subsequently the dashboard does not reconnect to the restarted prozy.

As you say, after restarting core log and moninj proxy, the only thing left to do is restart the dashboards. It's not terribly tedious and we can definitely work around it; I think a trigger to handle restarting would be a decent middle ground solution.

sbourdeauducq, 2022-07-02:
@Deepskyhunter Please test the latest changes

PR #1923 (2022-06-23): Disconnect Proxy after connection failure with core device
Public the task in comm_moninj for the usage in proxy
Add log message when connected/reconnected

# ARTIQ Pull Request

## Description of Changes

### Related Issue #1897 

## Type of Changes

|   | Type |
| ------------- | ------------- |
| ✓  | :bug: Bug fix  |

## Steps (Choose relevant, delete irrelevant before submitting)

### All Pull Requests

- [x] Use correct spelling and grammar.
- [x] Update [RELEASE_NOTES.rst](../RELEASE_NOTES.rst) if there are noteworthy changes, especially if there are changes to existing APIs.
- [x] Clos [... cut]
PR #1924 (2022-07-02): Disconnect corelog properly when core device connection failed

# ARTIQ Pull Request

## Description of Changes
Replace asyncio.open_connection with async_open_connection that
has keepalive
Put get_logs_task inside asyncio.wait() for termination when
disconnected

### Related Issue #1897

## Type of Changes

|   | Type |
| ------------- | ------------- |
| ✓  | :bug: Bug fix  |

## Steps (Choose relevant, delete irrelevant before submitting)

### All Pull Requests

- [x] Use correct spelling and grammar.
- [x] Close/update issues.

### Code Changes

- [x] Run `flake8` to check code style (foll [... cut]