m-labs/artiq #1941: Urukul injection ignores switch and configuration state

jordens, 2022-07-27:
Since core_cache `_{cpld}_cfg` is only used by injection, everything else (i.e. all experiments using that urukul) is unaware of it and conversely injection will be unaware of what `cfg` the experiments set. That state becomes inconsistent immediately in practice. Instead of `core_cache`, read the switch state (`sta_read()`) for each injection and build `cfg` from that. Also acknowledge and document that (1) all other `cfg` changes relative to ddb will be overwritten, (2) the channel will be reset (all DDS chip configuration lost), and (3) if the switch TTL RTIO channel is used and set high, injection is incapable of turning of the channel.

https://github.com/m-labs/artiq/blob/2a7a72b27a093ae5287d5abcb2124eccdb5c1155/artiq/dashboard/moninj.py#L616-L626

jordens, 2022-07-27:
@Spaqin

Spaqin, 2022-07-27:
The injection was made with several limitations in mind, mostly that cfg register is not readable from Urukul itself...

``core_cache`` was used to determine whether or not CPLD needs initialization or not. Status register on its own won't tell me, as default values pre-init are as valid as they would be post validation. 

However you have a point, this can be improved. We can also check for abnormalities in the cfg if the cached version doesn't exist - or maybe even cache the status register and compare against it, and update the cache when an external experiment changed something as well.

In similar manner, I'm thinking if it's possible to determine whether a dds channel was initialized too (read from dds register?).

Either way, thank you for the report :)

jordens, 2022-07-27:
Right. Redesigning the CPLD gateware to remove those limitations is tracked at https://github.com/quartiq/urukul/issues/7