m-labs/artiq #1947: Kasli Error LED does not light up on panic

nkrackow, 2022-08-08:
# Bug Report

## One-Line Summary

The Kasli Error LED does not light up on an error, aka when the rust firmware panics. 

## Issue Details

### Steps to Reproduce

E.g. don't feed a clock to a  Kasli which expects an external reference at startup.

### Expected Behavior

The LED should light up.

### Actual (undesired) Behavior

```
Starting firmware.
[     0.000014s]  INFO(runtime): ARTIQ runtime starting...
[     0.003938s]  INFO(runtime): software ident 7.8116.eba143a;npl9master
[     0.010390s]  INFO(runtime): gateware ident 7.8116.eba143a;npl9master
[     0.016845s]  INFO(runtime): log level set to INFO by default
[     0.022574s]  INFO(runtime): UART log level set to INFO by default
[     0.139378s]  WARN(runtime::rtio_clocking): rtio_clock setting not recognised. Falling back to default.
[     0.147478s]  WARN(runtime::rtio_clocking): si5324_ext_ref and ext_ref_frequency compile-time options are deprecated. Please use the rtio_clock coreconfig settings instead.
[     0.162890s]  INFO(runtime::rtio_clocking): using 10MHz reference to make 125MHz RTIO clock with PLL
panic at runtime/rtio_clocking.rs:225:55: cannot initialize Si5324: "Si5324 misses clock input signal"
backtrace for software version 7.8116.eba143a;npl9master:
0x4001e660
0x4000aca0
0x4000a2a0
0x4001cf70
0x4000770c
0x4002fcc0
0x4002fc68
0x4001d7e8
halting.
use `artiq_coremgmt config write -s panic_reset 1` to restart instead
```

Error LED doesn't turn on.

### Your System (omit irrelevant parts)

* ARTIQ version: ARTIQ-7 [eba143a](https://github.com/m-labs/artiq/commit/eba143a47530adbd9d402a9736d0fdc06dcf836f)

jordens, 2022-08-08:
Note that this is a regression from a very useful behavior in release 6.

sbourdeauducq, 2022-08-10:
It's not a regression, it doesn't work on any ARTIQ version when DRTIO is enabled.

PR #1952 (2022-08-11): Add Error LED to the Kasli's MasterBase and SatelliteBase submodule

# ARTIQ Pull Request

## Description of Changes
add Error LED to the Kasli's MasterBase and SatelliteBase submodule.

Test setup
1. Get a Kasli board
2. Insert a `panic!()` statement into the main function of files artiq/firmware/runtime/main.rs and artiq/firmware/satman/main.rs to simulate the rust firmware panic.

| before changes | after changes |
|---------------- | -----------------|
| Error LED Off | Error LED On |

### Related Issue
Closes #1947 

## Type of Changes

|   | Type |
| ------------- | ------------- |
| ✓   [... cut]
closed 2022-08-11 by sbourdeauducq