m-labs/artiq #1948: LLVM IR parsing error

fanmingyu212, 2022-08-09:
# Bug Report

## One-Line Summary

Sometimes a kernel runs with a LLVM IR parsing error saying `error: value for 'column' too large, limit is 65535`. It typically happens in a kernel following another long-running kernel.

## Issue Details

### Steps to Reproduce

We have an experiment that raises a LLVM IR error when we choose parameters to run many repetitions in the `self.run_kernel()` function. The error happen in a separate kernel function to reset the dds states after the experiment finishes. The experiment does not fail when the number of repetitions are small. Only relevant code is attached as the entire experiment is too long and complicated to be attached, and I don't have a MWE for the error.

```python
from artiq import *

class TestExperiment(EnvExperiment):
    def run(self):
        self.run_kernel()  # run a very long kernel function(~10 mins)
        self.reset_sinara_hardware()

    @host_only
    def reset_sinara_hardware(self):
        """Sets all hardware back to pre-experiment values."""
        if len(self.devices.urukuls_used) > 0:
            self._reset_urukuls()

        if len(self.devices.ad9910s_used) > 0:
            params_used_ad9910s = []
            for kk in self.devices.ad9910s_used:
                params_used_ad9910s.append(self._preexp_dds_params[kk[0]])
            self._reset_ad9910s(params_used_ad9910s)  # experiment sometimes fails here.

    @kernel
    def _reset_urukuls(self):
        """Sets all urukuls back to profile 7 (the default profile)."""
        for name, urukul in self.devices.urukuls_used:
            self.core.break_realtime()
            delay_mu(self.rtio_cycle_mu)
            urukul.set_profile(7)

    @kernel
    def _reset_ad9910s(self, params):
        for kk in range(len(self.devices.ad9910s_used)):
            name, ad9910 = self.devices.ad9910s_used[kk]
            values = params[kk]
            self._set_dds(ad9910, values)

    @kernel(flags={"fast-math"})
    def _set_dds(self, device, values):
        self.core.break_realtime()
        delay_mu(self.rtio_cycle_mu)
        delay_mu(self.dds_set_delay_mu)
        device.set(values[0], values[1], values[2])
        self.core.break_realtime()
        device.get_att_mu()
        self.core.break_realtime()
        delay_mu(self.dds_set_delay_mu)
        device.set_att(values[3])
        self.core.break_realtime()
        if values[4] > 0.:
            device.sw.on()
        else:
            device.sw.off()
```

### Expected Behavior

Finishes with no error.

### Actual (undesired) Behavior

With certain parameters, it raises the following error:
```python
root:Terminating with exception (RuntimeError: LLVM IR parsing error)
Traceback (most recent call last):
File "c:\users\scientist\code\artiq\artiq\master\worker_impl.py", line 300, in main
exp_inst.run()
File "C:\Users\scientist\code\jax\base\experiments\scan.py", line 44, in run
self.host_cleanup()
File "C:\Users\scientist\code\jax\base\experiments\scan.py", line 53, in host_cleanup
self.reset_sinara_hardware()
File "C:\Users\scientist\code\jax\base\environments\sinara_environment.py", line 80, in reset_sinara_hardware
self._reset_ad9910s(params_used_ad9910s)
File "c:\users\scientist\code\artiq\artiq\language\core.py", line 54, in run_on_core
return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
File "c:\users\scientist\code\artiq\artiq\coredevice\core.py", line 131, in run
self.compile(function, args, kwargs, set_result)
File "c:\users\scientist\code\artiq\artiq\coredevice\core.py", line 114, in compile
library = target.compile_and_link([module])
File "c:\users\scientist\code\artiq\artiq\compiler\targets.py", line 199, in compile_and_link
return self.link([self.assemble(self.compile(module)) for module in modules])
File "c:\users\scientist\code\artiq\artiq\compiler\targets.py", line 199, in <listcomp>
return self.link([self.assemble(self.compile(module)) for module in modules])
File "c:\users\scientist\code\artiq\artiq\compiler\targets.py", line 154, in compile
llparsedmod = llvm.parse_assembly(str(llmod))
File "C:\Users\scientist\Anaconda3\envs\jlab\lib\site-packages\llvmlite\binding\module.py", line 25, in parse_assembly
raise RuntimeError("LLVM IR parsing error\n{0}".format(errmsg))
RuntimeError: LLVM IR parsing error
<string>:50970:29: error: value for 'column' too large, limit is 65535
!1878 = !DILocation(column: 65544, line: 1, scope: !8)
^
```

The error is deterministic with the parameters we choose.

### Your System (omit irrelevant parts)

* Operating System: Windows 10
* ARTIQ version: ARTIQ-7, on 3f812c4c
* Version of the gateware and runtime loaded in the core device: (in the output of ``artiq_coremgmt -D .... log``): Same as the ARTIQ version
* If using Conda, output of `conda list` (please submit as a file attachment, as this tends to be long): A relevant one is `llvmlite==0.38.0`.
* Hardware involved: Kasli 2.0 and EEM modules.

fanmingyu212, 2022-08-23:
I am able to confirm the same problem on the latest ARTIQ-7 (9aa81e12) with the corresponding firmware. We see the same problem on three setups (2x Kasli 2.0 and 1x Kasli-SoC). This error happens typically when running a kernel after a long kernel function as described in the above message.

fanmingyu212, 2022-08-23:
The latest ARTIQ-7 saves a `LLVM IR (BROKEN) DUMP` file though. The dump file is attached here: [tmpikxl060q.ll.txt](https://github.com/m-labs/artiq/files/9401306/tmpikxl060q.ll.txt). The error is related to line 68660: `!1272 = !DILocation(column: 65580, line: 1, scope: !8)`.

fanmingyu212, 2022-08-23:
We found that the error was due to a code mistake. When calling `self._reset_ad9910s(params_used_ad9910s)`, `params_used_ad9910s` was incorrectly set to a very long list (~thousands of elements) with a lot of repeated devices in the list elements. After fixing the problem the code works with no issue.

The error message is not clear for this problem though. At least this issue may help future users if someone make the same mistake as I did.

sbourdeauducq, 2022-08-23:
Do you have a minimal example to reproduce the issue? The compiler should not behave like this.

fanmingyu212, 2022-08-25:
MWE on Kasli 2.0:
```python
from artiq.experiment import *

class TestLLVM(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("urukul1_ch3")  # ad9910 dds

    def run(self):
        reset_cmds = [(self.urukul1_ch3, 257.0, 1.0) for kk in range(1000)]
        self.reset_ddses(reset_cmds)

    @kernel
    def reset_ddses(self, reset_commands):
        self.core.reset()
        for device, freq_MHz, ampl in reset_commands:
            device.set(freq_MHz * MHz, 0.0, ampl)
```

Error message:
```
root:Terminating with exception (RuntimeError: LLVM IR parsing error)
Traceback (most recent call last):
  File "c:\users\scientist\code\artiq\artiq\master\worker_impl.py", line 343, in main
    exp_inst.run()
  File "C:\Users\scientist\code\spock\experiments\run\tests\test_llvm.py", line 11, in run
    self.reset_ddses(reset_cmds)
  File "c:\users\scientist\code\artiq\artiq\language\core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "c:\users\scientist\code\artiq\artiq\coredevice\core.py", line 139, in run
    self.compile(function, args, kwargs, set_result)
  File "c:\users\scientist\code\artiq\artiq\coredevice\core.py", line 115, in compile
    library = target.compile_and_link([module])
  File "c:\users\scientist\code\artiq\artiq\compiler\targets.py", line 203, in compile_and_link
    return self.link([self.assemble(self.compile(module)) for module in modules])
  File "c:\users\scientist\code\artiq\artiq\compiler\targets.py", line 203, in <listcomp>
    return self.link([self.assemble(self.compile(module)) for module in modules])
  File "c:\users\scientist\code\artiq\artiq\compiler\targets.py", line 157, in compile
    llparsedmod = llvm.parse_assembly(str(llmod))
  File "C:\Users\scientist\anaconda3\envs\jlab\lib\site-packages\llvmlite\binding\module.py", line 25, in parse_assembly
    raise RuntimeError("LLVM IR parsing error\n{0}".format(errmsg))
RuntimeError: LLVM IR parsing error
<string>:11649:29: error: value for 'column' too large, limit is 65535
!1668 = !DILocation(column: 65600, line: 1, scope: !8)
                            ^
```