m-labs/artiq #1953: release-7: Phaser init underflows in some scenarios

airwoodix, 2022-08-10:
# Bug Report

## One-Line Summary

On `release-7`, the Phaser board initialization routine reproducibly triggers a RTIO underflow in some, but not all, scenarios.

## Issue Details

### Steps to Reproduce

Consider the following experiments:

```python
from artiq.experiment import *

class Basic(EnvExperiment):  # works
    def build(self):
        self.setattr_device("core")
        self.setattr_device("phaser0")

    @kernel
    def run(self):
        self.core.reset()
        self.phaser0.init()

class DeviceArray(EnvExperiment):  # fails
    def build(self):
        self.setattr_device("core")
        self.phasers = [self.get_device("phaser0")]

    @kernel
    def run(self):
        self.core.reset()
        self.phasers[0].init()

class PhaserAndUrukul(EnvExperiment):  # fails
    def build(self):
        self.setattr_device("core")
        self.setattr_device("phaser0")
        self.setattr_device("urukul0_cpld")

    @kernel
    def run(self):
        self.core.reset()
        self.phaser0.init()
        self.urukul0_cpld.init()
```

### Expected Behavior

All three experiments above run without errors.

### Actual (undesired) Behavior

Only the `Basic` experiment runs without errors. Both `DeviceArray` and `PhaserAndUrukul` fail with an RTIO underflow in the `Phaser.init()` method. The location reported in the coredevice traceback is the same in both failing cases.

Traceback for `DeviceArray` (RTIO channel 48 is that of the Phaser device):

```
# Core Device Traceback:
# Traceback (most recent call first):
#   File "ksupport/rtio.rs", line 69, column 14, in (Rust function)
#     <unknown>
#                  ^
#   File "sandbox/phaser/phaser_basic.py", line 23, in artiq_run_phaser_basic.DeviceArray.run(..., ...) (RA=+0x144)
#     self.phasers[0].init()
#   File "<artiq>/coredevice/phaser.py", line 245, in ... artiq.coredevice.phaser.Phaser.init<artiq.coredevice.phaser.Phaser>(...) (RA=+0x7b0)
#     config18 = self.dac_read(0x18)
#   File "<artiq>/coredevice/phaser.py", line 582, in ... artiq.coredevice.phaser.Phaser.dac_read<artiq.coredevice.phaser.Phaser>(...) (RA=+0x2668)
#     self.spi_cfg(select=PHASER_SEL_DAC, div=div, end=0)
#   File "<artiq>/coredevice/phaser.py", line 371, in ... artiq.coredevice.phaser.Phaser.write8<artiq.coredevice.phaser.Phaser>(...) (RA=+0x2f38)
#     rtio_output((self.channel_base << 8) | (addr & 0x7f) | 0x80, data)
#   File "<artiq>/coredevice/phaser.py", line 537, in ... artiq.coredevice.phaser.Phaser.spi_cfg<artiq.coredevice.phaser.Phaser>(...) (inlined)
#     self.write8(PHASER_ADDR_SPI_SEL, select)
# artiq.coredevice.exceptions.RTIOUnderflow(0): RTIO underflow at 2043332480036 mu, channel 48, slack -19804 mu

# End of Core Device Traceback
```

Traceback for `PhaserAndUrukul`:

```
# Core Device Traceback:
# Traceback (most recent call first):
#   File "ksupport/rtio.rs", line 69, column 14, in (Rust function)
#     <unknown>
#                  ^
#   File "sandbox/phaser/phaser_basic.py", line 54, in artiq_run_phaser_basic.PhaserAndUrukul.run(..., ...) (RA=+0x120)
#     self.phaser0.init()
#   File "<artiq>/coredevice/phaser.py", line 245, in ... artiq.coredevice.phaser.Phaser.init<artiq.coredevice.phaser.Phaser>(...) (RA=+0x704)
#     config18 = self.dac_read(0x18)
#   File "<artiq>/coredevice/phaser.py", line 582, in ... artiq.coredevice.phaser.Phaser.dac_read<artiq.coredevice.phaser.Phaser>(...) (RA=+0x2558)
#     self.spi_cfg(select=PHASER_SEL_DAC, div=div, end=0)
#   File "<artiq>/coredevice/phaser.py", line 371, in ... artiq.coredevice.phaser.Phaser.write8<artiq.coredevice.phaser.Phaser>(...) (RA=+0x2dfc)
#     rtio_output((self.channel_base << 8) | (addr & 0x7f) | 0x80, data)
#   File "<artiq>/coredevice/phaser.py", line 537, in ... artiq.coredevice.phaser.Phaser.spi_cfg<artiq.coredevice.phaser.Phaser>(...) (inlined)
#     self.write8(PHASER_ADDR_SPI_SEL, select)
# artiq.coredevice.exceptions.RTIOUnderflow(0): RTIO underflow at 2164145976108 mu, channel 48, slack -15412 mu

# End of Core Device Traceback
```

These experiments work as expected against ARTIQ v7.0.06ad76b.beta.

### Your System

* ARTIQ version: ARTIQ v7.0.eba143a
* Version of the gateware and runtime loaded in the core device: 7.0.eba143a
* Hardware involved: Kasli v1.1, Phaser v1.1, Urukul v1.3

airwoodix, 2022-08-11:
@nkrackow @jordens did you see something similar while working on #1933 or other Phaser-related changes? Or is the issue specific to `release-7` (unlikely given the divergence to the main branch)? Or to pre-servo Phaser gateware? (also looks very unlikely).

@sbourdeauducq have there been specific changes between 06ad76b and the `release-7` cut that would qualify to trigger such behavior? As-is, Phaser is only usable in isolation in `release-7`. It seems likely that the issue also exists on f312794 (is not `release-7`-specific) but it's hard for me to test because of #1930.

jordens, 2022-08-11:
We haven't seen this. But it may well be related to newer changes re numpy arrays and or the new CPU.

nkrackow, 2022-08-11:
I presume the `artiq_sinara_tester` with more than a phaser also fails?

sbourdeauducq, 2022-08-11:
It does not. But it's compiling kernels one by one for each tested device.

airwoodix, 2022-08-11:
Rebuilding the gateware/firmware against 97cba3f solved the issue. But I also couldn't reproduce the issue with an independent build against eba143a. Although I had built twice yesterday, this was probably an inconsistency in the build or in the dependencies.

I also understand from @jordens's comment that this has not been reproduced elsewhere.

I'll therefore close this issue. Thanks for the feedback and sorry for the noise.

closed 2022-08-11 by airwoodix