m-labs/artiq #1987: extreme SRTIO latency when SED depth is exceeded

jbqubit, 2022-11-05:
## One-Line Summary

The Serial Event Dispatcher (SED) is a mechanism that enables RTIO PHYs to use a shared pool of FIFOs for queuing RTIO events -- SED was proposed [here](https://github.com/m-labs/artiq/issues/778). The SED FIFOs supports some number of distinct RTIO timestamps (depth) and multiple simultaneous events at each timestamp (width). We observe that when events are scheduled that saturate the SED depth limit there is surprising and undesirable behavior. 

1. There is no error emitted. 
2. The core device experiences a poorly bounded, sometimes extreme delay. 
3. The core device has anomalous high latency when scheduling the first event. 

## Issue Details

In a test program (detailed below) a single RTIO channel is populated in a loop. After SED depth is saturated (event 64) the core device blocks for a 835 us. The second time the SED depth is saturated (event 128) the core device blocks for 63,901 us. The blocking continues with anomalous high durations at subsequent multiples of the SED depth. 

### Steps to Reproduce

```
from artiq.experiment import *
import numpy as np

class CoreDeviceProcessDurationsJWB(EnvExperiment):
  
   kernel_invariants = {"ftw0", "asf0", 'mindelay'}

   def build(self):
       self.trials = 200
       self.setattr_device("core")
       self.setattr_device("ttl7")
       self.ttl = self.ttl7
       self.urukul_channel = self.get_device('urukul1_ch3')
       self.urukul_switch = self.get_device('ttl_urukul1_sw3')
       self.zotino = self.get_device('zotino0')

       self.ftw0 = self.urukul_channel.frequency_to_ftw(100*MHz)
       self.asf0 = self.urukul_channel.amplitude_to_asf(0.85)
       self.mindelay = 8*ns

   def prepare(self):
       self.durations = np.full(self.trials, 0, dtype=np.int64)

   def print_results(self, test_name, durations):
       s = "### {} ### \n".format(test_name)
       mean = np.mean(durations)
       std = np.std(durations)
       maxdev = np.max(durations) - np.min(durations)
       maxfdev = (np.max(durations) - np.min(durations)) / np.mean(durations)
       deviations_found = 0
       for i in range(len(durations)):
           s += "{:3}: {:9}, ".format(i, durations[i])
           if (i+1)%8 == 0:
               s += '\n'
       s = s + "\nmean: {:9.0f}, std: {:9.0f}, max deviation: {:9.0f}, max frac deviation: {:.3f}\n".format(
           mean, std, maxdev, maxfdev)
       print(s)

   def run(self):
       tests = ['test_ttl_pulse']
       for test in tests:
           testf = getattr(self, test)
           durations = self.timer(testf)
           self.print_results(test, durations)

   @kernel
   def timer(self, kernel_method):
       self.core.reset()
       delay(1*ms)
       for i in range(self.trials):
           t1 = self.core.get_rtio_counter_mu()
           td = kernel_method()
           delay(td)  # default td=self.mindelay, required so kernel_method() is not, in general, optimized away 
           t2 = self.core.get_rtio_counter_mu()
           self.durations[i] = t2 - t1
       return self.durations

   @kernel
   def test_ttl_pulse(self):
       self.ttl.pulse(self.mindelay)
       return self.mindelay
```

### Expected Behavior

SED is not well documented so my expectations can't be said to deviate from technical specifications. Generally, I can say say what I do and don't expect.
* I don't expect the core device to block for longer than self.mindelay*depth
* I do expect the core device to resume stuffing the SED FIFOS as soon as there is space available

### Actual (undesired) Behavior

```
$ artiq_run repository/calibration/core_device_process_durations_jwb.py
### test_ttl_pulse ###
  0:  	4200,   1:  	1288,   2:  	1168,   3:   	952,   4:   	936,   5:   	936,   6:   	936,   7:   	936,
  8:   	936,   9:   	936,  10:   	936,  11:   	936,  12:   	936,  13:   	936,  14:   	936,  15:   	936,
 16:   	936,  17:   	936,  18:   	936,  19:   	936,  20:   	936,  21:   	936,  22:   	936,  23:   	936,
 24:   	936,  25:   	936,  26:   	936,  27:   	936,  28:   	936,  29:   	936,  30:   	936,  31:   	936,
 32:   	936,  33:   	936,  34:   	936,  35:   	936,  36:   	936,  37:   	936,  38:   	936,  39:   	936,
 40:   	936,  41:   	936,  42:   	936,  43:   	936,  44:   	936,  45:   	936,  46:   	936,  47:   	936,
 48:   	936,  49:   	936,  50:   	936,  51:   	936,  52:   	936,  53:   	936,  54:   	936,  55:   	936,
 56:   	936,  57:   	936,  58:   	936,  59:   	936,  60:   	936,  61:   	936,  62:   	936,  63:   	936,
 64:	835600,  65:   	936,  66:   	936,  67:   	936,  68:   	928,  69:   	936,  70:   	936,  71:   	936,
 72:   	936,  73:   	936,  74:   	936,  75:   	936,  76:   	936,  77:   	936,  78:   	936,  79:   	936,
 80:   	936,  81:   	936,  82:   	936,  83:   	936,  84:   	936,  85:   	936,  86:   	936,  87:   	936,
 88:   	936,  89:   	936,  90:   	936,  91:   	936,  92:   	936,  93:   	936,  94:   	936,  95:   	936,
 96:   	936,  97:   	936,  98:   	936,  99:   	936, 100:   	936, 101:   	936, 102:   	936, 103:   	936,
104:   	936, 105:   	936, 106:   	936, 107:   	936, 108:   	936, 109:   	936, 110:   	936, 111:   	928,
112:   	936, 113:   	936, 114:   	936, 115:   	936, 116:   	936, 117:   	936, 118:   	936, 119:   	936,
120:   	936, 121:   	936, 122:   	936, 123:   	936, 124:   	936, 125:   	936, 126:   	936, 127:   	936,
128:  63901288, 129:   	936, 130:   	936, 131:   	936, 132:   	936, 133:   	936, 134:   	936, 135:   	936,
136:   	928, 137:   	936, 138:   	936, 139:   	936, 140:   	936, 141:   	936, 142:   	936, 143:   	936,
144:   	936, 145:   	936, 146:   	936, 147:   	936, 148:   	936, 149:   	936, 150:   	936, 151:   	936,
152:   	936, 153:   	936, 154:   	936, 155:   	936, 156:   	936, 157:   	936, 158:   	936, 159:   	936,
160:   	936, 161:   	936, 162:   	936, 163:   	936, 164:   	936, 165:   	936, 166:   	928, 167:   	928,
168:   	936, 169:   	936, 170:   	936, 171:   	936, 172:   	936, 173:   	936, 174:   	936, 175:   	936,
176:   	936, 177:   	936, 178:   	936, 179:   	936, 180:   	936, 181:   	936, 182:   	936, 183:   	936,
184:   	936, 185:   	936, 186:   	936, 187:   	936, 188:   	936, 189:   	936, 190:   	936, 191:   	936,
192:   	936, 193:  64900232, 194:   	936, 195:   	936, 196:   	936, 197:   	936, 198:   	936, 199:   	936,

mean:	649127, std:   6407745, max deviation:  64899304, max frac deviation: 99.979 
```

### Your System (omit irrelevant parts)

* Operating System:
* ARTIQ version: latest commit of tag version 7
* Environment is nix shell
* Hardware involved: Kasli v1.x

jbqubit, 2022-11-07:
Yes, that makes sense for ~1 ms delay at rep 64. However, I don't understand the > 60 ms delays at 128 and 193.

jbqubit, 2022-11-09:
Let me try this again with an clearer example. 

```
from artiq.experiment import *
import numpy as np

class CoreDeviceComputationDelaysJWB(EnvExperiment):
    
    kernel_invariants = {"ftw0", "asf0", 'std_delay'}

    def build(self):
        self.trials = 70
        self.setattr_device("core")
        self.setattr_device("ttl8")
        self.ttl = self.ttl8
        self.urukul_channel = self.get_device('urukul0_ch0')
        self.urukul_switch = self.get_device('ttl_urukul0_sw0')
        #self.zotino = self.get_device('zotino0')

        self.ftw0 = self.urukul_channel.frequency_to_ftw(100*MHz)
        self.asf0 = self.urukul_channel.amplitude_to_asf(0.85)
        self.std_delay = 1*us

    def prepare(self):
        self.rtios = np.full(self.trials, 0, dtype=np.int64)
        self.nows = np.full(self.trials, 0, dtype=np.int64)
        self.slack = np.full(self.trials, 0, dtype=np.int64)

    def print_results(self):
        sheader = "\n{:>3s}: {:>12s} {:>12s} {:>12s} {:>12s} {:>12s}\n".format(
            'rep', 'now','delta now', 'rtio', 'delta rtio', 'slack')
        s = sheader
        for i in range(self.trials-1):
            s += "{:3}: {:12,.0f} {:12,.0f} {:12,.0f} {:12,.0f} {:12,.0f}\n".format(
                i, 
                self.nows[i], self.nows[i]-self.nows[i+1],
                self.rtios[i], self.rtios[i]-self.rtios[i+1],
                self.slack[i]
            )
            if (i+1)%35 == 0:
                s += sheader
        print(s)

    def run(self):
        self.ktest()
        self.print_results()

    @kernel
    def ktest(self):
        self.core.reset()
        delay(-100*us)
        rtio0 = self.core.get_rtio_counter_mu()
        now0 = now_mu()
        for i in range(self.trials):
            self.ttl.pulse(1*us)
            delay(99*us)
            self.rtios[i] = self.core.get_rtio_counter_mu() - rtio0
            self.nows[i] = now_mu() - rtio0
            self.slack[i] = now_mu() - self.core.get_rtio_counter_mu()
```
```
$ artiq_run repository/core_device_process_durations_jwb.py 

rep:          now    delta now         rtio   delta rtio        slack
  0:      121,984     -100,000        7,080       -5,848      111,552
  1:      221,984     -100,000       12,928       -2,104      208,120
  2:      321,984     -100,000       15,032       -2,032      306,080
  3:      421,984     -100,000       17,064       -1,928      404,160
  4:      521,984     -100,000       18,992       -2,080      502,072
  5:      621,984     -100,000       21,072       -1,920      600,160
[... 47 lines cut ...]
 51:    5,221,984     -100,000      113,424       -1,920    5,107,800
 52:    5,321,984     -100,000      115,344       -2,184    5,205,624
 53:    5,421,984     -100,000      117,528       -1,920    5,303,696
 54:    5,521,984     -100,000      119,448       -1,928    5,401,776
 55:    5,621,984     -100,000      121,376       -1,936    5,499,840
 56:    5,721,984     -100,000      123,312       -1,960    5,597,880
 57:    5,821,984     -100,000      125,272       -1,920    5,695,952
 58:    5,921,984     -100,000      127,192       -1,920    5,794,040
 59:    6,021,984     -100,000      129,112       -1,920    5,892,112
 60:    6,121,984     -100,000      131,032       -1,920    5,990,200
 61:    6,221,984     -100,000      132,952       -1,928    6,088,264
 62:    6,321,984     -100,000      134,880       -2,016    6,186,256
 63:    6,421,984     -100,000      136,896       -1,920    6,284,328
 64:    6,521,984     -100,000      138,816       -1,944    6,382,384
 65:    6,621,984     -100,000      140,760      -83,864    6,480,472
 66:    6,721,984     -100,000      224,624       -1,984    6,496,536
 67:    6,821,984     -100,000      226,608       -1,944    6,594,592
 68:    6,921,984     -100,000      228,552       -1,936    6,692,664

```
The default Kasli DIO FIFO depth is 128. Each pulse requires two FIFO slots (on and off). Here's where what I observe first breaks my expectation. 
* Rep 65 blocks on scheduling ttl.pulse(1*us) -- this makes sense since the FIFO  is full until after wall time 221,640.
* Rep 66 should block on scheduling ttl.pulse(1*us) until the wall time passes 321,984. This is because the FIFO is full until after 321,984.

I don't see how reps 66-68 can be scheduled with DRTIO.... the FIFO is full.

occheung, 2025-03-07:
> I don't see how reps 66-68 can be scheduled with DRTIO.... the FIFO is full.

Yes but the other 7 aren't. See the lane switch logic in the release-7 branch when event spreading is enabled (by default).
https://github.com/m-labs/artiq/blob/3bd0502030b0b60450eb37af827b91b2b49db5b4/artiq/gateware/rtio/sed/lane_distributor.py#L174-L184

Lane switch can be triggered by a falling edge of `writable`. Consuming an event from a full FIFO achieves this.

Event spread in release-8 needs to be enabled manually. When enabled, a sufficiently-filled FIFO will trigger a lane switch.
See docs for details. https://m-labs.hk/artiq/manual/rtio.html#event-spreading